raspberrypi/linux #5168: Unable to choose Fully preemtible model in version 5.15.65

Larsinator7, 2022-09-08:
### Describe the bug

I'm trying to get a fully preemtible model to work on a Raspberry Pi using cross compiling. I've encountered an issue on the newest version of the github repo that I'm unable to choose the option of the fully preemtible model in the menuconfig. I'm trying to add a fully preemtible kernel on rpi version 5.15.65 with the corresponding patch. Any reasons why this problem might occur, alternatively what corresponding version and patches will work? I've tried several but I get different issues. 

### Steps to reproduce the behaviour

Downloading the github repo

### Device (s)

Other

### System

Kernel version: 5.15.55-rt48 #2 SMP PREEMPT_RT Mon Sep 5 19:09:07 CEST 2022 x86_64 x86_64 x86_64 GNU/Linux
Done on a host computer running Ubuntu version 22.04

### Additional context

I did the patch of the Kernel on my computer myself if that is relevant.

Larsinator7, 2022-09-08:
PS: Could it be the patch that is troublesome?

pelwell, 2022-09-08:
It works for me:
```
$ diff 1 2
108,109c108,111
< CONFIG_PREEMPT_VOLUNTARY=y
< # CONFIG_PREEMPT is not set
---
> # CONFIG_PREEMPT_VOLUNTARY is not set
> CONFIG_PREEMPT=y
> CONFIG_PREEMPT_COUNT=y
> CONFIG_PREEMPTION=y
131a134
> CONFIG_PREEMPT_RCU=y
135a139
> CONFIG_TASKS_RCU=y
749,753c753
< CONFIG_INLINE_SPIN_UNLOCK_IRQ=y
< CONFIG_INLINE_READ_UNLOCK=y
< CONFIG_INLINE_READ_UNLOCK_IRQ=y
< CONFIG_INLINE_WRITE_UNLOCK=y
< CONFIG_INLINE_WRITE_UNLOCK_IRQ=y
---
> CONFIG_UNINLINE_SPIN_UNLOCK=y
4150a4151
> # CONFIG_CEC_GPIO is not set
8088a8090
> CONFIG_DEBUG_PREEMPT=y
8166a8169
> # CONFIG_PREEMPT_TRACER is not set
```
(where `1` is the result of `make ARCH=arm bcm2711_defconfig` and `2` is after `make ARCH=arm menuconfig` and changing to `Preemptible Kernel`).

Which sequence of commands did you run?

Larsinator7, 2022-09-08:
I did:

$ gzip -cd ../patch-5.15.65-rt49.patch.gz | patch -p1 --verbose
$ make O=../kernel-out/ ARCH=arm64 CROSS_COMPILE=/opt/aarch64/bin/aarch64-linux-gnu- bcm2711_defconfig
$ make O=../kernel-out/ ARCH=arm64 CROSS_COMPILE=/opt/aarch64/bin/aarch64-linux-gnu- menuconfig

Patch version is [[patch-5.15.65-rt49.patch.gz](https://mirrors.edge.kernel.org/pub/linux/kernel/projects/rt/5.15/patch-5.15.65-rt49.patch.gz)

and it just didn't show up as an option in the menu

pelwell, 2022-09-08:
I think the RT patch may have replaced or disabled the `CONFIG_PREEMPT` option. If you've installed the RT patches. wouldn't you want to enable `CONFIG_PREEMPT_RT` instead?

Larsinator7, 2022-09-08:
I went into the config file and tried to enable the CONFIG_PREEMPT_RT option but this doesn't seems to exist in the config file with the current patch. Atleast I can't find it when searching for it

nbuchwitz, 2022-09-08:
Try with CONFIG_KVM=n. See link in https://github.com/RevolutionPi/linux/commit/5418e26a472d42324a75614df8cb9c9fc6ff5587 for more details

Larsinator7, 2022-09-08:
Thank you so much. This resolved my error and I was able to choose it in the menuconfig.

closed 2022-09-08 by Larsinator7