raspberrypi/linux #5169: Singapore default language is English not Chinese

AjayShanker-geek, 2022-09-09:
### Describe the bug

I wanted to inform that every time I set my country to Singapore on Raspberry Pi OS, the default language will be Chinese. Our Country main language is English, and btw we are not part of China. This has been bugging me since childhood.

Cheers!

### Steps to reproduce the behaviour

On initial boot

### Device (s)

Raspberry Pi 4 Mod. B

### System

Latest Raspbian OS

pelwell, 2022-09-09:
I understand your frustration, but I don't think the situation is quite as you see it. Assuming you are talking about the "Welcome to Raspberry Pi" wizard that greets users on first use, the list of languages for each country is sorted alphabetically, not in order of importance. For example, the languages listed for Ukraine are, in order, Crimean Tatar, Russian and Ukrainian, and I don't think anyone would claim that Crimean Tatar is the main language of Ukraine. The list box is displayed with the first entry as the default value. In your case, Chinese comes before English in the roman alphabet/ASCII ordering, hence Chinese is the default.

pelwell, 2022-09-09:
Not that it's only because of the use of the term "British English" that the default language for the UK isn't Cornish!

AjayShanker-geek, 2022-09-10:
Ohh my bet, thanks for the clarification. Cheers