raspberrypi/linux #5171: CONFIG_NFT_XFRM is not set

hailfinger, 2022-09-10:
### Describe the bug

The default kernel in Debian Bullseye has CONFIG_NFT_XFRM=m.
The kernel in Raspberry Pi OS has CONFIG_NFT_XFRM not set.

This causes the following error when using nftables rules with ipsec matches:
`/etc/nftables.conf:52:17-29: Error: Could not process rule: No such file or directory
                ipsec out spi 1-65536 accept comment "VPN encapsulated traffic"
                ^^^^^^^^^^^^^`

The same rule works fine in debian.

### Steps to reproduce the behaviour

Use the following content for /etc/nftables.conf:

`table ip filter {
        chain output {
                type filter hook output priority 0; policy accept;

                ipsec out spi 1-65536 accept comment "VPN encapsulated traffic"
        }
}`

Run
`nft -f /etc/nftables.conf`

### Device (s)

Raspberry Pi 4 Mod. B

### System

# cat /etc/rpi-issue
Raspberry Pi reference 2022-09-06
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 827affcc11aaf7aa577d15daf02fb40b64392380, stage2
# vcgencmd version
Aug 26 2022 14:04:10 
Copyright (c) 2012 Broadcom
version 102f1e848393c2112206fadffaaf86db04e98326 (clean) (release) (start_x)
# uname -a
Linux pi-test-cdh4 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux

### Logs

No dmesg output for the bug.

### Additional context

Having the same packet filtering features as the Debian Bullseye kernel would be highly appreciated. I'm using a few hundred Raspberry Pi 4B in a VPN deployment and I'd love to use nftables for packet filtering instead of relying on iptables to get the "will this traffic be sent via VPN" matching functionality.

Thank you!

pelwell, 2022-09-12:
Done - see https://github.com/raspberrypi/linux/commit/e621efdde748f4ade7969d91fe3a9924ca0ac603.

hailfinger, 2022-09-12:
Thank you! Will test later today.

hailfinger, 2022-09-13:
Tested, works.
Thank you!

closed 2022-09-13 by pelwell

hailfinger, 2022-09-13:
I noticed a typo in the changelog. It's CONFIG_NFT_XFRM, not CONFIG_NET_XFRM.
Probably only relevant for trees which get rebased (linux-6.0) as changing changelogs of already committed trees is a can of worms.

pelwell, 2022-09-13:
It's even less relevant for trees that get rebased because all the defconfig patches get squashed into one commit.