raspberrypi/linux #5172: CVE-2022-32250 Linux Kernel privilege escalation vulnerability, a use-after-free bug - unclear situation

fst-net, 2022-09-12:
### Describe the bug

Dear Raspi Team,
since June I'm waiting of a patch for **CVE-2022-32250**
Linux Kernel privilege escalation vulnerability, a use-after-free bug that was found in the Linux kernel’s net/netfilter/nf_tables_api.c.

It looks like that isn't patched yet.

In stock Debian this flaw is patched long time ago, of course for Kernel 5.10 an 5.18,
[https://security-tracker.debian.org/tracker/CVE-2022-32250](https://security-tracker.debian.org/tracker/CVE-2022-32250)
but RaspiOS uses Kernel 5.15?

One of my four Raspis is normally online with ssh access for external users. Because of this security flaw it must going offline for three month until now.

A fix or reply with a clarification would be nice.
See more information about it:
[https://securityonline.info/cve-2022-32250-linux-kernel-privilege-escalation-vulnerability/](https://securityonline.info/cve-2022-32250-linux-kernel-privilege-escalation-vulnerability/)

Best Regards
Macomaniac

### Steps to reproduce the behaviour

as normal user pi
`unshare -Unr /bin/bash`
looks to get a root shell

and as root
`capsh --decode=$(grep CapEff /proc/self/status | cut -f2)`
`0x0000000000000000=`
list still the vulnarable cap_net_admin right.

I try also the latest kernel via rpi-update with no difference (see below under system).

### Device (s)

Raspberry Pi 2 Mod. B

### System

Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage5

Aug 26 2022 14:04:36 
Copyright (c) 2012 Broadcom
version 102f1e848393c2112206fadffaaf86db04e98326 (clean) (release) (start)

Linux Raspi-2 5.15.65-v7+ #1583 SMP Wed Sep 7 15:40:11 BST 2022 armv7l GNU/Linux

### Additional context

Davin from the Raspberrypi forum said I should open an issue here on github.

pelwell, 2022-09-12:
As an LTS kernel, 5.15 received the back-port of the fix on Jun 6 2022 as part of 5.15.45 - see https://github.com/raspberrypi/linux/commit/f692bcffd1f2ce5488d24fbcb8eab5f351abf79d. This was merged into rpi-5.15.y on the same day (https://github.com/raspberrypi/linux/commit/b52941068bcc4f64501e4719103d3e4000ed4873) and first appeared in the rpi-firmware releases the following day, hitting the stable branch on August 11 2022.

fst-net, 2022-09-12:
Dear kralo,
Yes you didn't get a real root shell, but I didn't found another test as the both I posted first.

It seems you are German speaking. It was discussed first at the heise Forum here:
[https://www.heise.de/forum/heise-online/Kommentare/Fehler-in-Linux-Kernel-ermoeglicht-Rechteausweitung/Leider-ungenaue-bis-falsche-Beschreibung-der-Rahmenbedingungen-Mitigation/posting-41113548/show/](https://www.heise.de/forum/heise-online/Kommentare/Fehler-in-Linux-Kernel-ermoeglicht-Rechteausweitung/Leider-ungenaue-bis-falsche-Beschreibung-der-Rahmenbedingungen-Mitigation/posting-41113548/show/)

What I know is, that on some other systems the unshare "Test" didn't work anymore after the kernel was patched.

I only want to be sure that Raspios with Kernel 5.15.x is safe, because I do not find anything about this security flaw directly belonging to Raspios before.
Regards

kralo, 2022-09-12:
Reading the link you posted, changing into the namespace is a prerequisite to trigger the bug, not the bug.
The bug is fixed in the linux-mainline kernel and the change is in the raspi-kernel, as pointed out.
Unless you are going to compile and try the POC, I wouldn't bother researching this further.
The raspi should be safe.

fst-net, 2022-09-12:
I think yes that I try before and after patch, but 3 month ago, I can't swear an oath on it. ;-)

I don't want to make a POC. After pelwell's and your statement, I would trust, that the actual RaspiOS Kernel 5.15.x is safe. So my little system can go back online.
Thanks for your help.

fst-net, 2022-09-12:
I just figured out, that my Test System (Raspi 2) was bumped by me via rpi-update to Kernel 5.15.65 in between watching incoming updates.
But my online system (Raspi 1 powered back on after 3 month) is still at Kernel 5.10.
Linux 5.10.103+ #1529 Tue Mar 8 12:19:18 GMT 2022 armv6l GNU/Linux

It seems, that the mainline RaspiOS get no Kernel Update to 5.15 or a newer patched 5.10 within the standard apt dist-upgrade procedure.

But before I reopen my question to Kernel 5.10 I will use also rpi-update to go to 5.15 line at this system too.

closed 2022-09-13 by fst-net