raspberrypi/linux #5174: RPi4 DSI interface

billru05, 2022-09-16:
### Describe the bug

I am needing help with a RPi4 on getting the DSI interface activated so I can use a (OSOYOO 5 Inch DSI Touch Sceen)
I am using a (Raspberry Pi Compute Module 4, 1GB, 16GB - CM4001016) its attached to a (Waveshare Mini 
Base Board for Raspberry Pi Compute Module ( I have tried both 4CM4-IO-BASE-A and 4CM4-IO-BASE-B).
I have tried the 32bit & 64 bitSystem: 32-bit Kernel version:5.15 Debian version: 11 (bullseye) desktop
and lite version also tried the Raspberry Pi OS (Legacy).
The screen dose not boot its stays black as its not connected, I tried the screen on a RPI3 B+ and it boots
correctly but for some reason I can not figuare out how to activate the DSI on the RPi4

### Steps to reproduce the behaviour

flashing, boot Pi without anything connected, once on network, shut down, connect DSI screen then reboot but the DSI screen never boots

### Device (s)

Raspberry Pi CM4

### System

Raspberry Pi OS with desktop

    Release date: September 6th 2022
    System: 64-bit
    Kernel version: 5.15
    Debian version: 11 (bullseye)

6by9, 2022-09-16:
DSI is not a plug-and-play interface as it has no generic mechanism for probing attached devices.
The Raspberry Pi DSI display has additional I2C connectivity to provide power control, backlight, and access to the touchscreen driver IC.

CM4 makes no assumptions about GPIO usage, therefore does not define any default connections for displays, cameras, or anything else. Documentation for configuring that with the legacy stack is at https://www.raspberrypi.com/documentation/computers/compute-module.html#attaching-the-official-7-inch-display.

With Bullseye and vc4-kms-v3d, the simplest step is to add the additional line ```dtoverlay=vc4-kms-dsi-7inch``` to config.txt to explicitly enable the display. This is all the firmware is doing if ```display_auto_detect=1``` is set, and a display is found based on the fimware configuration.
Almost all the instructions in https://osoyoo.com/2019/09/20/instruction-for-raspberry-pi-5-dsi-touch-screen/#5.3 are out of date for Bullseye.

JamesH65, 2022-09-16:
This documentation may help explain.

https://pip.raspberrypi.com/categories/685/documents/RP-003472-WP/RP-003472-WP-1.pdf

6by9, 2022-09-16:
> This documentation may help explain.
https://pip.raspberrypi.com/categories/685/documents/RP-003472-WP/RP-003472-WP-1.pdf

The OSOYOO 5 Inch DSI Touch Sceen is a clone of the Pi 7" display, so getting into the detail of implementing drivers shouldn't be necessary in this case.
It is a 3rd party panel, therefore Raspberry Pi offer no direct support for it. Should the cloning be less than perfect, it is not our problem to solve.