raspberrypi/linux #5177: Package ceph-fuse is missing

DavidZidar, 2022-09-20:
### Describe the bug

I want to use `ceph-fuse` which is available in Debian but it is missing.

### Steps to reproduce the behaviour

`apt install ceph-fuse`

### Device (s)

Raspberry Pi 3 Mod. B

### System

## OS
Raspberry Pi reference 2018-10-09
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7b40fd11bc13d6496bb6bd1bc02f946b7e0ac7cb, stage2

## Firmware
Aug 26 2022 14:04:36
Copyright (c) 2012 Broadcom
version 102f1e848393c2112206fadffaaf86db04e98326 (clean) (release) (start)

## Kernel
Linux jarvis 5.15.61-v7+ #1579 SMP Fri Aug 26 11:10:59 BST 2022 armv7l GNU/Linux

### Additional context

I asked about `ceph-fuse` in another related issue but was told the question was just going to be ignored so I'm opening this issue.

DavidZidar, 2022-09-20:
Thanks for the response, my first post was just a question, nothing more. I didn't realize this was the wrong place so thanks for pointing that out.

Regarding `ceph-fuse`, I've been having issues getting the kernel module to work properly, the FUSE module on the other hand just works and is easier to configure to boot. Since Ceph support is missing on Raspberry Pi I'm currently using an NFSv4 mount instead but it's not been reliable.

XECDesign, 2022-09-21:
> I want to use `ceph-fuse` which is available in Debian but it is missing.

It's only available in Debian (and Ubuntu) for amd64. If there's no reason it can't be built for armhf and/or arm64, it should be reported to Debian.

DavidZidar, 2022-09-21:
@XECDesign Oh I see, thanks. I'll have a look there then. But just so I understand, does that mean that if Debian fixes it upstream it would automatically be picked up by Raspbian at some point down the line?

XECDesign, 2022-09-21:
Yes, but also if the maintainers of the package verify that there's not reason for it to be disabled, we can pull it into Raspberry Pi OS earlier.

DavidZidar, 2022-09-21:
@XECDesign Cool, I understand, thanks for the help.

closed 2022-09-21 by DavidZidar