raspberrypi/linux #5178: No signal using vc4-kms-v3d dtoverlay on pi4

giovariot, 2022-09-20:
### Describe the bug

I'm using a raspberry pi 4B, unless I set dtoverlay to use vc4-fkms-v3d I can't manage to get any signal through the HDMI after the raspberry boot logo with the kms driver. Last raspbian version then doesn't really work unless I fix the config.txt file. Older raspbian versions such as the stable version seem to work as expected (apart from some hiccups such as the system getting stuck at boot showing the "_" at the top left of the screen, behaviour that also applies to the newer version too sometimes, even using fkms

I also have the same problem on libreelec version 10.0.1 onwards, which I think it might be using this linux kernel, but I'm gonna ask on their forum too. Their version 10.0.0 seems to work fine though but I think it was using an older graphics driver.

At first I thought it was because I was dual booting the two using PINN but even flashing the sdcard with raspian alone I get the same result.

I tried resetting the EEPROM a few times, tried different AC adapter and tried different sdcards and two different TVs getting the same result.

### Steps to reproduce the behaviour

- reset the EEPROM
- flash raspbian last version
- boot it

### Device (s)

Raspberry Pi 4 Mod. B

### System

```
$ cat /etc/rpi-issue
Raspberry Pi reference 2022-09-06
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 66255495f29be5d09b765d081aff6fc0f11e59b4, stage4
```

```
$ vcgencmd version
Aug 26 2022 14:03:16 
Copyright (c) 2012 Broadcom
version 102f1e848393c2112206fadffaaf86db04e98326 (clean) (release) (start)
```

```
$ uname -a
Linux raspberrypi 5.15.61-v7l+ #1579 SMP Fri Aug 26 11:13:03 BST 2022 armv7l GNU/Linux
```

### Logs

This is the log using vc4-fkms-v3d as I can't get the dmesg while using kms.
I'll try to add one through ssh when I'll be able to do so...

```
$ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.15.61-v7l+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1579 SMP Fri Aug 26 11:13:03 BST 2022
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] random: crng init done
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ac00000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   Normal   empty
[    0.000000]   HighMem  [mem 0x0000000030000000-0x00000000fbffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] percpu: Embedded 16 pages/cpu s35788 r8192 d21556 u65536
[    0.000000] pcpu-alloc: s35788 r8192 d21556 u65536 alloc=16*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1011008
[... 380 lines cut ...]
[   11.477425] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   16.235448] systemd-journald[137]: File /var/log/journal/13c7dcc8cae14fa8bf0df2919ba3a428/user-1000.journal corrupted or uncleanly shut down, renaming and replacing.
[   17.305227] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   17.366697] Bluetooth: Core ver 2.22
[   17.366762] NET: Registered PF_BLUETOOTH protocol family
[   17.366770] Bluetooth: HCI device and connection manager initialized
[   17.366788] Bluetooth: HCI socket layer initialized
[   17.366798] Bluetooth: L2CAP socket layer initialized
[   17.366815] Bluetooth: SCO socket layer initialized
[   17.376172] Bluetooth: HCI UART driver ver 2.3
[   17.376189] Bluetooth: HCI UART protocol H4 registered
[   17.376254] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   17.376417] Bluetooth: HCI UART protocol Broadcom registered
[   17.658973] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   17.658989] Bluetooth: BNEP filters: protocol multicast
[   17.659002] Bluetooth: BNEP socket layer initialized
[   17.867167] NET: Registered PF_ALG protocol family
[   18.341901] cryptd: max_cpu_qlen set to 1000
[   18.639093] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net.ipv6.neigh.wlan0.retrans_time - use net.ipv6.neigh.wlan0.retrans_time_ms instead
[   18.761105] Bluetooth: RFCOMM TTY layer initialized
[   18.761141] Bluetooth: RFCOMM socket layer initialized
[   18.761166] Bluetooth: RFCOMM ver 1.11
[   31.833854] cam-dummy-reg: disabling
[   44.737409] v3d fec00000.v3d: MMU error from client L2T (0) at 0x9921000, pte invalid
```

giovariot, 2022-09-20:
Well... I've enabled ssh and rebooted just to check everything was alright, still using the fkms driver but this time I simply got a black screen...

```
pi@raspberrypi:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.15.61-v7l+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1579 SMP Fri Aug 26 11:13:03 BST 2022
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] random: crng init done
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ac00000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   Normal   empty
[    0.000000]   HighMem  [mem 0x0000000030000000-0x00000000fbffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] percpu: Embedded 16 pages/cpu s35788 r8192 d21556 u65536
[    0.000000] pcpu-alloc: s35788 r8192 d21556 u65536 alloc=16*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1011008
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@60 smsc95xx.macaddr=DC:A6:32:32:68:2B vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=d9e688b1-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] Unknown kernel command line parameters "splash", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x0000000014890000-0x0000000018890000] (64MB)
[    0.000000] Memory: 3601324K/4050944K available (10240K kernel code, 1385K rwdata, 3304K rodata, 2048K init, 592K bss, 121940K reserved, 327680K cma-reserved, 3264512K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 35501 entries in 105 pages
[    0.000000] ftrace: allocated 105 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000000] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000018] Switching to timer-based delay loop, resolution 18ns
[    0.000366] Console: colour dummy device 80x30
[    0.000429] printk: console [tty1] enabled
[    0.000497] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.000528] pid_max: default: 32768 minimum: 301
[    0.000667] LSM: Security Framework initializing
[    0.000879] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.000908] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.002507] cgroup: Disabling memory control group subsystem
[    0.002756] CPU: Testing write buffer coherency: ok
[    0.002814] CPU0: Spectre BHB: enabling loop workaround for all CPUs
[    0.003283] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.004661] Setting up static identity map for 0x200000 - 0x20003c
[    0.004887] rcu: Hierarchical SRCU implementation.
[    0.005886] smp: Bringing up secondary CPUs ...
[    0.007098] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.008441] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.009753] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.009910] smp: Brought up 1 node, 4 CPUs
[    0.009934] SMP: Total of 4 processors activated (432.00 BogoMIPS).
[    0.009953] CPU: All CPU(s) started in HYP mode.
[    0.009967] CPU: Virtualization extensions available.
[    0.010773] devtmpfs: initialized
[... 824 lines cut ...]
[  489.433166]  r4:c1205048
[  489.433178] [<bf38d5ac>] (vc4_fkms_connector_get_modes [vc4]) from [<bf2e9b8c>] (drm_helper_probe_single_connector_modes+0x1c4/0x814 [drm_kms_helper])
[  489.433498]  r7:c275e8d8 r6:00000001 r5:c3c99e5c r4:bf30a01c
[... cut ...]
[  489.435756] [<c046f830>] (sys_ioctl) from [<c0200040>] (ret_fast_syscall+0x0/0x1c)
[  489.435778] Exception stack(0xc3c99fa8 to 0xc3c99ff0)
[  489.435794] 9fa0:                   00000001 becf78f0 0000000c c05064a7 becf78f0 becf7940
[  489.435812] 9fc0: 00000001 becf78f0 c05064a7 00000036 0054ec18 0000000c becf78f0 0020f100
[  489.435827] 9fe0: b6d8c08c becf78d4 b6d71c5c b69eed4c
[  489.435844]  r10:00000036 r9:c3c98000 r8:c0200244 r7:00000036 r6:c05064a7 r5:becf78f0
[  489.435857]  r4:00000001
```

I'll post another dmesg using kms soon

giovariot, 2022-09-20:
Here is using vc4-kms-v3d

```
pi@raspberrypi:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.15.61-v7l+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1579 SMP Fri Aug 26 11:13:03 BST 2022
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] random: crng init done
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ac00000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   Normal   empty
[    0.000000]   HighMem  [mem 0x0000000030000000-0x00000000fbffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] percpu: Embedded 16 pages/cpu s35788 r8192 d21556 u65536
[    0.000000] pcpu-alloc: s35788 r8192 d21556 u65536 alloc=16*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1011008
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@60 smsc95xx.macaddr=DC:A6:32:32:68:2B vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=d9e688b1-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] Unknown kernel command line parameters "splash", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x0000000014890000-0x0000000018890000] (64MB)
[    0.000000] Memory: 3601324K/4050944K available (10240K kernel code, 1385K rwdata, 3304K rodata, 2048K init, 592K bss, 121940K reserved, 327680K cma-reserved, 3264512K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 35501 entries in 105 pages
[    0.000000] ftrace: allocated 105 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] irq_brcmstb_l2: registered L2 intc (/soc/interrupt-controller@7ef00100, parent irq: 25)
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000000] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000018] Switching to timer-based delay loop, resolution 18ns
[    0.000365] Console: colour dummy device 80x30
[    0.000430] printk: console [tty1] enabled
[    0.000499] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.000530] pid_max: default: 32768 minimum: 301
[    0.000670] LSM: Security Framework initializing
[    0.000880] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.000909] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.002498] cgroup: Disabling memory control group subsystem
[    0.002749] CPU: Testing write buffer coherency: ok
[    0.002806] CPU0: Spectre BHB: enabling loop workaround for all CPUs
[    0.003272] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.004664] Setting up static identity map for 0x200000 - 0x20003c
[    0.004868] rcu: Hierarchical SRCU implementation.
[    0.005885] smp: Bringing up secondary CPUs ...
[    0.007092] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.008428] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.009732] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.009916] smp: Brought up 1 node, 4 CPUs
[    0.009940] SMP: Total of 4 processors activated (432.00 BogoMIPS).
[    0.009958] CPU: All CPU(s) started in HYP mode.
[    0.009971] CPU: Virtualization extensions available.
[... 381 lines cut ...]
[   17.705404]  MAI: soc_pcm_open() failed (-19)
[   17.730906] hdmi-audio-codec hdmi-audio-codec.2.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   17.730937]  MAI: soc_pcm_open() failed (-19)
[   17.731375] hdmi-audio-codec hdmi-audio-codec.2.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   17.731392]  MAI: soc_pcm_open() failed (-19)
[... cut ...]
[   17.777614] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   17.777630]  MAI: soc_pcm_open() failed (-19)
[   17.848648] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net.ipv6.neigh.wlan0.retrans_time - use net.ipv6.neigh.wlan0.retrans_time_ms instead
[   17.929816] Bluetooth: RFCOMM TTY layer initialized
[   17.929835] Bluetooth: RFCOMM socket layer initialized
[   17.929854] Bluetooth: RFCOMM ver 1.11
[   31.833294] cam-dummy-reg: disabling
```

popcornmix, 2022-09-21:
When booting with fkms, post output of `vcdbg log msg`
Also post output of
```
tvservice -s
tvservice -m CEA
tvservice -m DMT
tvservice -d edid.dat
base64 edid.dat
```

giovariot, 2022-09-21:
funny, it only correctly boots through fkms only if the time before I've tried booting using kms, if I'll reboot from a fkms "X session" to another fkms session all i get is a black screen (not a no signal one, just a black blank screen).

Nonetheless I managed to correctly boot fkms and here is the output of the commands you suggested

output of vcdbg log msc
```
$ sudo vcdbg log msg
005703.378: arasan: arasan_emmc_open
005852.393: brfs: File read: /mfs/sd/config.txt
005853.336: brfs: File read: 2076 bytes
006028.519: HDMI1:EDID error reading EDID block 0 attempt 0
006029.533: HDMI1:EDID giving up on reading EDID block 0
006031.458: brfs: File read: /mfs/sd/config.txt
006479.705: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
006482.220: *** Restart logging
006482.240: brfs: File read: 2076 bytes
006647.433: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
006648.467: hdmi: HDMI1:EDID giving up on reading EDID block 0
006748.681: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
006749.732: hdmi: HDMI1:EDID giving up on reading EDID block 0
006749.767: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
006749.795: HDMI0: hdmi_pixel_encoding: 300000000
006749.822: HDMI1: hdmi_pixel_encoding: 300000000
006766.650: dtb_file 'bcm2711-rpi-4-b.dtb'
006778.398: brfs: File read: /mfs/sd/bcm2711-rpi-4-b.dtb
006778.432: Loaded 'bcm2711-rpi-4-b.dtb' to 0x100 size 0xcaf3
006811.049: brfs: File read: 51955 bytes
006825.827: brfs: File read: /mfs/sd/overlays/overlay_map.dtb
007005.993: brfs: File read: 2043 bytes
007009.470: brfs: File read: /mfs/sd/config.txt
007011.458: brfs: File read: 2076 bytes
007027.174: brfs: File read: /mfs/sd/overlays/rpi-tv.dtbo
007045.807: Loaded overlay 'rpi-tv'
007045.822: Loaded HAT overlay
007045.840: dtparam: audio=on
007109.267: brfs: File read: 567 bytes
007129.238: brfs: File read: /mfs/sd/overlays/vc4-fkms-v3d-pi4.dtbo
007173.227: Loaded overlay 'vc4-fkms-v3d'
007275.733: brfs: File read: 1446 bytes
007278.449: brfs: File read: /mfs/sd/cmdline.txt
007278.529: Read command line from file 'cmdline.txt':
007278.556: 'console=serial0,115200 console=tty1 root=PARTUUID=d9e688b1-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles'
007645.853: brfs: File read: 148 bytes
009001.810: brfs: File read: /mfs/sd/kernel7l.img
009001.849: Loaded 'kernel7l.img' to 0x8000 size 0x6b6ca0
009001.914: Device tree loaded to 0x2eff2c00 (size 0xd331)
009009.068: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
012515.557: vchiq_core: vchiq_init_state: slot_zero = 0xdad80000, is_master = 1
012519.094: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
012526.153: TV service:host side not connected, dropping notification 0x00000002, 0x00000001, 0x00000010
018245.126: TV service:host side not connected, dropping notification 0x00000001, 0x00000002, 0x00000000
018291.572: TV service:host side not connected, dropping notification 0x00000002, 0x00000001, 0x00000010
020980.887: TV service:host side not connected, dropping notification 0x00000001, 0x00000002, 0x00000000
021024.234: TV service:host side not connected, dropping notification 0x00000002, 0x00000001, 0x00000010
021587.518: TV service:host side not connected, dropping notification 0x00000008, 0x00000004, 0x00000000
023478.345: TV service:host side not connected, dropping notification 0x00000001, 0x00000002, 0x00000000
023521.165: TV service:host side not connected, dropping notification 0x00000002, 0x00000004, 0x00000000
023533.007: TV service:host side not connected, dropping notification 0x00000001, 0x00000002, 0x00000000
023575.812: TV service:host side not connected, dropping notification 0x00000002, 0x00000004, 0x00000000
028704.792: TV service:host side not connected, dropping notification 0x00000001, 0x00000002, 0x00000000
028752.243: TV service:host side not connected, dropping notification 0x00000002, 0x00000004, 0x00000000
028761.869: TV service:host side not connected, dropping notification 0x00000001, 0x00000002, 0x00000000
028803.465: TV service:host side not connected, dropping notification 0x00000002, 0x00000004, 0x00000000
029019.260: TV service:host side not connected, dropping notification 0x00000001, 0x00000002, 0x00000000
029063.586: TV service:host side not connected, dropping notification 0x00000002, 0x00000004, 0x00000000
031457.516: TV service:host side not connected, dropping notification 0x00000001, 0x00000002, 0x00000000
031501.078: TV service:host side not connected, dropping notification 0x00000002, 0x00000004, 0x00000000
```

tvservice -s
[... 40 lines cut ...]
           mode 28: 1280x800 @ 60Hz 16:10, clock:83MHz progressive 
           mode 32: 1280x960 @ 60Hz 4:3, clock:108MHz progressive 
           mode 35: 1280x1024 @ 60Hz 5:4, clock:108MHz progressive 
           mode 36: 1280x1024 @ 75Hz 5:4, clock:135MHz progressive 
           mode 39: 1360x768 @ 60Hz 16:9, clock:85MHz progressive 
           mode 47: 1440x900 @ 60Hz 16:10, clock:106MHz progressive 
           mode 48: 1440x900 @ 75Hz 16:10, clock:136MHz progressive 
           mode 58: 1680x1050 @ 60Hz 16:10, clock:146MHz progressive 
```

tvservice -d edid.dat
```
$ tvservice -d edid.dat
Written 256 bytes to edid.dat
```

base64 edid.dat
```
$ base64 edid.dat
AP///////wBMLXoIAAAAABkVAQOAPCJ4CrpBoVlVnSgNUFS/74BxT4EAgUCBgJUAlQ+zAAEBAjqA
GHE4LUBYLEUAVlAhAAAeZiFQsFEAGzBAcDYAVlAhAAAeAAAA/QAYSxpRFwAKICAgICAgAAAA/ABT
TVQyN0EzMDAKICAgAdACAyPxS5AfBBMFFAMSICEiIwkHB4MBAADiAA9nAwwAIAC4LQEdAHJR0B4g
bihVAKBaAAAAHgEdALxS0B4guChVQKBaAAAAHgEdgBhxHBYgWCwlAKBaAAAAngEdgNByHBYgECwl
gKBaAAAAngAAAAAAAAAAAAAAAAAAAAAAAAAA/Q==
```

popcornmix, 2022-09-22:
> funny, it only correctly boots through fkms only if the time before I've tried booting using kms (of course only getting the no signal screen), if I'll reboot from a fkms "X session" to another fkms session all i get is a black screen (not a no signal one, just a black blank screen).

In the black screen case, can you still ssh in and run commands? If so the output of the previous commands would be useful.
I'm suspecting edid reading issues, although on the successful run it worked okay.

I'm curious if in the fkms case you copy edid.dat to /boot and add to config.txt:
```
hdmi_edid_file=1
hdmi_force_hotplug=1
hdmi_group=1
hdmi_mode=16
```

Also in the kms case (with the fkms config.txt test lines removed), copy edid.dat to /lib/firmware and add to cmdline.txt (end of existing line)
```
drm.edid_firmware=edid.dat video=HDMI-A-1:1920x1080@60D
```
and report if either case is reliable.

Also have you tried a different hdmi cable? Are you using a case (such as the argon one) that reroutes hdmi?

giovariot, 2022-09-22:
## using just `vc4-fkms-v3d`

I kept the session open from yesterday, copied edid.dat to `/boot` and to `/lib/firmware`.

Just to check everything was(n't) working as expected I tried rebooting from the fkms X session before editing the config file. Got to a "_" on the top left of the screen: in this case it seems that tvservice gets stuck, can't even get out of the command using CTRL+C
```
pi@raspberrypi:~ $ sudo vcdbg log msc
vcdbg log - unrecognized command: 'msc'
pi@raspberrypi:~ $ tvservice -s
```
And it doesn't seem to want to reboot from "sudo reboot". Forced reboot a few times trying to get to the "black" screen but really couldn't reach it.

But anyways... of course at the next reboot it boots correctly to X, the next reboot gets me to the stuck "_"
- - -
## using the hdmi options
I edit the config.txt file adding what you told me

> ```
> hdmi_edid_file=1
> hdmi_force_hotplug=1
> hdmi_group=1
> hdmi_mode=16
> ```

now it's finally booting correctly every time
- - -
## using kms
Edited:
- config.txt commenting the aforementioned `hdmi` options and corrected `vc4-fkms-v3d` --> `vc4-kms-v3d` as per default
- cmdline.txt adding `drm.edid_firmware=edid.dat video=HDMI-A-1:1920x1080@60D`.

so now it works perfectly at every reboot even using kms.
- - -
## About my config
- tried both with my parents' tv (which is actually the one I want to configure the system for)
- tried 2 different hdmi cables (the one here and the one at my parents', which had been working with this same rpi for an year or so
- the case is the standard plastic raspberry one

- - -
## Two new questions
1. why is this happening both on this tv and my parents' one? Why did old raspbian images use to boot correctly every time and now they seem to work every now and then both on this tv and my parents' one? And why does PINN bootloader, for example, working as expected at every boot without needing to alter its config in any way?
2. now it works on my tv (thanks a lot @popcornmix!) but I'm only using it on my tv as I tried to setup it at my parent's place only to get to a no signal screen. Now I'd like to make it work on theirs. What can I do?

I was thinking to load an old raspbian version just to extract the `edid.dat` from my parents' tv and the export it to both raspbian and libreelec. Would that work?

Thanks a lot for helping me!

popcornmix, 2022-09-23:
> pi@raspberrypi:~ $ sudo vcdbg log msc
> vcdbg log - unrecognized command: 'msc'

This should be `sudo vcdbg log msg` (as you ran before).

> I was thinking to load an old raspbian version just to extract the `edid.dat` from my parents' tv and the export it to both raspbian and libreelec. Would that work?

Yes, the edid will be the same however you boot, so you can capture it from any operating system.
Libreelec has a slightly different scheme of [capturing the edid](https://wiki.libreelec.tv/configuration/edid)

All the evidence so far suggests a hardware issue in reading the edid. That could be in the cables, the TV or the Pi itself (the cable is most common cause, the TV less common and I've never hard of a Pi fault causing this, but it is possible).
If the TV has multiple HDMI ports, then try them all.

kms relies on the edid completely. fkms is more tolerant and often displays a picture without a valid edid, but it's just guessing what the TV actually supports. And you'll probably find issues with audio, cec and ability to change hdmi mode.

I can't think of a reason why newer fkms is failing but older works. It has been deprecated a long time so doesn't get any code changes. If you can capture "vcgencmd version" for a working image and a not working image (ideally find the oldest working/newest not-working) then I may be able to speculate if there's been any relevant change.

glyndon, 2023-03-18:
FWIW,  this same thing happened to me today, after upgrading RasPiOS to kernel 6.1.
I'd had the vc4-kms-v3d overlay enabled on that machine/display combo for years, so long that I've forgotten why.
Today the screen went blank at the same point yours did during boot. I swapped to a newer monitor and it worked fine.
But I wasn't satisfied, and googling the name of that overlay brought me to this issue report.
Something about that older [Dell] monitor wasn't compatible with the latest version of that overlay.
Removing the overlay did the trick. I wonder what I'm missing with it gone? Will keep reading ...

popcornmix, 2023-06-26:
Start a new issue please.

It's not clear what your issue is, but be aware that tvservice (and the fkms driver) are deprecated.
So please report what goes wrong with the (default) kms driver.
Reporting any difference in behaviour of fkms driver may be useful for diagnosing the issue, but any fixes are likely to come through the kms driver.

seaders, 2023-06-26:
No need now actually, I've figured my issue out.  I actually needed only the hotplug fix, but not for hdmi 1, for hdmi 2.  Issue with following the above steps for me was, because tvservice only works when in fkms mode, it was showing everything as hdmi 1, so, I thought I was dealing with hdmi 1.

When I finally copped that, I first changed A-1 to A-2 in cmdline.txt, then I changed hdmi_force_hotplug=1 to hdmi_force_hotplug:1=1, rebooted and pruned, and pruned again, and then everything was perfect.

Just a thing though, you said that tvservice is deprecated, and, sure enough it doesn't really work with the kms driver, but what's the alternatiive?  Specifically to try do what you suggest above for edid?  But I wasn't even able to find where that info was put out.  Played with kmsprint, as suggested by trying to use tvservice, but no luck.

popcornmix, 2023-06-27:
raspinfo is a debug tool that include a lot of information about display. Look at the output of that.

It can get all the information that tvservice does using standard linux interfaces when using kms.
It uses kmsprint to get info about modes supported, and `/sys/class/drm/card*/edid` to get the edid.