raspberrypi/linux #5182: No serial console on CM4 (5.19)

juergh, 2022-09-27:
### Describe the bug

With current rpi-5.19.y there is no serial console on a CM4 unless commandline option `8250.nr_uarts=1` is added explicitly. This seems to have been introduced by commit b9e62329e096c78dbf0c8aa1409acacfafd9fce1.

### Steps to reproduce the behaviour

Boot kernel built from rpi-5.19.y branch.

### Device (s)

Raspberry Pi CM4

### System

rpi-5.19.y

### Logs

```
[    1.866359] bcm2708_fb soc:fb: Unable to determine number of FBs. Disabling driver.
[    1.866390] bcm2708_fb: probe of soc:fb failed with error -2
[    1.937327] bcm2835-aux-uart fe215040.serial: there is not valid maps for state default
[    1.937663] bcm2835-aux-uart fe215040.serial: error -ENOSPC: unable to register 8250 port
[    1.937701] bcm2835-aux-uart: probe of fe215040.serial failed with error -28
[    1.939472] iproc-rng200 fe104000.rng: hwrng registered
[    1.939896] vc-mem: phys_addr:0x00000000 mem_base=0x3ec00000 mem_size:0x40000000(1024 MiB)
```

juergh, 2022-09-27:
See PR #5183.

pelwell, 2022-09-27:
You're right - CM4 and CM4S are both wrong, in opposite directions. CM4S is like CM3 in that it should be using UART0 as the console UART, and yet it includes `8250.nr_uarts=1` in its bootargs (partial command line), while as you've seen CM4 selects UART1 for the console but is missing `8250.nr_uarts=1`. I'll fix the CM4S bootargs so everything is consistent.

(But the fact that `8250.nr_uarts=1` is necessary in the first place in the Device Tree-equipped 21st century is absurd).

pelwell, 2022-09-27:
See ed40d79 for the companion patch (which will be forward-ported to the newer branches).

pelwell, 2022-09-27:
It should all be done now.