raspberrypi/linux #5185: IMX519.c - Compile Error |  error: 'MEDIA_BUS_FMT_SENSOR_DATA' undeclared

ds2k5, 2022-09-28:
### Describe the bug

Hi,
try to compile the IMX519 for Kernel 5.19.x on a SDM845 (SoC)
https://gitlab.com/sdm845-mainline/linux/

**Found the Source for IMX519 in your Repo.**

https://github.com/raspberrypi/linux/blob/rpi-5.19.y/drivers/media/i2c/imx519.c

**But the build fail - somebody so kind to help me ( I am noob and no developer - sorry )**

drivers/media/i2c/Kconfig
```

config VIDEO_IMX519
        tristate "Sony IMX519 sensor support"
        depends on OF_GPIO
        depends on I2C && VIDEO_DEV
        select VIDEO_V4L2_SUBDEV_API
        select MEDIA_CONTROLLER
        select V4L2_FWNODE
        help
          This is a Video4Linux2 sensor driver for the Sony
          IMX519 camera.

          To compile this driver as a module, choose M here: the
          module will be called imx519.

```

drivers/media/i2c/Makefile

`obj-$(CONFIG_VIDEO_IMX519) += imx519.o
`

.config

`CONFIG_VIDEO_IMX519=m
`

When do "make modules"
```

CC [M]  drivers/net/ipa/ipa_qmi.o
  CC [M]  drivers/net/ipa/ipa_qmi_msg.o
  CC [M]  drivers/rtc/rtc-pcf85063.o
  CC [M]  drivers/net/ipa/ipa_sysfs.o
  CC [M]  drivers/i2c/busses/i2c-cadence.o
  CC [M]  drivers/media/i2c/imx519.o
  CC [M]  drivers/net/ipa/ipa_data-v3.1.o
  CC [M]  drivers/rtc/rtc-pm8xxx.o
  CC [M]  drivers/i2c/busses/i2c-gpio.o
drivers/media/i2c/imx519.c: In function 'imx519_open':
drivers/media/i2c/imx519.c:1130:30: error: 'MEDIA_BUS_FMT_SENSOR_DATA' undeclared (first use in this function)
 1130 |         try_fmt_meta->code = MEDIA_BUS_FMT_SENSOR_DATA;
      |                              ^~~~~~~~~~~~~~~~~~~~~~~~~
drivers/media/i2c/imx519.c:1130:30: note: each undeclared identifier is reported only once for each function it appears in
  CC [M]  drivers/net/ipa/ipa_data-v3.5.1.o
drivers/media/i2c/imx519.c: In function 'imx519_enum_mbus_code':
drivers/media/i2c/imx519.c:1279:30: error: 'MEDIA_BUS_FMT_SENSOR_DATA' undeclared (first use in this function)
 1279 |                 code->code = MEDIA_BUS_FMT_SENSOR_DATA;
      |                              ^~~~~~~~~~~~~~~~~~~~~~~~~
drivers/media/i2c/imx519.c: In function 'imx519_enum_frame_size':
drivers/media/i2c/imx519.c:1306:34: error: 'MEDIA_BUS_FMT_SENSOR_DATA' undeclared (first use in this function)
 1306 |                 if (fse->code != MEDIA_BUS_FMT_SENSOR_DATA || fse->index > 0)
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  drivers/i2c/busses/i2c-qcom-cci.o
drivers/media/i2c/imx519.c: In function 'imx519_update_metadata_pad_format':
drivers/media/i2c/imx519.c:1342:28: error: 'MEDIA_BUS_FMT_SENSOR_DATA' undeclared (first use in this function)
 1342 |         fmt->format.code = MEDIA_BUS_FMT_SENSOR_DATA;
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~~
drivers/media/i2c/imx519.c: In function 'imx519_get_pad_format':
drivers/media/i2c/imx519.c:1364:33: error: 'MEDIA_BUS_FMT_SENSOR_DATA' undeclared (first use in this function)
 1364 |                                 MEDIA_BUS_FMT_SENSOR_DATA;
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~~
make[3]: *** [scripts/Makefile.build:249: drivers/media/i2c/imx519.o] Error 1
make[2]: *** [scripts/Makefile.build:466: drivers/media/i2c] Error 2
make[1]: *** [scripts/Makefile.build:466: drivers/media] Error 2
make[1]: *** Waiting for unfinished jobs....
  CC [M]  drivers/net/ipa/ipa_data-v4.2.o
  CC [M]  drivers/i2c/busses/i2c-qcom-geni.o
  CC [M]  drivers/net/ipa/ipa_data-v4.5.o
  CC [M]  drivers/net/ipa/ipa_data-v4.9.o
  CC [M]  drivers/net/ipa/ipa_data-v4.11.o
  LD [M]  drivers/net/ipa/ipa.o
make: *** [Makefile:1847: drivers] Error 2
 
```

### Steps to reproduce the behaviour

git clone https://gitlab.com/sdm845-mainline/linux/

cd linux

modify the files:  drivers/media/i2c/Kconfig & drivers/media/i2c/Makefile   (see above in Descripe the bug)

add the file:  imx519.c to drivers/media/i2c

make distclean
make ARCH=arm64 defconfig sdm845.config

modify the .config in source root dir (see above in Descripe the bug)

ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- make modules

### Device (s)

Other

### System

$ uname -a
Linux 5.19.7-sdm845 #1 SMP PREEMPT Wed Sep 17 18:35:19 CEST 2022 aarch64 GNU/Linux

6by9, 2022-09-28:
Mainline Linux has not merged support for CSI2 embedded metadata yet.

A plain IMX519 driver without that support has been sent to the linux-media list, and that should work against any tree. See https://patchwork.linuxtv.org/project/linux-media/patch/20220927114630.00004a4c@arducam.com/

Please note that the IMX519 driver is contributed by a third party (Arducam) and not Raspberry Pi. The level of support for it provided by Raspberry Pi personnel is low (basic questions only).

pelwell, 2022-09-28:
In other words, there will be a set of commits in rpi-5.19.y that you can cherry-pick on top of your Qualcomm kernel to add IMX519 support, but it may not be small, and there is the possibility of merge conflicts. Good luck in your non-Raspberry Pi-related task.

closed 2022-09-28 by pelwell