raspberrypi/linux #5193: Raspberry Pi 4: USB Wifi Adapter Hangs

leezu, 2022-10-02:
### Describe the bug

With USB Wifi devices including the CF-953AX (mt7921au) random hangs can occur when used with RPi. The hangs occur under heavy load, for example when operating the device as access point and running a speedtest + in parallel downloading many small files on a client [1]. This suggests further unmitigated firmware bugs in the the RPi USB Host Controller VL805. Disabling USB scatter gather via `/sys/module/mt76_usb/parameters/disable_usb_sg` appears to reduce the frequency of the issue, but the issue can still be reproduced with scatter gather disabled.

The issue has also been discussed at https://github.com/openwrt/mt76/issues/405 (for mt7612u devices) and https://github.com/morrownr/USB-WiFi/issues/107#issuecomment-1260976813 (for mt7921au devices)

### Steps to reproduce the behaviour

1. Obtain a CF-953AX and configure it as access point on a RPi [1]
2. Connect a client device via WiFi and run speedof.me on the client and in parallel download everything from nytimes.com via https://addons.mozilla.org/en-US/firefox/addon/downthemall/ from the same client. You may need to do this a few times in a row and perhaps connect other devices to the access points as well. To observe any hangs, you can run `ping -i 0.2 google.com` on the client.
3. Observe that the access point hangs and no more communication between client and access point happens
4. Observe the error in the kernel log on the RPi. If not error is displayed (yet) despite the hang, you can speed up printing the error by restarting the AP via `sudo systemctl restart hostapd@wlxe0e1a934a6a9`)

Optional: Disable USB Scatter Gather for the Mediatek USB Adapters. (The error can still be reproduced, but appear to occur less frequently)
```
% sudo cat /etc/modprobe.d/mt76_usb.conf
options mt76_usb disable_usb_sg=1
```

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspberry Pi reference 2021-05-07
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, dcfd74d7d1fa293065ac6d565711e9ff891fe2b8, stage2

Firmware: 
Aug 26 2022 14:03:16 
Copyright (c) 2012 Broadcom
version 102f1e848393c2112206fadffaaf86db04e98326 (clean) (release) (start)

Kernel: Linux raspberrypi 6.0.0-rc7-v8+ #4 SMP PREEMPT Wed Sep 28 02:12:59 UTC 2022 aarch64 GNU/Linux
 built from rpi-6.0.y https://github.com/raspberrypi/linux/commit/3fb5ca89c844eb9fda924a5e76ffab9e8c068675 which includes the https://github.com/raspberrypi/linux/pull/5173 USB fixes. (This issue also happened with earlier kernels that did not include the #5173 fixes)

### Logs

```
[ 1621.743502] mt7921u 2-1:1.3: Message 00020002 (seq 1) timeout
[ 1622.039473] mt7921u 2-1:1.3: timed out waiting for pending tx
[ 1622.053536] ------------[ cut here ]------------
[ 1622.053555] WARNING: CPU: 2 PID: 963 at kernel/kthread.c:659 kthread_park+0xb4/0xd0
[ 1622.053589] Modules linked in: xt_mark nft_chain_nat ctr aes_arm64 aes_generic ccm xt_MASQUERADE iptable_nat ip6table_nat nf_nat tun mt7921u mt7921_common mt76_connac_lib vc4 mt76_usb mt76 mac80211 snd_soc_hdmi_codec drm_display_helper cec drm_cma_helper drm_kms_helper brcmfmac btusb btrtl btintel btbcm bluetooth snd_soc_core brcmutil libarc4 sg cfg80211 ecdh_generic snd_compress snd_bcm2835(C) snd_pcm_dmaengine ecc snd_pcm bcm2835_isp(C) libaes bcm2835_codec(C) bcm2835_v4l2(C) rpivid_hevc(C) videobuf2_vmalloc bcm2835_mmal_vchiq(C) v3d v4l2_mem2mem videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common gpu_sched snd_timer rfkill vc_sm_cma(C) videodev drm_shmem_helper snd raspberrypi_hwmon mc syscopyarea sysfillrect sysimgblt fb_sys_fops nvmem_rmem uio_pdrv_genirq uio ip6t_REJECT nf_reject_ipv6 xt_hl ip6_tables ip6t_rt ipt_REJECT nf_reject_ipv4 xt_comment xt_multiport nft_limit xt_limit xt_addrtype xt_tcpudp xt_conntrack nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4
[ 1622.053938]  nft_compat nf_tables nfnetlink drm fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
[ 1622.053986] CPU: 2 PID: 963 Comm: kworker/u8:2 Tainted: G         C         6.0.0-rc7-v8+ #4
[ 1622.053999] Hardware name: Raspberry Pi 4 Model B Rev 1.4 (DT)
[ 1622.054007] Workqueue: mt76 mt7921_mac_reset_work [mt7921_common]
[ 1622.054053] pstate: 00000005 (nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[ 1622.054065] pc : kthread_park+0xb4/0xd0
[ 1622.054075] lr : mt76u_stop_tx+0x278/0x330 [mt76_usb]
[ 1622.054101] sp : ffffffc008f83c50
[ 1622.054107] x29: ffffffc008f83c50 x28: 0000000000000000 x27: ffffff8040838848
[ 1622.054127] x26: 0000000000000000 x25: ffffff8048c77280 x24: ffffff8040832068
[ 1622.054145] x23: ffffff8040834820 x22: ffffff8040836020 x21: ffffff8040832048
[ 1622.054162] x20: ffffff8049730580 x19: ffffff80456fdd00 x18: 0000000000000000
[ 1622.054179] x17: ffffffa650f08000 x16: ffffffdaa98b4e40 x15: 0000000000004000
[ 1622.054196] x14: 0000000000000000 x13: 00000000000000ba x12: 0000000000000000
[ 1622.054214] x11: 00000000000000ba x10: 0000000000001a90 x9 : ffffffda44eca8a8
[ 1622.054231] x8 : ffffff804af777f0 x7 : 0000000000000001 x6 : ffffffdaaad0b0c0
[ 1622.054248] x5 : ffffffdaaaba9000 x4 : ffffffdaaaba90b0 x3 : 0000000000002800
[ 1622.054265] x2 : 0000000000000000 x1 : 0000000000000000 x0 : 0000000000000004
[ 1622.054282] Call trace:
[ 1622.054288]  kthread_park+0xb4/0xd0
[ 1622.054299]  mt76u_stop_tx+0x278/0x330 [mt76_usb]
[ 1622.054320]  mt7921u_mac_reset+0x88/0x2d8 [mt7921u]
[ 1622.054336]  mt7921_mac_reset_work+0xac/0x1a0 [mt7921_common]
[ 1622.054364]  process_one_work+0x1dc/0x450
[ 1622.054382]  worker_thread+0x154/0x450
[ 1622.054392]  kthread+0x104/0x110
[... 150 lines cut ...]
# Maximum number of stations allowed
max_num_sta=255

# Bit field: bit0 = WPA, bit1 = (WPA2 or WPA3)
wpa=2
# Bit field: 1=wpa, 2=wep, 3=both
auth_algs=1

# Set of accepted cipher suites; disabling insecure TKIP
wpa_pairwise=CCMP
# Set of accepted key management algorithms
wpa_key_mgmt=WPA-PSK
wpa_passphrase=REDACTED

# hostapd event logger configuration
logger_stdout=-1
logger_stdout_level=2

# QoS support
wmm_enabled=1

# Use "iw list" to show device capabilities and modify ht_capab accordingly
ht_capab=[LDPC][HT40+][GF][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935]
vht_capab=[MAX-MPDU-7991][RXLDPC][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN]
```

P33M, 2022-10-03:
Please post the full output of `sudo lsusb -v`.

leezu, 2022-10-03:
```
Bus 002 Device 003: ID 7825:a2a4 Other World Computing External SATA Hard Drive Adapter cable PA023U3
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         9
  idVendor           0x7825 Other World Computing
  idProduct          0xa2a4 External SATA Hard Drive Adapter cable PA023U3
  bcdDevice           41.01
  iManufacturer           1 ULT-Best
  iProduct                2 Best USB Device
  iSerial                 3 032105264ABA
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0079
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              896mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              15
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              15
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       1
      bNumEndpoints           4
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     98 
      iInterface              0 
      Endpoint Descriptor:
[... 813 lines cut ...]
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x02
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0507 highspeed power suspend enable connect
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

leezu, 2022-10-04:
I'm unable to trigger the Wifi Adapter hang (yet?) when connecting the CF-953AX wifi adapter on usb3 extension lead but the SATA adapter on the RPi **usb2** port. Above error logs were all with CF-953AX on usb3 extension lead but the SATA adapter on the RPi **usb3** port. 

As "CF-953AX on usb3 extension lead but the SATA adapter on the RPi **usb3** port" is established as triggering errors upon unplugging the SATA adapter in https://github.com/raspberrypi/linux/issues/5192#issuecomment-1266364205 this raises the hypothesis that this Wifi adapter hanging issue and the USB disconnect causes failure on other USB devices #5192 are due to the same root cause.

leezu, 2022-10-06:
This issue reproduces on a fresh boot for which only the CF-953AX adapter was connected:

```
[89918.453283] mt7921u 2-1:1.3: Message 00020003 (seq 1) timeout
[89918.453343] wlxe0e1a934a6a9: failed to remove key (0, 2c:71:ff:8e:bd:7f) from hardware (-110)
[89918.741304] mt7921u 2-1:1.3: timed out waiting for pending tx
[89918.758984] ------------[ cut here ]------------
[89918.759003] WARNING: CPU: 0 PID: 7829 at kernel/kthread.c:659 kthread_park+0xb4/0xd0
[89918.759037] Modules linked in: ctr aes_arm64 aes_generic ccm xt_MASQUERADE iptable_nat xt_mark nft_chain_nat ip6table_nat nf_nat tun mt7921u mt7921_common mt76_connac_lib mt76_usb btusb mt76 btrtl mac80211 btintel btbcm bluetooth ecdh_generic vc4 ecc libarc4 libaes snd_soc_hdmi_codec drm_display_helper brcmfmac cec drm_cma_helper brcmutil bcm2835_codec(C) rpivid_hevc(C) drm_kms_helper bcm2835_isp(C) v3d cfg80211 v4l2_mem2mem bcm2835_v4l2(C) snd_soc_core gpu_sched bcm2835_mmal_vchiq(C) drm_shmem_helper videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev snd_bcm2835(C) snd_compress raspberrypi_hwmon snd_pcm_dmaengine rfkill snd_pcm vc_sm_cma(C) snd_timer mc snd syscopyarea sysfillrect sysimgblt uio_pdrv_genirq fb_sys_fops nvmem_rmem uio ip6t_REJECT nf_reject_ipv6 xt_hl ip6_tables ip6t_rt ipt_REJECT nf_reject_ipv4 xt_comment xt_multiport nft_limit xt_limit xt_addrtype xt_tcpudp xt_conntrack nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nft_compat
[89918.759451]  nf_tables nfnetlink drm fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
[89918.759502] CPU: 0 PID: 7829 Comm: kworker/u8:2 Tainted: G         C         6.0.0-rc7-v8+ #4
[89918.759518] Hardware name: Raspberry Pi 4 Model B Rev 1.4 (DT)
[89918.759528] Workqueue: mt76 mt7921_mac_reset_work [mt7921_common]
[89918.759579] pstate: 00000005 (nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[89918.759592] pc : kthread_park+0xb4/0xd0
[89918.759604] lr : mt76u_stop_tx+0x278/0x330 [mt76_usb]
[89918.759632] sp : ffffffc008f63c50
[89918.759639] x29: ffffffc008f63c50 x28: 0000000000000000 x27: ffffff8049818848
[89918.759662] x26: 0000000000000000 x25: ffffff8042e26c80 x24: ffffff8049812068
[89918.759684] x23: ffffff8049814820 x22: ffffff8049816020 x21: ffffff8049812048
[89918.759705] x20: ffffff8048bf5880 x19: ffffff8043cf5d00 x18: 0000000000000000
[89918.759725] x17: 0000000000000001 x16: ffffffd2e20b4e40 x15: 000efb46b4eb6234
[89918.759745] x14: 001163b91bdc1648 x13: 00000000000003dd x12: 00000000fa83b2da
[89918.759766] x11: 00000000000003dd x10: 0000000000001a90 x9 : ffffffd2d63fe8a8
[89918.759786] x8 : ffffff8040b8f7f0 x7 : 0000000000000001 x6 : ffffffd2e350b0c0
[89918.759806] x5 : ffffffd2e33a9000 x4 : ffffffd2e33a90b0 x3 : 0000000000002800
[89918.759825] x2 : 0000000000000000 x1 : 0000000000000000 x0 : 0000000000000004
[89918.759845] Call trace:
[89918.759851]  kthread_park+0xb4/0xd0
[89918.759863]  mt76u_stop_tx+0x278/0x330 [mt76_usb]
[89918.759887]  mt7921u_mac_reset+0x88/0x2d8 [mt7921u]
[89918.759906]  mt7921_mac_reset_work+0xac/0x1a0 [mt7921_common]
[89918.759938]  process_one_work+0x1dc/0x450
[89918.759953]  worker_thread+0x154/0x450
[89918.759965]  kthread+0x104/0x110
[89918.759975]  ret_from_fork+0x10/0x20
[89918.759990] ---[ end trace 0000000000000000 ]---
[89918.898590] mt7921u 2-1:1.3: HW/SW Version: 0x8a108a10, Build Time: 20220908210919a

[89918.910901] mt7921u 2-1:1.3: WM Firmware Version: ____010000, Build Time: 20220908211021
[89927.669357] mt7921u 2-1:1.3: Message 00020003 (seq 10) timeout
[89927.669421] wlxe0e1a934a6a9: failed to remove key (0, 8c:fd:f0:42:20:cf) from hardware (-110)
[89929.993368] mt7921u 2-1:1.3: timed out waiting for pending tx
[89930.133974] mt7921u 2-1:1.3: HW/SW Version: 0x8a108a10, Build Time: 20220908210919a

[89930.146118] mt7921u 2-1:1.3: WM Firmware Version: ____010000, Build Time: 20220908211021
[89935.349382] mt7921u 2-1:1.3: Message 00020001 (seq 12) timeout
```

Jibun-no-Kage, 2022-11-23:
A suggestion, maybe ping morrownr, here on Github, he has extensive experience with some of the mt7921 based devices.

P33M, 2022-12-19:
Your listing has bMaxBurst=0 for all of the endpoints in any of the configurations / alternate settings. So the burst OUT bug, and the corresponding fix, doesn't apply.

If the crash happens only under heavy load, then it's possible the wifi adapter momentarily exceeds the 1.2A downstream port current limit. Does it still happen if plugged into a self-powered USB3.0 hub?

gifter77, 2023-05-07:
I have the same crash happening even if plugged into a self-powered USB3.0 hub.

cronop, 2023-06-25:
Same thing happens on an Orange Pi 5 Plus with Openwrt running in a Proxmox VM:

[  332.940838] mt7921u 4-1:1.0: Message 00020003 (seq 7) timeout
[  332.943895] phy0-ap0: failed to set key (1, ff:ff:ff:ff:ff:ff) to hardware (-110)
[  335.980509] mt7921u 4-1:1.0: Message 00020003 (seq 8) timeout
[  335.991275] phy0-ap0: failed to set key (4, ff:ff:ff:ff:ff:ff) to hardware (-110)
[  339.235851] mt7921u 4-1:1.0: vendor request req:63 off:d02c failed:-110
[  342.470445] mt7921u 4-1:1.0: vendor request req:63 off:d054 failed:-110
[  345.672157] mt7921u 4-1:1.0: vendor request req:63 off:d058 failed:-110
[  348.890817] mt7921u 4-1:1.0: vendor request req:63 off:53b8 failed:-110
[  352.121498] mt7921u 4-1:1.0: vendor request req:63 off:53c4 failed:-110
[  355.362003] mt7921u 4-1:1.0: vendor request req:66 off:53c4 failed:-110
[  358.610738] mt7921u 4-1:1.0: vendor request req:63 off:d698 failed:-110
[  361.825873] mt7921u 4-1:1.0: vendor request req:63 off:d520 failed:-110
[  365.031865] mt7921u 4-1:1.0: vendor request req:63 off:d518 failed:-110
[  368.274826] mt7921u 4-1:1.0: vendor request req:63 off:d688 failed:-110
[  371.482260] mt7921u 4-1:1.0: vendor request req:63 off:d690 failed:-110
[  374.698523] mt7921u 4-1:1.0: vendor request req:63 off:d558 failed:-110
[  377.916391] mt7921u 4-1:1.0: vendor request req:63 off:d564 failed:-110
[  381.142789] mt7921u 4-1:1.0: vendor request req:63 off:d568 failed:-110
[  384.365935] mt7921u 4-1:1.0: vendor request req:63 off:d7a8 failed:-110
[  387.582269] mt7921u 4-1:1.0: vendor request req:63 off:a150 failed:-110
[  390.820761] mt7921u 4-1:1.0: vendor request req:63 off:a158 failed:-110
[  394.036455] mt7921u 4-1:1.0: vendor request req:63 off:d780 failed:-110
[  397.265631] mt7921u 4-1:1.0: vendor request req:63 off:d770 failed:-110
[  400.508192] mt7921u 4-1:1.0: vendor request req:63 off:d774 failed:-110
[  403.721153] mt7921u 4-1:1.0: vendor request req:63 off:d55c failed:-110
[  406.961058] mt7921u 4-1:1.0: vendor request req:63 off:10e0 failed:-110
[  410.185741] mt7921u 4-1:1.0: vendor request req:63 off:10e4 failed:-110
[  413.441741] mt7921u 4-1:1.0: vendor request req:63 off:10e8 failed:-110
[  416.689423] mt7921u 4-1:1.0: vendor request req:63 off:10ec failed:-110
[  419.912375] mt7921u 4-1:1.0: vendor request req:63 off:10f0 failed:-110
[  423.172345] mt7921u 4-1:1.0: vendor request req:63 off:10f4 failed:-110
[  426.412783] mt7921u 4-1:1.0: vendor request req:63 off:10f8 failed:-110
[  429.631078] mt7921u 4-1:1.0: vendor request req:63 off:10fc failed:-110
[  432.862377] mt7921u 4-1:1.0: vendor request req:63 off:d7dc failed:-110
[  436.082556] mt7921u 4-1:1.0: vendor request req:63 off:d7ec failed:-110
[  439.292609] mt7921u 4-1:1.0: vendor request req:63 off:d7e0 failed:-110
[  442.543617] mt7921u 4-1:1.0: vendor request req:63 off:d7f0 failed:-110
[  445.773236] mt7921u 4-1:1.0: vendor request req:63 off:d7e4 failed:-110
[  448.990723] mt7921u 4-1:1.0: vendor request req:63 off:d7f4 failed:-110
[  452.214152] mt7921u 4-1:1.0: vendor request req:63 off:d7e8 failed:-110
[  455.453004] mt7921u 4-1:1.0: vendor request req:63 off:d7f8 failed:-110
[  458.531282] mt7921u 4-1:1.0: Message 00020002 (seq 9) timeout
[  461.571373] mt7921u 4-1:1.0: Message 00020002 (seq 10) timeout
[  461.784163] mt7921u 4-1:1.0: HW/SW Version: 0x8a108a10, Build Time: 20230331110902a
[  461.784163] 
[  461.811386] mt7921u 4-1:1.0: WM Firmware Version: ____010000, Build Time: 20230331110939
[  464.565793] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-ap0: link becomes ready
[  464.569712] br-lan: port 2(phy0-ap0) entered blocking state
[  464.576405] br-lan: port 2(phy0-ap0) entered forwarding state

root@OpenWrt:/# uname -a
Linux OpenWrt 6.1.35 #0 SMP Fri Jun 23 21:07:17 2023 aarch64 GNU/Linux

efemerny, 2026-09-28:
I can still reproduce this issue on a Raspberry Pi 4 Model B Rev 1.5
using an MT7921U adapter (`0e8d:7961`) as a 5 GHz AP over USB 3.0.

Current environment:

- Kernel: `6.18.50+rpt-rpi-v8`
- Firmware: `____010000-20260224110949`
- USB link: SuperSpeed, 5000M
- `mt76_usb disable_usb_sg=Y`
- USB runtime power management disabled
- AP: channel 36, VHT80

The failure produces essentially the same trace as the original report:

```text
mt7921u 2-2:1.0: Message 00020003 (seq 13) timeout
mt7921u 2-2:1.0: timed out waiting for pending tx
WARNING at kernel/kthread.c:709 kthread_park
mt76u_stop_tx
mt7921u_mac_reset
mt7921_mac_reset_work
```

I also tested a backport of the current MT7921U USB recovery series from
mt76, including the `mt792xu_reset_on_bus_error()` recovery path. The
failure still occurred.

The firmware reloaded and a client re-associated about 14 seconds later,
but the AP was unavailable during recovery. Previous failures sometimes
required a manual USB unbind/bind.

Full sanitized log is attached. I also reported the driver-side details
here: https://github.com/openwrt/mt76/issues/1141