raspberrypi/linux #5195: RPI4B 64bit vc4-kms-v3d blank screen during post

JerryK13, 2022-10-03:


popcornmix, 2022-10-03:
> On a fresh Raspberry Pi OS Lite 64bit (September 22nd 2022) and also rpi-updated to ed6ef1c57077cc3969ae62a4bd9dc1abe5e2a7b7 (kernel 5.15.70-v8+) using the default vc4-kms-v3d the display turns blank during post on two different displays while vc4-fkms-v3d works as intended.

There are a few stages during boot where hdmi is output. Do you see output in some of them?
1) bootloader outputs a diagnostic screen with white text and a raspberry in corner (this can be quite quick so monitor may not switch mode in time - removing sdcard will make it stay up permanently).
2) start.elf outputs a rainbow square
3) kernel stars boot using a simple framebuffer (created with firmware settings)
4) kms driver is probed and takes over second half of boot (you'll see a screen clear here)
5) X starts and may choose a new hdmi mode

Booting to desktop does hide some of the console output, so booting to console may be a simpler case to investigate.

popcornmix, 2022-10-03:
I've tested your edid on my Pi and it seems fine. I boot to console in 1920x1080@60 mode. 
```mode: "1920x1080": 60 148500 1920 2008 2052 2200 1080 1084 1089 1125 0x48 0x5```

You could try capturing the edid:
sudo cp sys/devices/platform/gpu/drm/card1/card1-HDMI-A-1/edid /lib/firmware

and then use the captured edid and a specified mode. Add this to end of cmdline.txt (same line)
```drm.edid_firmware=edid video=HDMI-A-1:1920x1080@60 ```

Can you ssh in and run commands when display is blank?
I'd be interested in 
```
base64 /sys/devices/platform/gpu/drm/card1/card1-HDMI-A-1/edid # it may be invalid before the switch

JerryK13, 2022-10-03:
Thank you for the reply.
None of the displays are fast enough to display the diagnostic screen. The rainbow screen is displayed and then the framebuffer. After switching to KMS, the displays go blank. My Pi is not configured to enter X/Wayland. I have however tried turning that on, to see if it makes any difference and it does not. Neither starting X nor Sway produce any visible output. I can however at anytime powercycle the monitor (or switch source on the TV) to get the picture working. 
(just for the reference, If I use the obsolete FKMS, it does produce a clear but the picture is returned in a second.)

The first thing I tired, was capturing the monitor edid and loading it via cmdline.txt (with and without using the video attribute) but it hasn't worked (haven't tried on the TV yet).

I have now stored the edid (via SSH) while the monitor displayed blank (haven't tried the TV yet), and the edid is exactly the same as when power cycled:
```
AP///////wBMLdUEAAAAACATAQOAEAl4KmBBplZKnCUSUFQjCACpQIGAgUCBAJUAswABAQEBAjqA
GHE4LUBYLEUAoFoAAAAeAR2AGHEcFiBYLCUAoFoAAACeAAAA/QAyPBtREQAKICAgICAgAAAA/ABT
eW5jTWFzdGVyCiAgAboCAxzxSJAfBRQEEwMSIwkHB4MBAABmAwwAEACAAR0AclHQHiBuKFUAoFoA
AAAeAR0AvFLQHiC4KFVAoFoAAAAejArQkCBAMSAMQFUAoFoAAAAYAR2A0HIcFiAQLCWAoFoAAACe
jArQiiDgLRAQPpYAoFoAAAAYAAAAAAAAAAAADg==
```
Pressing the power button on the monitor also doesn't do it. I have to physically remove it's power.

JerryK13, 2022-10-03:
It is exactly the same on the TV. The /sys/devices/platform/gpu/drm/card1/card1-HDMI-A-1/edid is the same as after switching source (you just have to remember to remove the loaded edid from the cmdline first!). On this device just switching the source makes it work.
```
AP///////wAebQEAAQEBAQEWAQOAoFp4Cu6Ro1RMmSYPUFShCAAxQEVAYUBxQIGAAQEBAQEBAjqA
GHE4LUBYLEUAoFoAAAAeZiFQsFEAGzBAcDYAoFoAAAAeAAAA/QA6Ph5TEAAKICAgICAgAAAA/ABM
RyBUVgogICAgICAgAUMCAzPxThCfBBMFFAMCEiAhIhUBJhUHUAlXB3gDDAAgAIAeIMAOAUAKDwgQ
GBCYEFgQOBABHYAYcRwWIFgsJQAgwjEAAJ4BHQByUdAeIG4oVQAgwjEAAB4COoAYcTgtQFgsRQCg
WgAAAB4BHQC8UtAeILgoVUDEjiEAAB4AAAAAEw==
```
I also tried another HDMI cable to no avail. 

In the process I also discovered that using the stable EEPROM bootloader (Tue Aug 2 15:55:05 UTC 2022 (1659455705)) causes my Pi to hang at rainbow if I issue a reboot command with 7 power led flashes, while shutdown or power cycling boots normally. The critical branch (Tue Apr 26 10:24:28 UTC 2022 (1650968668)) does not do this and reboots normally. This probably warrants another bug report in the firmware section.

popcornmix, 2022-10-03:
> This probably warrants another bug report in the firmware section.

Yes. Sounds like an unrelated issue.

popcornmix, 2022-10-03:
> Switching source or unplugging/plugging the HDMI cable on this display does nothing. However if I unplug the display power cable and plug it back, the display comes to life.

I'd like to be sure whether, after the power cycle of the display, the display becomes happy with the existing signal we output, or whether some hotplug toggling that occurs causes us to switch to a hdmi signal the display likes better.

Can you use this in cmdline.txt:
```drm.edid_firmware=edid video=HDMI-A-1:1920x1080@60 vc4.force_hotplug=1```

I believe now the pi will not be able to detect any cable unplug or power cycle of display.
Does power cycling the display bring the picture back?

It may also be interesting to look at:
```sudo cat /sys/kernel/debug/dri/1/state```
(if there is no file use replace the 1 with a 0).

Perhaps copy that file somewhere before and after the power cycle and diff them just to see if anything changes.

JerryK13, 2022-10-04:
I've setup the Pi to startx on tty7 (lightdm) and did some testing. All the testing is on the monitor.

With `drm.edid_firmware=edid video=HDMI-A-1:1920x1080@60` in cmdlinte.txt and HDMI connected before booting up, it's blank across all ttys.
If I disconnect the HDMI (at the Pi end), boot up the Pi, connect the HDMI and then switch to tty1 I get a picture of the console. Switching back to tty7 however produces a blank screen again. I can then switch between the working tty1 and blank tty7 as much as I want to.
But if I disconnect the HDMI (at the Pi end), boot up the Pi, switch to tty1 and then connect the HDMI, I get blank across all the ttys. In this case, the only way to see some output is to power cycle the monitor. The same happens if I switch to tty1 and then back to tty7 and then connect the HDMI, blank across all the ttys.

With `drm.edid_firmware=edid video=HDMI-A-1:1920x1080@60 vc4.force_hotplug=1` in the cmdline.txt and HDMI connected before booting up, it's blank across all ttys. But if I boot with disconnected HDMI (at the Pi end) and plug it in after KMS took over, I get output on all ttys without any tty switching.

Here is the `/sys/kernel/debug/dri/1/state` with `vc4.force_hotplug=1` before and after
```
# diff statebefore stateafter
30,31c30,31
< 	fb=340
< 		allocated by = Xorg
---
> 	fb=338
> 		allocated by = [fbcon]
33c33
< 		format=XR24 little-endian (0x34325258)
---
> 		format=RG16 little-endian (0x36314752)
38c38
< 			pitch[0]=7680
---
> 			pitch[0]=3840
43,44c43,44
< 				start=001003f5
< 				size=8294400
---
> 				start=00100000
> 				size=4149248
46,47c46,47
< 				paddr=0x00000000cfb00000
< 				vaddr=00000000a69ab5ba
---
> 				paddr=0x00000000cf700000
> 				vaddr=00000000be273bc4
244,264c244,247
< 	crtc=crtc-3
< 	fb=342
< 		allocated by = Xorg
< 		refcount=1
< 		format=AR24 little-endian (0x34325241)
< 		modifier=0x0
< 		size=64x64
< 		layers:
< 			size[0]=64x64
< 			pitch[0]=256
< 			offset[0]=0
< 			obj[0]:
< 				name=0
< 				refcount=3
< 				start=00100bea
< 				size=16384
< 				imported=no
< 				paddr=0x00000000cf050000
< 				vaddr=000000001f264dee
< 	crtc-pos=64x64+949+528
< 	src-pos=64.000000x64.000000+0.000000+0.000000
---
> 	crtc=(null)
> 	fb=0
> 	crtc-pos=0x0+0+0
> 	src-pos=0.000000x0.000000+0.000000+0.000000
335c318
< 	plane_mask=2000008
---
> 	plane_mask=8
338c321
< 	mode: "": 60 148500 1920 2008 2052 2200 1080 1084 1089 1125 0x0 0x5
[... 714 lines cut ...]
	encoder_mask=0
	mode: "": 0 0 0 0 0 0 0 0 0 0 0x0 0x0
crtc[116]: crtc-5
	enable=0
	active=0
	self_refresh_active=0
	planes_changed=0
	mode_changed=0
	active_changed=0
	connectors_changed=0
	color_mgmt_changed=0
	plane_mask=0
	connector_mask=0
	encoder_mask=0
	mode: "": 0 0 0 0 0 0 0 0 0 0 0x0 0x0
connector[32]: HDMI-A-1
	crtc=crtc-3
	self_refresh_aware=0
connector[41]: HDMI-A-2
	crtc=(null)
	self_refresh_aware=0
connector[50]: Writeback-1
	crtc=(null)
	self_refresh_aware=0
```

popcornmix, 2022-10-04:
I don't think you've run the test I described in my previous post.
My understanding was that if you to boot to console (with hdmi attached), you see some console text but screen goes blank when kernel starts the kms driver. You can get output back by power cycling the monitor.

I want you to repeat this with this in config.txt
```drm.edid_firmware=edid video=HDMI-A-1:1920x1080@60 vc4.force_hotplug=1```

Do you still get blank screen at end of boot, and do you get output back after power cycling the monitor?

popcornmix, 2022-10-04:
```
< 	fb=340
< 		allocated by = Xorg
---
> 	fb=338
> 		allocated by = [fbcon]
```
You seem to be running with X active in the first case and X inactive (possibly switched to a different VT) in the second case.
That makes any comparison impossible. The useful comparison would be for the no display case and the display working case with everything else the same. Ideally run in the simple boot to console case for each of these cases.

JerryK13, 2022-10-04:
> I don't think you've run the test I described in my previous post. My understanding was that if you to boot to console (with hdmi attached), you see some console text but screen goes blank when [...]

Yes. Everything stays the same as before. I see the rainbow screen and the framebuffer but after kms takes over it goes blank. I can bring it back with power cycling the monitor. The only difference is, with vc4.force_hotplug=1, I can boot without hdmi cable attached and if I attach it later when the KMS is in charge, I get output without the need to power cycle the monitor. This does not work without vc4.force_hotplug=1.

> You seem to be running with X active in the first case and X inactive (possibly switched to a different VT) in the second case.
> That makes any comparison impossible. The useful comparison would be [...]

Sorry about that one, didn't notice I switched ttys. Retested again with X disabled, the dri state before and after remain the same.

JerryK13, 2022-10-04:
And it it exactly the same on the TV. I can even use the monitors edid in the cmdline.txt and produce the same results (get output by plugging the HDMI after KMS takes over or by switching source on the TV).

Just to rule out if this particular Pi is a bit wonky, I tried this sdcard with this image in two other RPI4B on both the TV and the monitor and the results are the same.

popcornmix, 2022-10-04:
That is strange. I believe the signal we are outputting is identical before and after the display power cycle, so the fix is the display resetting its state which sounds like a display problem.

But as this is a very rare issue (I'm not sure I've seen this reported), it seems inconceivable you have two unrelated displays with this issue.

Just a sanity check. What does `vcgencmd get_throttled` report after boot?

popcornmix, 2022-10-04:
I wonder if the issue is the switchover that occurs from firmware to kms driver (effectively the TV sees two hdmi mode changes a few seconds apart). It's a longshot but add to cmdline.txt:
`drm.debug=0x14 boot_delay=1000` 

That will generate more drm debugging and also delay debug messages to 1 per second. 
It will take a couple of minutes to boot, but if it works it may indicate issue is timing related.

JerryK13, 2022-10-04:
The `drm.debug=0x14` was already included in the default of the foundation supplied image (and my kern.log and syslog were already  couple of GB in size. I wander if the periodic freezes I was experiencing with this setup were due to this excessive logging) so I just added the `boot_delay=1000`.

Now the boot up is taking about 180s, and once systemd starts spawning services it starts flying (no more ~1/s, more like 10/s, not sure if any slower than without boot delay). I noticed, that the mode change occurs somewhere near the end of systemd spawning services. Before that the output is always visible and not changing mode. Now I am not sure where KMS mode change occurs (4th point in your first reply), perhaps something else is changing mode later on?

Anyway, did a couple of test with `drm.debug=0x14 boot_delay=1000`. The first ones I removed the edid and video from cmdline.txt. Tried with lightdm enabled and disabled and nothing changed - all blank until I power cycle the monitor.
The second tests I did, were in addition to debug and delay with added edid and video. I did roughly 10 start ups, and on about half of them, during the systemd spawning, the output went blank and returned, so I could see the console. If I enabled lightdm, then the output was always blanked.

You can compare [kernwoxwcmd.log](https://github.com/raspberrypi/linux/files/9708157/kernwoxwcmd.log) (kern.log with debug, delay, edid and video and without starting X) with [kernmonitor.log](https://github.com/raspberrypi/firmware/files/9696624/kernmonitor.log) (kern.log without those bootloader/kernel parameters also without starting X) from opening post.

popcornmix, 2022-10-04:
> The drm.debug=0x14 was already included in the default of the foundation supplied image

I don't believe that to be true.

JerryK13, 2022-10-04:
Yes, you are right! I've must have missed copy&pasting that while messing with loading the edid from file before opening this report. Sorry for the miss information.

macmpi, 2022-10-12:
> It will take a couple of minutes to boot, but if it works it may indicate issue is timing related.

aside this debug workaround, is a permanent firmware/driver fix possible?

popcornmix, 2022-10-12:
macmpi, this issue is a very specific failure mode which sounds like a problem with the display (power cycling the display fixes it, even though the signal received is the same before and after power cycle).

I don't believe there is a reliable workaround yet (sounds like significant delays only sometimes avoids the issue).

Are you saying you have exactly the same issue?

HiassofT, 2022-10-12:
The issue with the LG TV in the initial report reads a lot like this firmware issue  https://github.com/raspberrypi/firmware/issues/1647 - which also mentions an LG TV of the same vintage.

As that issue was narrowed down to a firmware-only change I suspect something in the FW->KMS handover (timing maybe?) might have changed that results in the TV not wanting to show a picture.

A shot in the dark: the vc4 KMS driver doesn't seem to implement AVMUTE support while changing modes. ISTR I read some reports on LE slack or forums (could have been on Rockchip or Allwinner devices) that some TVs didn't like (some?) mode changes unless AVMUTE was set before mode change and cleared after.

Might be worth adding AVMUTE support to the driver, maybe that helps (getting the timing right might be a bit tricky as IIRC AVMUTE can only be changed at the beginning of each frame and any clock changes have to be delayed until the GCP has been transmitted)

macmpi, 2022-10-13:
> Are you saying you have exactly the same issue?

It looked similar on a Toshiba TV with PiZeroW, but I do not have that TV with me right now to confirm if it is exactly same issue.
Will check at next opportunity but may take time...

bruceb85, 2022-10-14:
I had same issue with LG 34WK95U-W

commenting out this lines solves the issue, but why?

# Enable DRM VC4 V3D driver
#dtoverlay=vc4-kms-v3d

macmpi, 2022-10-14:
> A shot in the dark: the vc4 KMS driver doesn't seem to implement AVMUTE support while changing modes.[...]
> Might be worth adding AVMUTE support to the driver, [...]

@HiassofT : very enlightening indeed thanks.
I guess AVMUTE support is probably key for supporting AVR use-cases as per https://github.com/raspberrypi/firmware/issues/1647#issuecomment-1059281180 (Yamaha AVR + Sony TV in that specific case)?

popcornmix, 2022-10-14:
@HiassofT 
HDMI 1.4b spec:
> The General Control Packet contains fields for indicating AVMUTE information and color-depth
information. Each transmitted GCP may contain valid indications for AVMUTE and/or color-depth
or may contain no information (all fields zero).
General Control packets indicating Set_AVMUTE or Clear_AVMUTE may only be transmitted
between the active edge of VSYNC and 384 pixels following this edge. A Source may not send a
General Control Packet with the Clear_AVMUTE and Set_AVMUTE flags set simultaneously.
Source transmission of the General Control Packet is optional. Sinks may optionally interpret
General Control Packet contents. Sinks shall be capable of receiving any General Control Packet.
The General Control packet’s Set_AVMUTE and Clear_AVMUTE flags may be used by a Source
to reduce the negative impact on the Sink of TMDS clock changes or interruptions. Use of the
AVMUTE function may prevent spurious pops or noises in the audio during these clock changes.
When AVMUTE is set, the Sink may assume that no valid audio or video data is being received.
The Sink may optionally apply a mute function to the audio data and/or a blank function to the
video.

"Source transmission of the General Control Packet is optional" suggests this isn't something required and a display shouldn't behave badly if it is missing.

picov, 2022-11-01:
Same problem with 2022-09-22-raspios-bullseye-arm64.img
Same problem updating LibreElec from 9.x to 10.x series.
The problem occours with both TV (Panasonic VIERA TX P46GT30) and Monitor (Asus QHD 27") 
I've a RPI4.
If I modify the config.txt by replacing dtoverlay=vc4-kms-v3d with dtoverlay=vc4-fkms-v3d the problem disappears.

popcornmix, 2023-01-24:
There is a potential update for this issue (screen going blank at point of transition of kms driver on a pi4) in latest rpi-update kernel. It would be useful if anyone who experiences this issue can test and report back.

JerryK13, 2023-01-24:
I can confirm that with the rpi-update commit 20169be87eee199ab8c958a02ef923597f15f944 the firmware to kms handover works on both displays from the opening comment! Good work!

popcornmix, 2023-01-24:
@JerryK13 Great to hear. This issue has taken a while to understand.
Hopefully the fix resolves a number of blank screen issues.

JerryK13, 2023-02-06:
#5317 fixed this issue.

closed 2023-02-06 by JerryK13

markusand, 2023-06-12:
> Same problem with 2022-09-22-raspios-bullseye-arm64.img Same problem updating LibreElec from 9.x to 10.x series. The problem occours with both TV (Panasonic VIERA TX P46GT30) and Monitor (Asus QHD [...]

This solved my problem