raspberrypi/linux #5200: Artefact in IMX462 camera frames after rpi-update

sandyol55, 2022-10-06:
### Describe the bug

When in the 1280 : 720 : 12bit mode the sensor outputs a noisy coloured 'overlay' over the rightmost 32 pixels. Colour seems to vary depending on the content of main image.
Does not happen in any of the three other sensor modes. The artefact appears in preview, captured stills, or video,  and in both libcamera-apps and Picamera2 applications, (so presumed to be a low level change perhaps associated with recent IMX290 driver modifications???) 

### Steps to reproduce the behaviour

IMX462 board directly driven by 2G Pi4b via short (150mm) cable.
Using imx290 dtoverlay.
A simple libcamera-hello is enough to demonstrate  the issue, as it opens in the 1280:720:12bit mode, the only mode that seems to exhibit the artefact. 
Artefact appeared after recent rpi-updates to both .70 and .72 kernels.
Reverting to 5.15.61-v7l+ reverts to normal behaviour.

### Device (s)

Raspberry Pi 4 Mod. B

### System

OS 32bit Bullseye 

$ vcgencmd version
Oct  5 2022 15:51:11 
Copyright (c) 2012 Broadcom
version 6c57d97fa8066929a2fee6e4fac6d64ad62506c9 (clean) (release) (start)

$ uname -a
Linux raspberrypi 5.15.72-v7l+ #1591 SMP Wed Oct 5 12:05:33 BST 2022 armv7l GNU/Linux

6by9, 2022-10-10:
I've reproduced this with an Arducam IMX462 (UC-755). ```dtoverlay=imx290``` is all that I've applied for configuration as the oscillator is 37.125MHz on that board.

It's an odd error - it looks like the CSI side of the sensor is running too slow to keep up with the sensor array. The numbers look to match the datasheet though, so it needs some more investigation.

6by9, 2022-10-10:
Found.
In 720p mode the CSI link is running slower, and therefore the HMAX value (width + HBLANK) needs to be increased to 3300 rather than being left at 2200 (the same ratio as the 891 vs 594Mbit/s/lane speed).
The latest patches had set blanking based on a constant HMAX of 2200.

Fixing this is relatively simple, although please note that as the HBLANK control doesn't get set to a particular value on mode change (only the ranges change, and value is adjusted to fit those _if required_), you are likely to need the patch proposed in https://lists.libcamera.org/pipermail/libcamera-devel/2022-October/034679.html to make libcamera understand HBLANK.

sandyol55, 2022-10-11:
Speedy response for an 'as-time-allows' investigation!! ;-)
Many Thanks.
Glad to see it was a real issue and not a fault in my hardware.
When libcamera reaches the stage of understanding variable horizontal_blanking is it likely to be in the form of an additional control or will it be incorporated in to the AEC algorithm and be pretty much invisible to us users, other than a wider effective range of possible exposure times?

6by9, 2022-10-11:
> Speedy response for an 'as-time-allows' investigation!! ;-)

I'm currently trying to sort out my changes to push upstream, so it was a useful one to track down.

> When libcamera reaches the stage of understanding variable horizontal_blanking is it likely to be in the form of an additional control or will it be incorporated in to the AEC algorithm and be pretty much invisible to us users, other than a wider effective range of possible exposure times?

It'll be a wider range of exposure time / frame duration.
It'll adopt the minimum HBLANK as standard to minimize any rolling shutter effects, but increase it should we hit max VBLANK and still not be able to achieve the requested frame time. Exposing it explicitly just adds complication for the user.

closed 2022-10-11 by sandyol55