raspberrypi/linux #5203: 64-bit OS, RPI4B, DPMS stops after video played

notGMman, 2022-10-08:
### Describe the bug

RPi4B, running 64-bit OS, upgraded using dist-upgrade on Mon 3rd October 2022.
Linux XXX 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux

"xset dpms force off" works after booting. Various apps run to test and DPMS continues to work afterwards. 
However, after VLC is run and video played (which stops DPMS whilst running, as expected) AND also after Chromium is run and a video played on Youtube, DPMS not longer works - screen briefly (sub 1s) blanks then returns.

I use bluetooth headphones but see this behaviour even when they've not been switched on. 

### Steps to reproduce the behaviour

As above.
reboot PI.
"xset DPMS force off" works.
 Open CHromium or VLC, play a video. Stop the app. 
"xset DMPS force off" doesn't work. Screen returns < 1s after command actioned.

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspberry Pi reference 2022-01-28
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, fbe448ccdc995d295d24c7596e5f0ef62cc2488f, stage4

Aug 26 2022 14:04:10 
Copyright (c) 2012 Broadcom
version 102f1e848393c2112206fadffaaf86db04e98326 (clean) (release) (start_x)

Linux ron10 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux

### Logs

Not sure where to look. Google not offering any clues.

### Additional context

Was running old Bullseye 64-bit release from ~late Feb before this upgrade. Worked fine.

popcornmix, 2022-10-10:
I've just updated a 64-bit RpiOS bullseye and I can't reproduce this issue. What I've tried.
Boot to desktop. I've ssh-ed in and run
```export DISPLAY=:0```
```xset dpms force off``` makes display go blank and ```xset dpms force on``` makes it come back.
I launch chromium.
```xset dpms force off``` makes display go blank and ```xset dpms force on``` makes it come back.
I browser to youtube and play a video (embedded on web page)
```xset dpms force off``` makes display go blank and ```xset dpms force on``` makes it come back.
I switch to full screen
```xset dpms force off``` makes display go blank and ```xset dpms force on``` makes it come back.
I quit chromium
```xset dpms force off``` makes display go blank and ```xset dpms force on``` makes it come back.
I run VLC
```xset dpms force off``` makes display go blank and ```xset dpms force on``` makes it come back.
I play a video (hevc, embedded in window)
```xset dpms force off``` makes display go blank and ```xset dpms force on``` makes it come back.
I switch to fullscreen
```xset dpms force off``` and display remains on(*) ```xset dpms force on``` and display remains on
I switch back to desktop
```xset dpms force off``` makes display go blank and ```xset dpms force on``` makes it come back.
I quit vlc
```xset dpms force off``` makes display go blank and ```xset dpms force on``` makes it come back.

(*) looking at layers on screen, with kmsprint, the force off did remove the desktop, but the fullscreen overlay of video remained in place.

So I'm not reproducing the issue. The screen is still blank minutes later if I just run ```xset dpms force off``` after running chromium and vlc.

Can you explain if I've missed any step you are running?
If you have a spare sdcard, are you able to reproduce with a clean install of RPiOS64 bullseye?

notGMman, 2022-10-10:
Hi Popcornmix,
Thanks for jumping on this so quickly. I've got a lot of Pi's so I'll see what I can do.
A couple of observations - 
1. VLC,preferences, advanced, video - "disable screensaver" is ticked for me.
2. For some reason (and maybe I'm making it up) I thought the same disable screen saving was a feature of Chromium playing videos. In other words, yours isn't working properly :-)
However, I see that mine isn't consistent. I play an MKV in VLC - which I grabbed with Youtube-DL - and it disables the screensaver every time. I play a video grabbed by my trailcam - mp4. It doesn't. So this is inconsistent/variable.
I'll check on another device. You can probably tell I hate upgrading for exactly this reason.
Thx,