raspberrypi/linux #5207: Camera device is not detecting with 5.15.65 rpi kernel.

savidha777, 2022-10-14:
### Describe the bug

Upgraded rpi kernel version from rpi-5.4.y to rpi-5.15.y.

Camera device (/dev/video0) device is not detecting with rpi-5.15.y version.

Linux version : 5.15.65
revision : 08a29ba5fb33886d3d1095d66110bdd38f6820a2

### Steps to reproduce the behaviour

Install 5.15.65 kernel on RPI 3B+ and notice video0 device detection.

### Device (s)

Raspberry Pi 3 Mod. B+

### System

root@raspberrypi3-rdk-camera:/usr/local/rms/bin# vcgencmd version
May 27 2021 14:05:31 
Copyright (c) 2012 Broadcom
version 7d9a298cda813f747b51fe17e1e417e7bf5ca94d (clean) (release) (start_x)

root@raspberrypi3-rdk-camera:/usr/local/rms/bin# uname -a
Linux raspberrypi3-rdk-camera 5.15.65-v7 #1 SMP Fri Sep 23 10:49:53 UTC 2022 armv7l GNU/Linux

### Logs

root@raspberrypi3-rdk-camera:/usr/local/rms/bin# uname -r
5.15.65-v7

root@raspberrypi3-rdk-camera:/usr/local/rms/bin# ls /dev/video0
ls: cannot access /dev/video0: No such file or directory

root@raspberrypi3-rdk-camera:/usr/local/rms/bin# modprobe bcm2835-v4l2
modprobe: FATAL: Module bcm2835-v4l2 not found in directory /lib/modules/5.15.65-v7

root@raspberrypi3-rdk-camera:/usr/local/rms/bin# lsmod
Module                  Size  Used by

pelwell, 2022-10-14:
1. Which camera is this?

2. What non-standard settings (i.e. any changes you've made) are in config.txt?

savidha777, 2022-10-14:
Camera : RPi V2 model (imx219)

Not did any changes in config.txt file. whatever we used for 5.4 kernel same file only we are using for 5.15.

For the reference attaching config.txt file 
[config.txt](https://github.com/raspberrypi/linux/files/9785437/config.txt)

pelwell, 2022-10-17:
Your installation looks broken - the standard configurations enable and build bcm2835-v4l2, and our releases include it:
https://github.com/raspberrypi/rpi-firmware/tree/stable/modules/5.15.61-v7%2B/kernel/drivers/staging/vc04_services/bcm2835-camera
This is not a kernel bug.

closed 2022-10-17 by pelwell

savidha777, 2022-10-20:
Hi @pelwell 

Installation are not getting break, all the packages are there in part of image, but modules only not getting load.

root@raspberrypi3-rdk-camera:~# ls /lib/modules/5.15.74-v7/
kernel/                  modules.builtin          modules.dep              modules.order            modules.symbols.bin
modules.alias            modules.builtin.bin      modules.dep.bin          modules.softdep          
modules.alias.bin        modules.builtin.modinfo  modules.devname          modules.symbols 

root@raspberrypi3-rdk-camera:~# ls /lib/modules/5.15.74-v7/kernel/drivers/staging/vc04_services/bcm2835-camera/bcm2835-v4l2.ko.xz 
/lib/modules/5.15.74-v7/kernel/drivers/staging/vc04_services/bcm2835-camera/bcm2835-v4l2.ko.xz

root@raspberrypi3-rdk-camera:~# modprobe bcm2835-v4l2
modprobe: FATAL: Module bcm2835-v4l2 not found in directory /lib/modules/5.15.74-v7
root@raspberrypi3-rdk-camera:~#

pelwell, 2022-10-20:
The module loads for me - you are the only person reporting a problem. List the steps you use when building your kernel, including any config changes after "make ARCH=arm bcm2709_defconfig".

pelwell, 2022-10-20:
This is a big clue that your installation is broken:
```
root@raspberrypi3-rdk-camera:/usr/local/rms/bin# lsmod
Module Size Used by
```
No modules are loaded - on a Pi 4 I have 66.

savidha777, 2022-10-20:
We are using linux-raspberrypi yocto recipe to compile kernel module.

please find below recipe
**5.10.52 recipe:**
https://code.rdkcentral.com/r/plugins/gitiles/rdk/components/generic/rdk-oe/meta-cmf-raspberrypi/+/refs/heads/rdk-next/recipes-kernel/linux/linux-raspberrypi_5.10.52.bb

Modules are properly loading with 5.10.52
**root@raspberrypi3-rdk-camera:~# uname -r
5.10.52-v7**
root@raspberrypi3-rdk-camera:~# lsmod
Module                  Size  Used by
snd_seq_dummy          16384  0
snd_hrtimer            16384  1
snd_seq                69632  7 snd_seq_dummy
snd_seq_device         16384  1 snd_seq
hci_uart               40960  1
btbcm                  16384  1 hci_uart
bluetooth             413696  3 hci_uart,btbcm
ecdh_generic           16384  1 bluetooth
ecc                    40960  1 ecdh_generic
brcmfmac              331776  0
brcmutil               20480  1 brcmfmac
sha256_generic         16384  0
bcm2835_codec          40960  0
bcm2835_isp            32768  0
raspberrypi_hwmon      16384  0
bcm2835_v4l2           49152  1
snd_bcm2835            24576  1
v4l2_mem2mem           36864  1 bcm2835_codec
videobuf2_dma_contig    20480  2 bcm2835_isp,bcm2835_codec
bcm2835_mmal_vchiq     28672  3 bcm2835_isp,bcm2835_codec,bcm2835_v4l2
cfg80211              782336  1 brcmfmac
videobuf2_vmalloc      16384  1 bcm2835_v4l2
videobuf2_memops       16384  2 videobuf2_dma_contig,videobuf2_vmalloc
videobuf2_v4l2         32768  4 bcm2835_isp,bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem
videobuf2_common       61440  5 bcm2835_isp,bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
snd_pcm               114688  1 snd_bcm2835
rfkill                 32768  3 bluetooth,cfg80211
videodev              253952  7 bcm2835_isp,bcm2835_codec,videobuf2_common,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
snd_timer              32768  3 snd_seq,snd_hrtimer,snd_pcm
snd                    77824  9 snd_seq,snd_timer,snd_seq_device,snd_bcm2835,snd_pcm
mc                     45056  6 bcm2835_isp,bcm2835_codec,videobuf2_common,videodev,v4l2_mem2mem,videobuf2_v4l2
vc_sm_cma              32768  2 bcm2835_isp,bcm2835_mmal_vchiq
uio_pdrv_genirq        16384  0
uio                    20480  1 uio_pdrv_genirq
fixed                  16384  0
sch_fq_codel           24576  3
ipv6                  503808  28

Using same recipe with modified 5.15.74 revision and compiled 5.15.74 kernel

Getting issue only with 5.15 not with 5.10

I have noted 5.10 and 5.4 having ***.ko** but 5.15 having ***.ko.xz** file for all the module, Is this ***.ko.xz** creating the issue? or any specific configuration required for 5.15.

pelwell, 2022-10-20:
You need `CONFIG_MODULE_COMPRESS_XZ=y`, but that's set in `bcm2709_defconfig`.

savidha777, 2022-10-21:
Modules are loading with disabled compression flag "CONFIG_MODULE_COMPRESS_XZ=n".

root@raspberrypi3-rdk-camera:~# uname -r
5.15.74-v7
root@raspberrypi3-rdk-camera:~# lsmod
Module                  Size  Used by
hci_uart               40960  1
btbcm                  20480  1 hci_uart
bluetooth             413696  3 hci_uart,btbcm
ecdh_generic           16384  1 bluetooth
ecc                    40960  1 ecdh_generic
imx219                 20480  1
i2c_mux_pinctrl        16384  0
i2c_mux                16384  1 i2c_mux_pinctrl
brcmfmac              335872  0
brcmutil               20480  1 brcmfmac
raspberrypi_hwmon      16384  0
bcm2835_codec          45056  0
bcm2835_v4l2           45056  0
v4l2_mem2mem           36864  1 bcm2835_codec
bcm2835_unicam         45056  0
bcm2835_isp            32768  0
v4l2_dv_timings        36864  1 bcm2835_unicam
sha256_generic         16384  0
v4l2_fwnode            24576  2 bcm2835_unicam,imx219
bcm2835_mmal_vchiq     32768  3 bcm2835_isp,bcm2835_codec,bcm2835_v4l2
v4l2_async             24576  3 bcm2835_unicam,imx219,v4l2_fwnode
videobuf2_dma_contig    20480  3 bcm2835_unicam,bcm2835_isp,bcm2835_codec
videobuf2_vmalloc      16384  1 bcm2835_v4l2
videobuf2_memops       16384  2 videobuf2_dma_contig,videobuf2_vmalloc
videobuf2_v4l2         32768  5 bcm2835_unicam,bcm2835_isp,bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem
snd_bcm2835            24576  0
cfg80211              770048  1 brcmfmac
videobuf2_common       65536  9 bcm2835_unicam,bcm2835_isp,bcm2835_codec,videobuf2_dma_contig,videobuf2_vmalloc,videobuf2_memops,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
snd_pcm               114688  1 snd_bcm2835
rfkill                 32768  3 bluetooth,cfg80211
videodev              249856  9 bcm2835_unicam,imx219,bcm2835_isp,bcm2835_codec,videobuf2_common,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2,v4l2_async
snd_timer              32768  1 snd_pcm
mc                     45056  9 bcm2835_unicam,imx219,bcm2835_isp,bcm2835_codec,videobuf2_common,videodev,v4l2_mem2mem,videobuf2_v4l2,v4l2_async
snd                    86016  3 snd_timer,snd_bcm2835,snd_pcm
vc_sm_cma              32768  2 bcm2835_isp,bcm2835_mmal_vchiq
uio_pdrv_genirq        16384  0
uio                    20480  1 uio_pdrv_genirq
fixed                  16384  3
sch_fq_codel           24576  3
ipv6                  524288  26
root@raspberrypi3-rdk-camera:~#