raspberrypi/linux #5208: Missing FSIA6B driver/module.

pgold, 2022-10-15:
### Describe the bug

The fsia6b driver is not available on Raspberry Pi's Linux kernel build.

### Steps to reproduce the behaviour

When trying to load the fsia6b module:
`$ sudo modprobe fsia6b`

I get the following message: "modprobe: FATAL: Module fsia6b not found in directory /lib/modules/5.4.0-1071-raspi"

### Device (s)

Raspberry Pi 4 Mod. B

### System

`cat /etc/rpi-issue` (not available, I'm using Ubuntu 20.04).

```
$ vcgencmd version

Nov 18 2021 16:17:39 
Copyright (c) 2012 Broadcom
version d9b293558b4cef6aabedcc53c178e7604de90788 (clean) (release) (start_x)
```

```
$ uname -a

Linux roompi 5.4.0-1071-raspi #81-Ubuntu SMP PREEMPT Mon Sep 26 15:53:11 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux
```

### Additional context

I'm trying to use `inputattach` with a fsia6b compatible controller, but it does not work unless the kernel module is loaded. It works if I build the module out-of-tree and `insmod` it, but it'd be nice if the module was part of the build (I see that other joystick drivers are included already).

Happy to open PRs if it is agreed that the driver should be added as a module to the kernel.

pelwell, 2022-10-15:
I'm happy to add a missing module, but 5.4 has already been superceded by two major kernel versions - 5.10 and 5.15. Even if we add the required config setting in rpi-5.4.y (which shouldn't really happen because it is in legacy, bug fix only mode), would it make its way into an Ubuntu release?

pgold, 2022-10-15:
Thanks for looking into it, @pelwell .

Regarding the kernel version, I agree that it should only be added to the last one (for my particular use case, I plan on upgrading to Ubuntu 22.04 soon (linux 5.15)).

I'd expect it to make to an Ubuntu release, as 20.04 is an LTS version and AFAIK they still push new 5.4.y kernels.

pelwell, 2022-10-16:
The `fsia6b` module has been enabled in all current (non-`oldstable`) branches.

pgold, 2022-10-17:
Many thanks, @pelwell !

closed 2022-10-17 by pgold

juergh, 2022-10-17:
FWIW, Ubuntu is not following the raspberrypi kernel configs so rpi config changes won't make it magically into an Ubuntu kernel. @pgold You need to open an Ubuntu bug for that: https://bugs.launchpad.net/ubuntu/+source/linux-raspi/+filebug