raspberrypi/linux #5213: .DTB files not generated in 6.1

by, 2022-10-18:
### Describe the bug

When compiling the new kernel 6.1, the .DTB files are not generated (only .DTS), so the final installation fails with
`cp: cannot stat 'arch/arm64/boot/dts/broadcom/*.dtb': No such file or directory
cp: cannot stat 'arch/arm64/boot/dts/overlays/*.dtb*': No such file or directory`

The respective directory shows
`~/linux/arch/arm64/boot/dts/broadcom $ ls -al
total 88
drwxr-xr-x  5 pi pi 4096 Oct 18 09:31 .
drwxr-xr-x 35 pi pi 4096 Oct 18 09:31 ..
-rw-r--r--  1 pi pi   56 Oct 18 09:31 bcm2710-rpi-2-b.dts
-rw-r--r--  1 pi pi   56 Oct 18 09:31 bcm2710-rpi-3-b.dts
-rw-r--r--  1 pi pi   61 Oct 18 09:31 bcm2710-rpi-3-b-plus.dts
-rw-r--r--  1 pi pi   56 Oct 18 09:31 bcm2710-rpi-cm3.dts
-rw-r--r--  1 pi pi   61 Oct 18 09:31 bcm2710-rpi-zero-2.dts
-rw-r--r--  1 pi pi   61 Oct 18 09:31 bcm2710-rpi-zero-2-w.dts
-rw-r--r--  1 pi pi   56 Oct 18 09:31 bcm2711-rpi-400.dts
-rw-r--r--  1 pi pi   56 Oct 18 09:31 bcm2711-rpi-4-b.dts
-rw-r--r--  1 pi pi   56 Oct 18 09:31 bcm2711-rpi-cm4.dts
-rw-r--r--  1 pi pi   74 Oct 18 09:31 bcm2711-rpi-cm4-io.dts
-rw-r--r--  1 pi pi   57 Oct 18 09:31 bcm2711-rpi-cm4s.dts
-rw-r--r--  1 pi pi   76 Oct 18 09:31 bcm2837-rpi-3-a-plus.dts
-rw-r--r--  1 pi pi   71 Oct 18 09:31 bcm2837-rpi-3-b.dts
-rw-r--r--  1 pi pi   76 Oct 18 09:31 bcm2837-rpi-3-b-plus.dts
-rw-r--r--  1 pi pi   75 Oct 18 09:31 bcm2837-rpi-cm3-io3.dts
-rw-r--r--  1 pi pi   76 Oct 18 09:31 bcm2837-rpi-zero-2-w.dts
lrwxrwxrwx  1 pi pi   52 Oct 18 09:31 bcm283x-rpi-csi1-2lane.dtsi -> ../../../../arm/boot/dts/bcm283x-rpi-csi1-2lane.dtsi
lrwxrwxrwx  1 pi pi   49 Oct 18 09:31 bcm283x-rpi-lan7515.dtsi -> ../../../../arm/boot/dts/bcm283x-rpi-lan7515.dtsi
drwxr-xr-x  2 pi pi 4096 Oct 18 09:31 bcmbca
-rw-r--r--  1 pi pi  916 Oct 18 09:31 Makefile
drwxr-xr-x  2 pi pi 4096 Oct 18 09:31 northstar2
drwxr-xr-x  2 pi pi 4096 Oct 18 09:31 stingray`

### Steps to reproduce the behaviour

`git clone --branch rpi-6.1.y https://github.com/raspberrypi/linux
cd linux
KERNEL=kernel8
make bcm2711_defconfig
make prepare
make -j4 Image.gz modules dtbs
sudo make modules_install`
`# (all directories in /boot are in place already)`
`sudo cp arch/arm64/boot/dts/broadcom/*.dtb /boot/NTP/; sudo cp arch/arm64/boot/dts/overlays/*.dtb* /boot/NTP/overlays-NTP/; sudo cp arch/arm64/boot/dts/overlays/README /boot/NTP/overlays-NTP/; sudo cp arch/arm64/boot/Image.gz /boot/kernel8-NTP.img`

### Device (s)

Raspberry Pi 4 Mod. B

### System

`Raspberry Pi reference 2021-10-30`
`Aug 26 2022 14:03:16 
Copyright (c) 2012 Broadcom
version 102f1e848393c2112206fadffaaf86db04e98326 (clean) (release) (start)`
`Linux chronos 6.0.2-v8-NP-NTP+ #1 SMP Mon Oct 17 17:15:26 CEST 2022 aarch64 GNU/Linux`

pelwell, 2022-10-18:
Hmm, the arm64 defconfigs are broken in a big way - they don't even set `CONFIG_ARCH_BCM2835`. Give us a while to check it over.

pelwell, 2022-10-18:
You're OK - you managed to stay on the "efficient" side of the line.

I've pushed part one of the config fix - defining CONFIG_ARCH_BCM=y was enough to fix the majority of the breakage. The rest will take a bit longer.

pelwell, 2022-10-18:
The defconfigs should be equivalent to those in rpi-6.0.y now. It wouldn't have take half as long if the kernel tool `scripts/config` didn't turn `scripts/config -m I2C_MUX_PCA954x` into `CONFIG_I2C_MUX_PCA954X=m` (note the capital `X`), which is ignored by Kconfig.

by, 2022-10-18:
Thank you Phil!

by, 2022-10-18:
Builds and runs great, thank you!

closed 2022-10-18 by by

pelwell, 2022-10-18:
Do report any further issues you find, especially if they're regressions.