raspberrypi/linux #5217: WD_BLACK SN850X (Gen 4) NVMe not detected (via PCIe, CM4)

agners, 2022-10-19:
### Describe the bug

We try to use a WD_BLACK SN850X NVMe (with PCIe Gen 4) with the CM4. It seems that on our custom hardware as well as on the CM4IO board (with a PCIe slot to M.2 adapter) the NVMe is not detected. It seems the PCIe link stays down:

```
[    1.357362] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    1.357401] brcm-pcie fd500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    1.357494] brcm-pcie fd500000.pcie:      MEM 0x0600000000..0x063fffffff -> 0x00c0000000
[    1.357587] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x007fffffff -> 0x0400000000
[    1.953444] brcm-pcie fd500000.pcie: link down
```

### Steps to reproduce the behaviour

Connect a WD_BLACK SN850X NVMe with a CM4, observe link down message in the kernel logs.

### Device (s)

Raspberry Pi CM4 Lite

### System

```
pi@raspberrypi:~$ cat /etc/rpi-issue
Raspberry Pi reference 2022-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 8a42abcd1dbd8c9c1fdfca4e0c3778255b2f9cc4, stage4
pi@raspberrypi:~$ vcgencmd version
Aug 26 2022 14:03:16 
Copyright (c) 2012 Broadcom
version 102f1e848393c2112206fadffaaf86db04e98326 (clean) (release) (start)
pi@raspberrypi:~$ uname -a
Linux raspberrypi 5.15.61-v7l+ #1579 SMP Fri Aug 26 11:13:03 BST 2022 armv7l GNU/Linux

```

### Logs

[dmesg.txt](https://github.com/raspberrypi/linux/files/9822480/dmesg.txt)

timg236, 2022-10-19:
Does this happen with SD/EMMC boot or only if NVMe boot is enabled?

agners, 2022-10-19:
This is with SD boot. I also tried `usbboot -d mass-storage-gadget` and checked the console.

I did not try to install something on the drive and boot, I guess I'd have to connect to a PC to do that.

timg236, 2022-10-20:
If it’s failing from SD boot then it’s more likely to be a hardware issue. PCI is initialised once by Linux and the link never comes up.
Do you have any other PCI devices to try?

It’s worth checking the pins on the CM4 and making sure that none are bent out of shape

agners, 2022-10-20:
> Do you have any other PCI devices to try?

@timg236 yeah I have quite some hardware around, as I am part of the team building [Yellow](https://www.crowdsupply.com/nabu-casa/home-assistant-yellow/).

So the following two work on my desk with the same setup:
- Samsung 970 EVO Plus 1TB (MZ-V7S1T0)
- ADATA SX8200PNP 512GB

I also used another CM4 (with eMMC), and the WD_BLACK SN850X seems to fail there too.

agners, 2022-10-20:
I am not a PCIe expert, but it seems to me that the link training fails for some reason. These seems to be true when Linux is trying to do the link as well as U-Boot:
```
HAOS> pci
PCIe BRCM: link down
```

I am guessing the U-Boot PCIe initialization sequence is lifted from Linux, but not sure.

With NVMe boot it would be the EEPROM firmware which first initiates the link, maybe that can behave differently? Or is the exact same sequence used in firmware?

I've talked to @geerlingguy briefly, he confirms that WD black are problematic with CM4. He says also the SN750 cause problems.

agners, 2022-10-20:
> With NVMe boot it would be the EEPROM firmware which first initiates the link, maybe that can behave differently? Or is the exact same sequence used in firmware?

I tried directly booting from the NVMe, and it seems not to work either. The green LED stays steadily on.

The screen shows timeout errors (this is with the CM4 Lite & without SD card plugged in):
[image]

timg236, 2022-10-20:
There’s very little software involved in link setup it just enabled the hardware and waits for the link to come up or timeout. It’s unlikely that the bootloader would succeed if Linux fails. I was just checking in case there was a failure handing over the PCI setup ft boot loader to Linux but it appears that the link never comes up.

agners, 2022-10-20:
Ugh, I see, so its likely a hardware problem then? Technically these things should be compatible, as PCIe is backwards compatible. And the WD devices do work in PCs, so maybe the BCM2711 PCIe IP is somehow getting in trouble with that particular device? If that is the case, I am guessing there is not much we can do for current CM4 users.

timg236, 2022-10-20:
I’ll pass this on to our PCI experts, not sure if there is anything that can be done if the link won’t come up though

timg236, 2022-10-20:
What sort of M2 adapter are you using, is it powered? It's possible that this NVMe requires more power than the CM can supply just through the PCIe socket.

agners, 2022-10-20:
I am using the [Delock PCI Express x1 to M.2 Key M Adapter (64105)](https://www.delock.com/produkt/64105/kaufen.html), so no, it's not powered.

However, on our Home Assistant Yellow PCB, we use a relatively high powered DC/DC to provide 3.3V. Our hardware engineer has verified that it can supply the amperage specified in the PCIe M.2 standard (I think 2.7A). The device does not work on that platform either.

agners, 2022-10-20:
FWIW, you can find the schematics of the Home Assistant Yellow PCB here: https://yellow.home-assistant.io/resources/Yellow_1.2c_Schematic_git.0b4049bec10a.pdf.

I consider the Yellow HW a bit better for NVMe M.2 tests as it has the actual M.2 connector on-board, so no additional damping due to adapters etc...

timg236, 2022-10-20:
> @timg236 yeah I have quite some hardware around, as I am part of the team building [Yellow](https://www.crowdsupply.com/nabu-casa/home-assistant-yellow/).
>
> So the following two work on my desk with [...]

We'll probably get one. Please could you confirm the exact model e.g.
https://www.amazon.co.uk/WD_BLACK-Internal-Gaming-Technology-Heatsink/dp/B08PHSVW7K

agners, 2022-10-20:
> The board appears to be using the CM4's onboard 3.3v regulator to power the NVMe slot. The maximum current draw permissible from the 3.3v pins is 0.6A, which is a lot less than most NVMe drives will consume.

+3.3VP is supplied by the on-board DC/DC U17 (BD9D321EFJ). The power curcuit is designed to provide 3A.

P33M, 2022-10-21:
Huh that's weird. This drive doesn't work on a CM4IO board, with or without the PCIe analyzer interposer in-circuit. The adapter I'm using has a 12V->3.3V 5A regulator. No response at all from the drive's PCIe TX.

P33M, 2022-10-21:
The drive isn't detected in 2 different NVMe to USB adapters either. Not much we can do here.

agners, 2022-10-24:
Thanks for your investigation, very much appreciated!

> My guess is the drive simply doesn't do downscaled numbers of active lanes (i.e. it needs 4).

@P33M interesting finding, shouldn't downscaling be mandatory?

I've looked up a bit of PCIe spec, and found this descriptions:

> PCI Express Links must consist of 1, 2, 4, 8, 12, 16, or 32 Lanes in parallel, referred to as x1, x2, x4,
x8, x12, x16, and x32 Links, respectively. All Lanes within a Link must simultaneously (transmit
data based on the same frequency with a skew between Lanes not to exceed LTX-SKEW (Table 4-9).
The negotiation process is described as a sequence of steps.
> The negotiation establishes values for Link number and Lane number for each Lane that is part of a
valid Link; each Lane that is not part of a valid Link exits the negotiation to become a separate Link
or remains in Electrical Idle.

> The ability for a xN Port to form a xN Link as well as a x1 Link (where N can be 32, 16, 12, 8, 4,15
2, and 1) is required.

So that would mean this drive violates the PCIe spec?

Maybe it is able to do it, but the negotiation fails for some reason?

> The drive isn't detected in 2 different NVMe to USB adapters either.

Do they happen to be non-x4 adapters?

P33M, 2022-10-24:
> Thanks for your investigation, very much appreciated!
> 
> > My guess is the drive simply doesn't do downscaled numbers of active lanes (i.e. it needs 4).
> 
> @P33M shouldn't downscaling be mandatory?
> I've looked up a bit of PCIe spec, and found this descriptions:
> 
> So that would mean this drive violates the PCIe spec?
Yes
> Maybe it is able to do it, but the negotiation fails for some reason?
I don't see any Gen1 traffic out of TX at all, so it never gets to the Polling.Active state.
> > The drive isn't detected in 2 different NVMe to USB adapters either.
> 
> Do they happen to be non-x4 adapters?
Quite likely.

agners, 2022-10-24:
Just realized my [Delock adapter](https://www.delock.com/produkt/64105/kaufen.html) is x1 only (of course), and Linux detects the NVMe on my ASUS ROG STRIX X570-E motherboard. :thinking:

6by9, 2022-10-24:
@P33M One that I hit previously for lower performance devices is that they didn't like ```(struct brcm_pcie *pcie)->gen``` defaulting to 0 and resulted in "link down".
Adding ```max-link-speed = <1>``` to the pcie host controller config to force gen 1 resulted in them working.

I thought I'd added that as an option to one of the overlays, but I can't see it.

P33M, 2022-10-24:
Link training always starts in gen 1, and I see that on the analyzer - max-link-speed will suppress the subseqent directed link speed change if it's set to 1.

agners, 2022-10-26:
FWIW, that is the kernel log of that device in my computer using the Delock adapter (x1 only)

```
kernel: pci 0000:05:00.0: [15b7:5030] type 00 class 0x010802
kernel: pci 0000:05:00.0: reg 0x10: [mem 0xfc900000-0xfc903fff 64bit]
kernel: pci 0000:05:00.0: 15.753 Gb/s available PCIe bandwidth, limited by 16.0 GT/s PCIe x1 link at 0000:03:02.0 (capable of 63.012 Gb/s with 16.0 GT/s PCIe x4 link)
kernel: pci 0000:05:00.0: Adding to iommu group 25
kernel: nvme nvme2: pci function 0000:05:00.0
kernel: nvme nvme2: 32/0/0 default/read/poll queues
kernel:  nvme2n1: p1 p2
kernel:  nvme2n1: p1 p2
```

Blindmikey, 2023-03-03:
Possibly relevant: https://community.wd.com/t/wd-black-sn850-not-detected-by-bios-on-bootup-from-power-off/259168

P33M, 2023-09-20:
This device needs the SUSCLK (32kHz) to be present on NVMe connector pin 68 for it to wake up. As the PCIe CEM connector (as used on the CM4IO board) doesn't have a pin that could perform this function, you need a custom CEM to NVMe adapter that generates the clock.

The Startech PEX4M2E1 adapter that I have has a header for DEVSLP/SUSCLK, so I could inject 32kHz and prove that the drive needs it, but on the website it looks like a different PCB version has been created without it.

The vast majority of $random_cheap_adapters don't break out anything other than an LED at most, but someone adventurous could take something like https://www.amazon.co.uk/dp/B07ZJXNW4Y and bolt a Pico to it that's programmed to output 32khz on a gpio. Other than that, there's nothing more that can be done here.

closed 2023-09-20 by P33M

agners, 2023-09-20:
> This device needs the SUSCLK (32kHz) to be present on NVMe connector pin 68 for it to wake up.

Interesting, how did you find this?

I've previously used the WD_BLACK on an ODROID-M1 board, and from what I can tell from its schematic the signal you mention is not connected there either. Yet the NVMe SSD seems to work fine on that board.

We'll do some investigation based on this finds.

havokentity, 2025-05-11:
Hi,

I have nothing to do with this thread or rasberrypi/linux but I've seen the SN850X have this issue with SEVERAL motherboards and MSI boards especially. It will detect and work after 1-2 warm boots.

You can look online and see many people have this issue with the WD BLACK SN850X. Even with the newer AM4/AM5 platforms and x670e motherboards