raspberrypi/linux #5219: Raspberry is frozen when restart many time (every 3 minutes)

vuongvt-rights, 2022-10-21:
### Describe the bug

I installed OS 5.15 in my raspberry. When I try to restart my raspberry every 3 minutes, my raspberry is frozen.
I tried to install OS 5.10 in other raspberry and I can not reproduce this bug.

### Steps to reproduce the behaviour

1/ Installed OS 5.15
2/ Restart raspberry in every 3 minutes

### Device (s)

Raspberry Pi 4 Mod. B

### System

> Which OS and version (cat /etc/rpi-issue)?
```
Raspberry Pi reference 2022-01-28
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, fbe448ccdc995d295d24c7596e5f0ef62cc2488f, stage4
```

> Which firmware version (vcgencmd version)?
```
Aug 26 2022 14:03:16 
Copyright (c) 2012 Broadcom
version 102f1e848393c2112206fadffaaf86db04e98326 (clean) (release) (start)
```

> Which kernel version (uname -a)?
```
Linux rp7 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux
```

pucgenie, 2022-10-28:
Happens to me too, regardless of Kernel version (5.15, 6.0.y, 6.1.y).
The last firmware update now changed the ratio of "working restarts to non-working restarts" from ~ 2:8 (yes, that bad) to ~ 7:3 (unusable anyways). Changing WAKE_ON_GPIO didn't fix it.
The green LEDs blink code says "Kernel image not found". config.txt is being read correctly, at least up until the rainbow screen setting - changing that works and reflects the 0/1 state of the option, just booting the Kernel doesn't work every time (when restarting).
journalctl tells me something about a damaged journal file at every boot (regardless of poweroff or restart).

OT: On a Pi3B+ I had to take out the microSDXC for a few hours because it was unreadable after a restart. So as a workaround you could try using other SD cards.

What's the make and model of your boot medium?

hailfinger, 2023-01-28:
Some older Raspberry Pi 4 B (before revision 1.4) may have trouble reading the SD card on reboot. This has been fixed in newer EEPROM versions. Run rpi-eeprom-update to check if an update is available.

See https://github.com/raspberrypi/rpi-eeprom/blob/debian/bullseye/firmware/release-notes.md#2022-12-07---fix-sd-voltage-reset-on-pi4-r11-defaultstablebeta for details.