raspberrypi/linux #5221: RPi4 - 5.15.74-v8+ Power LED can't be turned off anymore

klslz, 2022-10-22:
### Describe the bug

On my RPI4 with

kernel:
```
Linux devpi 5.15.74-v8+ #1594 SMP PREEMPT Fri Oct 21 16:38:33 BST 2022 aarch64 GNU/Linux
```

firmware:
```
CURRENT: Tue 18 Oct 10:57:44 UTC 2022 (1666090664)
```

and with following settings in config.txt:

```
###turn off onboard LEDs###################
dtoverlay=act-led

###disable the ACT LED
dtparam=act_led_trigger=none
dtparam=act_led_activelow=off  

###disable the PWR LED
dtparam=pwr_led_trigger=none
dtparam=pwr_led_activelow=off

###turn off ethernet port LEDs
dtparam=eth_led0=4
dtparam=eth_led1=4

#########################################
```
my PWR LED won't turn off anymore as it used to do.
ACT_LED and ETH_LED turn off as expected.

Anybody else experienced this?

THX

### Steps to reproduce the behaviour

I tried two different PI4 with above config.

### Device (s)

Raspberry Pi 4 Mod. B

### System

```
raspinfo
System Information
------------------

Raspberry Pi 4 Model B Rev 1.2
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"

Raspberry Pi reference 2022-05-14
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, ca06d62f2ef63bd00b5729e48c64a2688e46ae14, stage2

Linux audian 5.15.74-v8+ #1594 SMP PREEMPT Fri Oct 21 16:38:33 BST 2022 aarch64 GNU/Linux
Revision	: b03112
Serial		: 100000009b4ffd80
Model		: Raspberry Pi 4 Model B Rev 1.2
Throttled flag  : throttled=0x0
[... 519 lines cut ...]
[    1.453643] systemd[1]: Finished Load Kernel Module fuse.
[    1.456439] systemd[1]: Mounting FUSE Control File System...
[    1.463672] systemd[1]: Mounting Kernel Configuration File System...
[    1.466193] systemd[1]: Started File System Check Daemon to report status.
[    1.470612] systemd[1]: Finished Load Kernel Modules.
[    1.475382] systemd[1]: Starting Apply Kernel Variables...
[    1.495220] systemd[1]: Mounted Kernel Configuration File System.
[    1.495916] systemd[1]: Mounted FUSE Control File System.
[    1.510909] systemd[1]: Finished File System Check on Root Device.
[    1.514574] systemd[1]: Starting Remount Root and Kernel File Systems...
[    1.544553] systemd[1]: Finished Apply Kernel Variables.
[    1.556756] systemd[1]: Started Journal Service.
[    1.604305] EXT4-fs (sda2): re-mounted. Opts: (null). Quota mode: none.
[    1.625247] systemd-journald[131]: Received client request to flush runtime journal.
[    1.926433] mc: Linux media interface: v0.10
[    1.980907] videodev: Linux video capture interface: v2.00
[    2.070238] rpivid_hevc: module is from the staging directory, the quality is unknown, you have been warned.
[    2.083561] rpivid feb10000.codec: Device registered as /dev/video19
[    2.186181] usb 1-1.3: 1:3 : unsupported format bits 0x100000000
[    2.216722] usbcore: registered new interface driver snd-usb-audio
[    2.901460] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[    2.902073] bcmgenet fd580000.ethernet eth0: Link is Down
[    6.975102] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   31.710996] cam-dummy-reg: disabling
```

pelwell, 2022-10-22:
Duplicate of https://github.com/raspberrypi/firmware/issues/1742 and https://github.com/raspberrypi/linux/issues/5151.

closed 2022-10-22 by pelwell

klslz, 2022-10-23:
OK.  Below config.txt snippet is the way it works now.
All LEDs are off.

```
###turn off onboard LEDs###################
dtoverlay=act-led

###disable the ACT LED
dtparam=act_led_trigger=none
dtparam=act_led_activelow=off 

###disable the PWR LED
dtparam=pwr_led_trigger=default-on
dtparam=pwr_led_activelow=off

###turn off ethernet port LEDs
dtparam=eth_led0=4
dtparam=eth_led1=4

##########################################

```

However:

These new settings need to be described in ./overlays/README
The README doesn't mention these at all. 
The README also falsely claims that pwr_led_activelow is OFF by default, since
the pwr_leds would equal act_leds. That doesn't work. You need to specify
dtparam=pwr_led_activelow=off to turn it off.

README snippet

```
        act_led_trigger       Choose which activity the LED tracks.
                                Use "heartbeat" for a nice load indicator.
                                (default "mmc")

        act_led_activelow       Set to "on" to invert the sense of the LED
                                (default "off")
                                N.B. For Pi 3B, 3B+, 3A+ and 4B, use the act-led
                                overlay.

        act_led_gpio            Set which GPIO to use for the activity LED
                                (in case you want to connect it to an external
                                device)
                                (default "16" on a non-Plus board, "47" on a
                                Plus or Pi 2)
                                N.B. For Pi 3B, 3B+, 3A+ and 4B, use the act-led
                                overlay.

        pwr_led_trigger
        pwr_led_activelow
        pwr_led_gpio
                                As for act_led_*, but using the PWR LED.
                                Not available on Model A/B boards.

```

The ticket shouldn't be closed until the README is updated properly and accordingly.

Keep also in mind, there are numerous HowTos and manuals on the net describing the old 
behaviour! This change has a major impact out there. This ALL-LED-off feature is widely 
used out there!

As mentioned in the other ticket. The act_led  should act in a consistent way.
It's not nice that it doesn't.

THX

pelwell, 2022-10-23:
The documentation is correct. The only thing that was wrong was the behaviour of the _old_ firmware, which is something we can't change (except by getting people to replace it with the new firmware).