raspberrypi/linux #5222: Please add support (dtoverlay) for AHTX (i.e. AHT10) family of sensors

Jibun-no-Kage, 2022-10-24:
### Describe the bug

Not a but but asked where to post this request, and the firmware repo guys suggested here.

### Steps to reproduce the behaviour

Not a but but asked where to post this request, and the firmware repo guys suggested here.

### Device (s)

Other

### System

Not a but but asked where to post this request, and the firmware repo guys suggested here.

pelwell, 2022-10-25:
Commits 6dcf32395b596d25e5c36233fd20fb623a58dc16 and c20376da5e61323410d1ffb076db1ae818ccbf59 enable the driver module and add an `aht10` parameter to the `i2c-sensor` overlay. Use with:
```
dtoverlay=i2c-sensor,aht10
```

popcornmix, 2022-10-26:
Updates in latest rpi-update kernel.

Jibun-no-Kage, 2022-10-26:
Excellent, believed that was so, but wanted to confirm.  Should be testing in the next couple of days.  Thank you again for doing this, really appreciate it!

Jibun-no-Kage, 2022-10-31:
Just a footnote, tested, and works great, thanks again!

closed 2022-10-31 by Jibun-no-Kage