raspberrypi/linux #5230: CAN port not found - Failed to detect MCP251xFD

dav2dg, 2022-10-31:
### Describe the bug

Intermittently startup does not recognise the can port.
[    9.855675] mcp251xfd spi0.0 (unnamed net_device) (uninitialized): Failed to detect MCP251xFD (osc=0x00000000).

RPi Device: CM4102032
CAN Device: WaveShare Isolate CAN FD Duo 2.1
Driver: dtoverlay=waveshare-canfdhat-v21-a
Kernel: Linux e45f013eb397 5.15.56-v8+ #1575 SMP PREEMPT Fri Jul 22 20:31:26 BST 2022 aarch64 GNU/Linux

FAILED:
```
pi@e45f013eb397:~ $ dmesg | grep -Ei 'can|spi|mcp2'
[    9.677539] CAN device driver interface
[    9.805941] vc4-drm gpu: [drm] Cannot find any crtc or sizes
[    9.817116] spi_master spi0: will run message pump with realtime priority
[    9.855675] mcp251xfd spi0.0 (unnamed net_device) (uninitialized): Failed to detect MCP251xFD (osc=0x00000000).
[    9.869528] spi_master spi1: will run message pump with realtime priority
[    9.878443] mcp251xfd spi1.0 can0: MCP2518FD rev0.0 (-RX_INT -MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD c:40.00MHz m:20.00MHz r:17.00MHz e:16.66MHz) successfully initialized.
[   11.402957] IPv6: ADDRCONF(NETDEV_CHANGE): can0: link becomes ready
[   22.009467] can: controller area network core
[   22.009589] NET: Registered PF_CAN protocol family
[   22.018445] can: raw protocol
```

WORKED:
```
pi@e45f013eb397:~ $ dmesg | grep -Ei 'can|spi|mcp2'
[    9.496855] CAN device driver interface
[    9.572447] spi_master spi0: will run message pump with realtime priority
[    9.612490] mcp251xfd spi0.0 can0: MCP2518FD rev0.0 (-RX_INT -MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD c:40.00MHz m:20.00MHz r:17.00MHz e:16.66MHz) successfully initialized.
[    9.624664] spi_master spi1: will run message pump with realtime priority
[    9.700235] mcp251xfd spi1.0 can1: MCP2518FD rev0.0 (-RX_INT -MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD c:40.00MHz m:20.00MHz r:17.00MHz e:16.66MHz) successfully initialized.
[    9.745377] vc4-drm gpu: [drm] Cannot find any crtc or sizes
[   11.285772] IPv6: ADDRCONF(NETDEV_CHANGE): can0: link becomes ready
[   12.292164] IPv6: ADDRCONF(NETDEV_CHANGE): can1: link becomes ready
[   23.771614] can: controller area network core
[   23.771734] NET: Registered PF_CAN protocol family
[   23.784423] can: raw protocol
```

### Steps to reproduce the behaviour

Intermittent after reboot

### Device (s)

Raspberry Pi CM4

### System

```
pi@e45f013eb397:~ $ vcgencmd version
Aug  9 2022 13:44:40
Copyright (c) 2012 Broadcom
version 273b410636cf8854ca35af91fd738a3d5f8b39b6 (clean) (release) (start)
pi@e45f013eb397:~ $ uname -a
Linux e45f013eb397 5.15.56-v8+ #1575 SMP PREEMPT Fri Jul 22 20:31:26 BST 2022 aarch64 GNU/Linux
```

### Logs

dmesg:
```

[    8.729339] bcm2835-isp bcm2835-isp: Loaded V4L2 bcm2835-isp
[    8.963056] [drm] Initialized v3d 1.0.0 20180419 for fec00000.v3d on minor 0
[    9.245132] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    9.293763] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    9.326652] cfg80211: loaded regulatory.db is malformed or signature is missing/invalid
[    9.373484] vc4-drm gpu: bound fe400000.hvs (ops vc4_hvs_ops [vc4])
[    9.401144] Registered IR keymap rc-cec
[    9.401397] rc rc0: vc4 as /devices/platform/soc/fef00700.hdmi/rc/rc0
[    9.401640] input: vc4 as /devices/platform/soc/fef00700.hdmi/rc/rc0/input0
[    9.583424] vc4-drm gpu: bound fe400000.hvs (ops vc4_hvs_ops [vc4])
[    9.591573] brcmfmac: F1 signature read @0x18000000=0x15264345
[    9.597449] Registered IR keymap rc-cec
[    9.597844] rc rc0: vc4 as /devices/platform/soc/fef00700.hdmi/rc/rc0
[    9.598093] input: vc4 as /devices/platform/soc/fef00700.hdmi/rc/rc0/input1
[    9.610907] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    9.625750] usbcore: registered new interface driver brcmfmac
[    9.630556] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,4-compute-module.bin failed with error -2
[    9.677539] CAN device driver interface
[    9.698232] vc4-drm gpu: bound fe400000.hvs (ops vc4_hvs_ops [vc4])
[    9.720596] Registered IR keymap rc-cec
[    9.721308] rc rc0: vc4 as /devices/platform/soc/fef00700.hdmi/rc/rc0
[    9.721600] input: vc4 as /devices/platform/soc/fef00700.hdmi/rc/rc0/input2
[    9.725437] vc4-drm gpu: bound fef00700.hdmi (ops vc4_hdmi_ops [vc4])
[    9.760636] Registered IR keymap rc-cec
[    9.760870] rc rc1: vc4 as /devices/platform/soc/fef05700.hdmi/rc/rc1
[    9.761094] input: vc4 as /devices/platform/soc/fef05700.hdmi/rc/rc1/input3
[    9.767006] vc4-drm gpu: bound fef05700.hdmi (ops vc4_hdmi_ops [vc4])
[    9.767821] vc4-drm gpu: bound fe004000.txp (ops vc4_txp_ops [vc4])
[    9.768753] vc4-drm gpu: bound fe206000.pixelvalve (ops vc4_crtc_ops [vc4])
[    9.769466] vc4-drm gpu: bound fe207000.pixelvalve (ops vc4_crtc_ops [vc4])
[    9.769942] vc4-drm gpu: bound fe20a000.pixelvalve (ops vc4_crtc_ops [vc4])
[    9.770454] vc4-drm gpu: bound fe216000.pixelvalve (ops vc4_crtc_ops [vc4])
[    9.771007] vc4-drm gpu: bound fec12000.pixelvalve (ops vc4_crtc_ops [vc4])
[    9.805671] [drm] Initialized vc4 0.0.0 20140616 for gpu on minor 1
[    9.805941] vc4-drm gpu: [drm] Cannot find any crtc or sizes
[    9.817116] spi_master spi0: will run message pump with realtime priority
[    9.855675] mcp251xfd spi0.0 (unnamed net_device) (uninitialized): Failed to detect MCP251xFD (osc=0x00000000).
[    9.869528] spi_master spi1: will run message pump with realtime priority
[    9.878443] mcp251xfd spi1.0 can0: MCP2518FD rev0.0 (-RX_INT -MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD c:40.00MHz m:20.00MHz r:17.00MHz e:16.66MHz) successfully initialized.
[    9.948553] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    9.949605] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    9.957780] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Nov  1 2021 00:37:25 version 7.45.241 (1a2f2fa CY) FWID 01-703fd60
[   10.168065] Process accounting resumed
[   10.446527] uart-pl011 fe201000.serial: no DMA platform data
[   10.606115] 8021q: 802.1Q VLAN Support v1.8
[   10.842611] Adding 3807228k swap on /var/swap.  Priority:-2 extents:9 across:5978112k SSFS
[   11.402957] IPv6: ADDRCONF(NETDEV_CHANGE): can0: link becomes ready
[   11.499004] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   11.912198] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[   11.914049] bcmgenet fd580000.ethernet eth0: Link is Down
[   11.939691] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   11.940769] r8152 2-4:1.0 eth1: carrier on
[   14.613484] systemd-journald[145]: File /var/log/journal/b7cb962df9ec4886967f0781db3b14b7/user-1000.journal corrupted or uncleanly shut down, renaming and replacing.
[   17.316098] Bluetooth: Core ver 2.22

```

pelwell, 2022-10-31:
Is this a power-related issue? A supply that is marginal could easily cause intermittent startup problems.

Try disabling the on-board WiFi (`dtoverlay=disable-wifi`) - it is relatively power hungry and is being started at the same time. If that works, look at your supply (and the USB cable that connects it to the Pi) and make sure it is adequate.

dav2dg, 2022-10-31:
Thanks, I agree this is a potential problem.

I have 30 of these deployed and this is the only one with this specific
problem. Yes so maybe a HAT problem.

The supply is 15W but I will change to direct supply to the CM4 baseboard
via 24V high current capability.

It is worth a shot. Maybe see if I can turn off Wifi.

Thanks

pelwell, 2022-10-31:
> I have 30 of these deployed and this is the only one with this specific
problem.

That sort of information should really go in the initial issue report, as it sets expectations about reproducibility and helps to point towards a possible cause.