raspberrypi/linux #5231: PTP: getting the time of the PHC can cause extts events to be lost

jclark, 2022-10-31:
### Describe the bug

This is an issue with the PTP support introduced in #4151 (drivers/net/phy/bcm-phy-ptp.c).

When a program is reading extts events (which are generated by calls to ptp_clock_event in bcm_ptp_extts_work) and another program is reading the PHC (which will end up calling bcm_ptp_gettimex), some extts events are lost: if a PPS signal is being fed into the CM4 SYNC_OUT pin, then instead of there being one event per second, there will be sometimes be two or more seconds between events.

This causes a problem when trying to use the PTP as a grandmaster and also synchronize the system time to the PHC (e.g. to run NTP as well). The ts2phc program (part of linuxptp) is used to synchronize the PHC to a PPS signal input to the SYNC_OUT pin; this reads extts events.  Then the system time is synchronized to the PHC using either phc2sys program from linuxptp or using chrony with the PHC refclock; both of these use the PTP_SYS_OFFSET_EXTENDED ioctl to get the time of the PHC.

### Steps to reproduce the behaviour

Connect up the SYNC_OUT pin to a PPS source.

Use the testptp program (in linux/master/tools/testing/selftests/ptp/) to  configure the SYNC_OUT pin to be an input pin:

```
sudo ./testptp  -d /dev/ptp0 -L 0,1
```

Then use testptp to read some PTP extts event

```
sudo ./testptp -e 10
```

You should see one event per second, e.g.:

```
external time stamp request okay
event index 0 at 1667216818.000468254
event index 0 at 1667216819.000468335
event index 0 at 1667216820.000468409
event index 0 at 1667216821.000468483
event index 0 at 1667216822.000468549
event index 0 at 1667216823.000468614
event index 0 at 1667216824.000468656
event index 0 at 1667216825.000468714
event index 0 at 1667216826.000468780
event index 0 at 1667216827.000468837
```

Now run chrony with a conf file including refclock line like this:

```
refclock PHC /dev/ptp0 dpoll -2 offset -37 precision 1e-7 pps
```

Now run testptp again (you may need more events):

```
sudo ./testptp -e 100
```

Instead of seeing one event per second, I see some events lost, e.g.

```
external time stamp request okay
event index 0 at 1667209592.000004850
event index 0 at 1667209597.000005139
event index 0 at 1667209598.000005181
event index 0 at 1667209599.000005230
event index 0 at 1667209606.000005523
```

In the above, we see we have lost the events for 1667209593-96 and 1667209600-05.

I believe this is incorrect behaviour and the occurrence of extts events should not be affected by getting the time from the PHC.

### Device (s)

Raspberry Pi CM4
[... 228 lines cut ...]
ptp_clock_event sys=775397380696984, phc=1667209414999999962
PTP_SYS_OFFSET_EXTENDED n_samples=10, sys=775397527975866, pid=42477
PTP_SYS_OFFSET_EXTENDED n_samples=10, sys=775397786856184, pid=42477
PTP_SYS_OFFSET_EXTENDED n_samples=10, sys=775398045666392, pid=42477
PTP_SYS_OFFSET_EXTENDED n_samples=10, sys=775398304528100, pid=42477
ptp_clock_event sys=775398388750785, phc=1667209415999999979
PTP_SYS_OFFSET_EXTENDED n_samples=10, sys=775398563355900, pid=42477
PTP_SYS_OFFSET_EXTENDED n_samples=10, sys=775398822208497, pid=42477
PTP_SYS_OFFSET_EXTENDED n_samples=10, sys=775399081034427, pid=42477
PTP_SYS_OFFSET_EXTENDED n_samples=10, sys=775399339952059, pid=42477
ptp_clock_event sys=775399397096434, phc=1667209416999999984
PTP_SYS_OFFSET_EXTENDED n_samples=10, sys=775399598816026, pid=42477
PTP_SYS_OFFSET_EXTENDED n_samples=10, sys=775399857677566, pid=42477
PTP_SYS_OFFSET_EXTENDED n_samples=10, sys=775400116607902, pid=42477
PTP_SYS_OFFSET_EXTENDED n_samples=10, sys=775400375386851, pid=42477
PTP_SYS_OFFSET_EXTENDED n_samples=10, sys=775400634254151, pid=42477
```

### Additional context

The dpoll value in the chrony refclock line affects how often the problem occurs.  With `dpoll 4` (which does a  PTP_SYS_OFFSET_EXTENDED ioctl every 16 seconds with 10 samples) it happens roughly every 10 to 15 minutes.  When it does happen it "clusters":  it will miss one timestamp and then another 16 or 32 seconds later; it will do this several times; then it will not miss one for 10 or 15 minutes. With both chrony and phc2sys, the interval between the ioctl is an integral number of seconds.

This is consistent with a hypothesis that an extts event will be missed if the edge happens while a call to bcm_ptp_gettimex is happening.

I noticed that the extts signal is enabled by setting the NSE_SYNC1_FRAMESYNC flag in bcm_ptp_extts_locked (https://github.com/raspberrypi/linux/blob/7f52d634c6e810fbc704d579e8e9614d3959c575/drivers/net/phy/bcm-phy-ptp.c#L677) but that flag appears to be cleared for the duration of bcm_ptp_framesync_ts (which is called by bcm_ptp_gettimex) by the call to bcm_ptp_framesync_disable at (https://github.com/raspberrypi/linux/blob/7f52d634c6e810fbc704d579e8e9614d3959c575/drivers/net/phy/bcm-phy-ptp.c#L225). I wonder if this might be relevant.

lasselj, 2022-11-02:
Hi Janes,

i like your recent work.

Yeah, the issue is that in order to get framesyncs from the PHY for the purposes of capturing a timestamp with say clock_gettime, we have to suspend the function that generates framesyncs as a result of a pulse on the I/O pin as we cannot otherwise tell them apart. The limitation is the hardware. The fewer calls to clock_gettime and clock_gettimex (including via PTP_SYS_OFFSET(_EXTENDED)) the more pulses will be captured. 

@jlemon did I forget anything?

P.S. in modern clients like Timebeat you can adjust per interface both the offset calculation strategy and filter used and the number of samples collected.

[image]

All the best,

Lasse

lasselj, 2024-08-23:
There is some problem here. I am on it. @jlemon any thoughts?

lasselj, 2024-08-24:
Hi,

Ok, I've reviewed again and I can clarify the position.

In the kernel driver during any type of "gettime" request, a framesync between the MAC and the PHY is done. This includes calling clock_gettime, during an IOCTL of type PTP_SYS_OFFSET_EXTEND and its relatives. 

If any of these transactions coincide with a pulse incoming on a pin, the pulse will not be registered as it uses also the framesync mechanism. 

In software that takes an integrated view of capturing pulses and syncing the PHC with the system clock the following approach can be taken (I thank @jclark for cutting this out in cardboard for me more or less): Whenever a pulse is due to arrive, don't ask the PHC what time it is. So in Timebeat for instance, when we see a pulse, we wait 900ms, then we lock a mutex for 200ms and during that time we don't allow any other PHC interaction. This works fine when the software is monolithic. However, if you run two different pieces of software like ts2phc and phc2sys, perhaps it is more difficult (I am not very familiar with them, there may be a way to sync their operation. I don't know if there is).

I am not aware of a way to fix this problem by modifying the driver in the kernel nor is it a problem that I know exists anywhere else than in the BCM54210PE PHY. (and so for that reason Timebeat checks /proc/cpuinfo to see if it is indeed running on a CM4 - and I suspect in the future a CM5 - before implementing the workaround discussed above).

I hope that makes sense. Only too happy to field questions on the subject.

All the best,

Lasse

returntoreality, 2024-09-03:
I run ts2phc and phc2sys. I found that setting the update rate of phc2sys ("-R") to something fractional helps to trigger it less often. Another idea to prevent the issue: comparing the clocks directly after receiving a timestamp event

jclark, 2024-09-04:
> Another idea to prevent the issue: comparing the clocks directly after receiving a timestamp event

I implemented this approach in my project and I can confirm that it works well.