raspberrypi/linux #5234: Please add MCP9808 sensor support to i2c-sensor dtoverlay list

Jibun-no-Kage, 2022-10-31:
### Describe the bug

Not a bug but a support request.  Please add MCP9808 sensor support to i2c-sensor dtoverlay list.  As I understand it, this sensor already is supported by the Linux kernel?  Thanks.

### Steps to reproduce the behaviour

NA

### Device (s)

Other

### System

Latest Raspberry Pi firmware image (or so I believe what is how support is implemented).

### Additional context

Just a suggestion but maybe setup a sub section on github under the Raspberry Pi area for such requests?

pelwell, 2022-11-01:
Done - see https://github.com/raspberrypi/linux/commit/ac4cf1b984f0b3473460af866eef536765b60865.

GitHub doesn't allow the creation of arbitrary sections, and requesting additions via the Issues page works for us.

Jibun-no-Kage, 2022-11-02:
Cool, I just did not want to post the requests the wrong way or such.  I should be able to test MCP9808 sensor first thing tomorrow, once I get the update.

Jibun-no-Kage, 2022-11-02:
Is this update published yet?  Ran rpi-update on a Pi4, then added dtoverlay=i2c-sensor,mcp980x,addr=0x18 to /boot/config.txt, and rebooted, but did not see the typical UU on i2c-detect -y 1 scan?  Shows 18 as address rather than UU.

pelwell, 2022-11-02:
Not yet - the last update was on Monday. You can see the release history here: https://github.com/raspberrypi/rpi-firmware/commits/master

Jibun-no-Kage, 2022-11-02:
Excellent, that is going to be a book mark!  Thanks.

popcornmix, 2022-11-04:
rpi-update contains this now.

Jibun-no-Kage, 2022-11-04:
Excellent!  Yup, now see kernel/dtoverlay control of the sensor, i.e. UU in i2c address slot (0x18) in the i2c detect result.

Jibun-no-Kage, 2022-11-04:
Odd... the direct query of temp1_input under kernel/dtoverlay... the result is '62' which does not make sense.  So I pulled the sample test script from Adafruit, and after loading the dependencies, and running the simple test, the results were as expected.

\# cd /root
\# apt install build-essential python3-dev python3-pip python3-smbus git
\# pip3 install RPi.GPIO
\# git clone https://github.com/adafruit/Adafruit_Python_MCP9808.git
\# cd Adafruit_Python_MCP9808
\# python3 setup.py install

\# cd examples
\# python3 simpletest.py
# python3 simpletest.py
Temperature: 27.250*C / 81.050*F
Temperature: 27.312*C / 81.162*F
Temperature: 27.312*C / 81.162*F
Temperature: 27.250*C / 81.050*F

Jibun-no-Kage, 2022-11-07:
Ah, thanks! Look forward to testing.

pelwell, 2022-11-07:
Oh dear. The `mcp980x` driver only supports the `MCP980<x>` family of devices when `x` is in the range 0-3. For MCP9804, MCP9805 and MCP9808 one needs the `jc42` driver (obviously). Looking back at my commit I think the only thing wrong was the README comment, and the fact that it didn't solve your problem.

I'll add a `jc42` option to the i2c-sensor overlay.

pelwell, 2022-11-07:
https://github.com/raspberrypi/linux/commit/41d7f478e800a31dc8d1e85d199bc9e660b12074 is the update to the overlay. To save you waiting for the next build you can download a patched overlay here: https://drive.google.com/file/d/1eWeA3oRhVTrYI1Y57HLoXzy_dhXmQOd8/view?usp=share_link

Just `sudo cp i2c-sensor.dtbo /boot/overlays/`, edit config.txt to say `dtoverlay=i2c-sensor,jc42` and reboot.

Jibun-no-Kage, 2022-11-07:
Wow, I pulled the C source for the support of the MCP9808, but have not had time to review it as yet.  Interesting.

First test is odd, after reboot...

\# cat /sys/devices/platform/soc/fe804000.i2c/i2c-1/1-0018/hwmon/hwmon2/temp1_input
0
\# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         08 09 0a 0b 0c 0d 0e 0f
10: 10 11 12 13 14 15 16 17 UU 19 1a 1b 1c 1d 1e 1f
20: 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f
30: 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f
40: 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f
50: 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f
60: 60 61 62 63 64 65 66 67 UU 69 6a 6b 6c 6d 6e 6f
70: 70 71 72 73 74 75 76 77

Look like the changes are flooding blasting the i2c bus?

I set boot/config.txt to...

dtoverlay=i2c-sensor,jc42
dtoverlay=i2c-sensor,mcp980x,addr=0x18

I pulled the VCC from the sensor, to reset it, and then got...

\# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

Not sure where 0x68 is coming from, I only have 1 device on the i2c bus.

But still can seem to read the sensor right...

\# cat /sys/devices/platform/soc/fe804000.i2c/i2c-1/1-0018/hwmon/hwmon2/temp1_input
62

Of course if I use the python script...

\# python3 simpletest.py
Press Ctrl-C to quit.
Temperature: 28.000*C / 82.400*F
Temperature: 27.938*C / 82.287*F
Temperature: 28.000*C / 82.400*F

pelwell, 2022-11-07:
You need to remove the usage of `mcp980x` - it's probably getting in the way.

Jibun-no-Kage, 2022-11-07:
Testing now.

Jibun-no-Kage, 2022-11-07:
But looking at the actual kernel patch?  They don't list the MCP9808...
https://linux.kernel.narkive.com/fFuXz9Ak/patch-hwmon-add-support-for-jedec-jc-42-4-compliant-temperature-sensors

+ * Microchip MCP9805, MCP98242, MCP98243, MCP9843
+ Prefixes: 'mcp9805', 'mcp98242', 'mcp98243', 'mcp9843'
+ Addresses scanned: I2C 0x18 - 0x1f

So I wonder if the actual kernel patch is missing what is otherwise documented?  I only see references to the MCP9805.

Jibun-no-Kage, 2022-11-07:
I don't have a Pi setup to execute C source correctly, and I have little experience with C in Linux, versus on Macintosh and PC.  If I can setup a C environment, but that would be the actual proof, to run the raw C code that the kernel has, right?

pelwell, 2022-11-07:
https://github.com/raspberrypi/linux/blob/rpi-5.15.y/drivers/hwmon/jc42.c#L175

Jibun-no-Kage, 2022-11-07:
Yup... { MCP_MANID, MCP9808_DEVID, MCP9808_DEVID_MASK },  That is good then,  I am trying to brute force load the driver.

Jibun-no-Kage, 2022-11-07:
I happen to do the above on 64 bit as well! LOL

Jibun-no-Kage, 2022-11-19:
Ah, I will test int he next day or so.  Thanks!

Jibun-no-Kage, 2022-11-22:
Either I have missed something, or this does not seem to be working  Have a Pi Zero, with i2c enabled, and latest rpi-update done.  The i2c module is loaded, but I don't see 'UU' at address 0x18 (for MCP9808.  The jc42 module is loaded as well.

\# lsmod | grep jc42
jc42                   16384  0
\# lsmod | grep i2c
i2c_bcm2835            16384  0
i2c_dev                16384  0

\# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- 18 -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

And /boot/config.txt...

# cat /boot/config.txt | grep i2c-sensor
dtoverlay=i2c-sensor,jc42

There is not temp1_input file, and there should be...
\# find / | grep 1-0018
/sys/devices/platform/soc/20804000.i2c/i2c-1/1-0018
/sys/devices/platform/soc/20804000.i2c/i2c-1/1-0018/subsystem
/sys/devices/platform/soc/20804000.i2c/i2c-1/1-0018/power
/sys/devices/platform/soc/20804000.i2c/i2c-1/1-0018/power/runtime_suspended_time
/sys/devices/platform/soc/20804000.i2c/i2c-1/1-0018/power/autosuspend_delay_ms
/sys/devices/platform/soc/20804000.i2c/i2c-1/1-0018/power/runtime_active_time
/sys/devices/platform/soc/20804000.i2c/i2c-1/1-0018/power/control
/sys/devices/platform/soc/20804000.i2c/i2c-1/1-0018/power/runtime_status
/sys/devices/platform/soc/20804000.i2c/i2c-1/1-0018/waiting_for_supplier
/sys/devices/platform/soc/20804000.i2c/i2c-1/1-0018/modalias
/sys/devices/platform/soc/20804000.i2c/i2c-1/1-0018/uevent
/sys/devices/platform/soc/20804000.i2c/i2c-1/1-0018/name
/sys/devices/platform/soc/20804000.i2c/i2c-1/1-0018/of_node

However if I remove the overlay from the configuration and reboot, then force the driver...
\# lsmod | grep jc42
[No Output]

\# modprobe jc42
\# echo jc42 0x18 > /sys/bus/i2c/devices/i2c-1/new_device

\# /usr/sbin/i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

\# find / | grep 1-0018 | grep input
/sys/devices/platform/soc/20804000.i2c/i2c-1/1-0018/hwmon/hwmon2/temp1_input

\# cat /sys/devices/platform/soc/20804000.i2c/i2c-1/1-0018/hwmon/hwmon2/temp1_input
26062

Then I can access the sensor data as expected.  So clearly the overlay is not working right?

pelwell, 2022-11-23:
Having got my hands on one of these I noticed a warning message in the kernel log:
```
[    9.567522] jc42: probe of 1-0018 failed with error -121
```
Unfortunately, not all devices support the disabling of the SMBUS timeout, and the driver treats attempting to do so on a non-supporting chip as a fatal error. I've pushed an update to the i2c-sensor overlay adding a `no_timeout` parameter - not using the parameter on an MCP9808 makes it work.

Jibun-no-Kage, 2022-11-23:
Dang, I missed that, I should have catch that.  So this will be live via rpi-update when?  As I recall you only push out changes like once a week?

pelwell, 2022-11-23:
Updates are pushed as and when we feel they are needed. It's approaching a week since the last update so I would expect one fairly soon, but in the meantime you can download a trial build from here: https://drive.google.com/file/d/1eWeA3oRhVTrYI1Y57HLoXzy_dhXmQOd8/view?usp=share_link

pelwell, 2022-11-23:
Without a device to test, overlay changes like this are strictly best-guess, which in this case wasn't good enough.

Kudos to Pimoroni for the next-day delivery.

Jibun-no-Kage, 2022-11-26:
Done and done... just got a chance to test this.  Not that I doubted mind you.  Thanks for the test file, worked fine.

closed 2022-11-26 by Jibun-no-Kage

Nodi-Rubrum, 2026-02-24:
I also tried... given the documentation reference to leverage the jc42 option...
```
dtoverlay=i2c-sensor,jc42,mcp980x,addr=0x18
```
Did not recall having to do this previously.  Results...
```
dmesg | grep i2c
[    7.184919] i2c_dev: i2c /dev entries driver
[    7.926749] i2c i2c-1: Failed to register i2c client jc-42.4-temp at 0x18 (-16)
[    7.926758] i2c i2c-1: of_i2c: Failure registering /axi/pcie@1000120000/rp1/i2c@74000/jc42@18
[    7.926763] i2c i2c-1: Failed to create I2C device for /axi/pcie@1000120000/rp1/i2c@74000/jc42@18
```

Nodi-Rubrum, 2026-02-26:
Yup... will give it a shot... I should have some time tomorrow to attempt a bit more testing.  Thanks for the help.

Nodi-Rubrum, 2026-02-26:
Confirmed just using jc42...
```
# cat /boot/firmware/config.txt | grep -E "^dtoverlay"
dtoverlay=dwc2,dr_mode=host
dtoverlay=disable-bt
dtoverlay=i2c-sensor,jc42,addr=0x18
```
Works for polling temperature...
```
# cat /sys/devices/platform/axi/1000120000.pcie/1f00074000.i2c/i2c-1/1-0018/hwmon/hwmon6/temp1_input
25187
```
Thanks again.