raspberrypi/linux #5235: Gpio-ir module does not work

nandlab, 2022-10-31:
### Describe the bug

Although I enabled the `gpio-ir` dtoverlay, I cannot see any lirc device like `/dev/lirc0`.

### Steps to reproduce the behaviour

Uncomment the line `dtoverlay=gpio-ir,gpio_pin=17` in `/boot/config.txt`.

### Device (s)

Raspberry Pi 4 Mod. B

### System

`cat /etc/rpi-issue`:
```
Raspberry Pi reference 2022-01-28
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f01430c9d8f67a4b9719cc00e74a2079d3834d5d, stage4
```
`vcgencmd version`:
```
Oct 26 2022 11:09:05 
Copyright (c) 2012 Broadcom
version c72ad6b26ff40c91ef776b847436094ee63fabee (clean) (release) (start)
```
`uname -a`:
```
Linux nextpi 5.15.74-v7l+ #1595 SMP Wed Oct 26 11:05:08 BST 2022 armv7l GNU/Linux
```

### Logs

`dmesg`:
```
[    5.396569] pinctrl-bcm2835 fe200000.gpio: pin gpio17 already requested by fe200000.gpio; cannot claim for ir-receiver@11
[    5.396600] pinctrl-bcm2835 fe200000.gpio: pin-17 (ir-receiver@11) status -22
[    5.396622] pinctrl-bcm2835 fe200000.gpio: could not request pin 17 (gpio17) from group gpio17  on device pinctrl-bcm2711
[    5.396643] gpio_ir_recv ir-receiver@11: Error applying setting, reverse things back
```

nandlab, 2022-10-31:
It works with gpio 18 though.

6by9, 2022-11-01:
It's told you
```pin gpio17 already requested by fe200000.gpio; cannot claim for ir-receiver@11```

Please quote the full contents of your /boot/config.txt file, as something else is configured to use GPIO 17. (Nothing is configured to use GPIO 18, and therefore it loads).

nandlab, 2022-11-01:
`/boot/config.txt`:
```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Disable compensation for displays with overscan
disable_overscan=1

dtoverlay=pisound

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]
```
How can I find out, which process uses a gpio pin?

6by9, 2022-11-02:
You can use gpioinfo or look in /sys/kernel/debug/pinctrl/fe200000.gpio-pinctrl-bcm2711, however in your case it is fairly obvious as there is only one line different from normal - ```dtoverlay=pisound```

https://github.com/raspberrypi/linux/blob/rpi-5.15.y/arch/arm/boot/dts/overlays/pisound-overlay.dts#L96 claims GPIO 17 for a button on the Pisound. If you don't actually have the Pisound connected, then remove the dtoverlay line.

pelwell, 2022-11-02:
I'm pretty sure that this line (https://github.com/raspberrypi/linux/blob/rpi-5.15.y/arch/arm/boot/dts/overlays/pisound-overlay.dts#L104) is the reason the culprit isn't obvious, since it obscures what is really responsible for claiming the GPIO. I think the line could be deleted, since there is already another pinctrl reference associated with the sound card (https://github.com/raspberrypi/linux/blob/rpi-5.15.y/arch/arm/boot/dts/overlays/pisound-overlay.dts#L83), but it may also be necessary to move the `pinctrl-names` line (https://github.com/raspberrypi/linux/blob/rpi-5.15.y/arch/arm/boot/dts/overlays/pisound-overlay.dts#L83) as well.

pelwell, 2022-11-02:
P.S. I'm aware that this makes mcp3004 different to the other mcp300x overlays, but this is better. At some point we may add the new style of parameters to the existing overlays, or even move them all under a single umbrella overlay, similar to `i2c-sensor`.

nandlab, 2022-12-01:
I disabled `dtoverlay=pisound` and now gpio-ir works with GPIO 17. Thank you!

closed 2022-12-01 by nandlab