raspberrypi/linux #5245: RPi2–3+; Missing from kernel7.img: CONFIG_VMSPLIT_3G_OPT=y

tiago-casal, 2022-11-09:
### Describe the bug

Hello!

If possible, officially in the "arch/arm/configs/bcm2709_defconfig" file, "CONFIG_VMSPLIT_2G=y" could be changed to "CONFIG_VMSPLIT_3G_OPT=y", so those who need to run applications in WINE and the like will have native support without requiring them to rebuild/recompile the kernel or specific knowledge.

Currently with CONFIG_VMSPLIT_2G=y, when trying to use WINE it shows the following error:
> Warning: memory above 0x80000000 doesn't seem to be accessible.
> Wine requires a 3G/1G user/kernel memory split to work properly.
> wine: failed to map the shared user data: c0000018

The old CONFIG_VMSPLIT_3G=y option provided less lowmem, less than 800 MB on the RPi2/RPi3, whereas the CONFIG_VMSPLIT_3G_OPT=y covers more than 920 MB of lowmem.

CONFIG_VMSPLIT_3G_OPT=y Via menuconfig: "Kernel Features" > "Memory split" > "3G/1G user/kernel split (for full 1G low memory)".
I did amateur testing and in my use there were no side effects, please experts could check and if there are no side effects could put such an option in the kernel stable.

Note: CONFIG_VMSPLIT_3G_OPT is available since Kernel 4.4-rc1:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/rawdiff/?id=v4.4-rc1&id2=v4.3

### Steps to reproduce the behaviour

In lxterminal:
1. sudo apt update && sudo apt install wine
2. wineboot

### Device (s)

Raspberry Pi 2 Mod. B, Raspberry Pi 3 Mod. B

### System

#### cat /etc/rpi-issue
> Raspberry Pi reference 2022-09-22
> Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 8a42abcd1dbd8c9c1fdfca4e0c3778255b2f9cc4, stage4

#### vcgencmd version
> Oct 26 2022 11:10:35
> Copyright (c) 2012 Broadcom
> version c72ad6b26ff40c91ef776b847436094ee63fabee (clean) (release) (start)

#### uname -a
> Linux raspberrypi 5.15.76-v7+ #1597 SMP Fri Nov 4 12:13:17 GMT 2022 armv7l GNU/Linux

pelwell, 2022-11-10:
A change like this has the potential to break lots of things in subtle (or not-so-subtle) ways that only affect a few users. It's something we could trial on one of the bleeding edge kernels - rpi-6.1.y, for example - but it sounds like a lot of effort (by the time all the testing has completed).

tiago-casal, 2022-11-13:
> A change like this has the potential to break lots of things in subtle (or not-so-subtle) ways that only affect a few users. It's something we could trial on one of the bleeding edge kernels - rpi-6.1.y, for example - but it sounds like a lot of effort (by the time all the testing has completed).

Hello @pelwell, your comments are interesting! The kernel7l.img for RPi4 has CONFIG_VMSPLIT_3G enabled and this is great, missing for RPi2/RPi3; for RPi2/RPi3 using CONFIG_VMSPLIT_3G doesn't look so good but CONFIG_VMSPLIT_3G_OPT seems to be a good choice, made for devices with up to 1 GB of memory; only when this is put to the test to be confirmed or not.

tiago-casal, 2022-12-15:
Hello friends! Now in December 2022 with Kernel 6.1 released, CONFIG_VMSPLIT_3G_OPT=y will be put to test in bcm2709_defconfig? Thanks!