raspberrypi/linux #5254: Full KMS video driver does not support same HDMI-CEC features as fake KMS

cihlarma, 2022-11-24:
### Describe the bug

With `dtoverlay=vc4-fkms-v3d` my CEC-capable TV automatically turns on and changes to the correct input source when the Raspberry Pi is turned on. With `dtoverlay=vc4-kms-v3d` this does not happen.

### Steps to reproduce the behaviour

1. Flash the latest RPi OS onto an SD card (tested with `2022-09-22-raspios-bullseye-armhf`)
2. Boot
3. Observe the TV not turning on automatically
4. Change `dtoverlay=vc4-kms-v3d` in `config.txt` to `dtoverlay=vc4-fkms-v3d`
5. Reboot
6. Observe the TV now turning on automatically

### Device (s)

Raspberry Pi 4 Mod. B

### System

```
$ cat /etc/rpi-issue 
Raspberry Pi reference 2022-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 8a42abcd1dbd8c9c1fdfca4e0c3778255b2f9cc4, stage4

$ vcgencmd version
Oct 26 2022 11:09:05 
Copyright (c) 2012 Broadcom
version c72ad6b26ff40c91ef776b847436094ee63fabee (clean) (release) (start)

$ uname -a
Linux raspberrypi 5.15.76-v8+ #1597 SMP PREEMPT Fri Nov 4 12:16:41 GMT 2022 aarch64 GNU/Linux
```

### Logs

<details><summary>Full boot dmesg output</summary>

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.15.76-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1597 SMP PREEMPT Fri Nov 4 12:16:41 GMT 2022
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.2
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ac00000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] On node 0, zone DMA32: 16384 pages in unavailable ranges
[    0.000000] percpu: Embedded 28 pages/cpu s77336 r8192 d29160 u114688
[    0.000000] pcpu-alloc: s77336 r8192 d29160 u114688 alloc=28*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v3a
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 1742098
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 951552
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@60 smsc95xx.macaddr=DC:A6:32:7D:A2:9B vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=82b0393a-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] Unknown kernel command line parameters "splash", will be passed to user space.
[... 508 lines cut ...]

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]
gpu_mem=256
[all]
dtoverlay=gpio-fan,gpiopin=14,temp=75000

[pi4]
arm_64bit=1
hdmi_enable_4kp60=1
```
</details>

Related: [documentation/#2699](https://github.com/raspberrypi/documentation/issues/2699)

6by9, 2022-11-24:
It's not a function that the kernel driver should undertake (it was the firmware under legacy and vc4-fkms-v3d).
libcec should be able to do that for you though - IIRC Kodi makes use of it in that way. I can't recall if you need to setup udev rules or not.

cihlarma, 2022-11-24:
Where in the userspace should this be implemented then? Should this be done by the graphical environment, or perhaps by a script run as a service by systemd at boot?

popcornmix, 2022-11-25:
I'd recommend cec-ctl rather than libcec (which is largely unmaintained).

Jibun-no-Kage, 2022-11-26:
Odd thing about cec-ctl there are times if does not seem to work.  For example, I have a Vizio TV that just seems to refuse to work with cec-ctl but works fine with libcec, i.e. cec-client.  Makes no sense but true.

SteelQuantum, 2024-01-07:
> I'd recommend cec-ctl rather than libcec (which is largely unmaintained).

Thank you for that tip. I tried, but unfortunately I didn't get  cec-ctl working either. It seems to have similar types of problems that I have in cec-client now.

I'm new here, and it's possible that I just haven't realized some basic things that I should know. Also I may have posted this to a wrong place - please notice me if so. Anyway I have searched the forums a lot and tested everything quite patiently.

I have been reading these discussions here and in Raspberry Pi forum to get a solution to a problem with libcec with Full KMS. It really seems to me like that, when I switch from kms to fake kms (fkms) in /boot/config.txt, cec-client works with absolutely no errors, just like it's supposed according to the manual. I have understood that probably libcec isn't just working well with Full KMS, but what about cec-ctl?

When Full kms is enabled, cec-client first seems to communicate with the TV: It wakes up with 'on 0' and switches the active source from the TV cable broadcast to the Raspberry Pi with 'as'. But when I switch the TV with its own remote control to another HDMI or to any other source, CEC seems mess up. After that the communication with the TV fails, and will not recover to the original.

My LG television uses CEC version 1.3a, so it might be also a compatibility problem with the versions. But it still works with the Fake KMS, so there's still something different in CEC functionality between the Fake and the Full KMS. And 1.3a isn't really so old, is it? :)  I'd like to test this with a TV that supports CEC 2.0 later. My goal is to get this working with new and old televisions, so trying TV with the CEC 2.0 support wouldn't be the solution, but just a part of the troubleshooting.

**Cec-client with Full KMS:**

When switching the TV to another HDMI from the TV itself, cec-client immediately prints out this:
'DEBUG:   [           48301]	changing physical address to FFFF
DEBUG:   [           48301]	SetDevicePhysicalAddress - not setting invalid physical address ffff'

'scan' command shows that the TV's power status has changed from 'on' to 'unknown', and it also shows up in the cec output:
TV (0): power status changed from 'on' to 'unknown'

After that I try to switch the Raspberry Pi to the active source with 'as', cec-client starts flooding these messages eternally:
DEBUG:   [          340915]	TransmitImageViewOn - 'image view on' already sent
NOTICE:  [          340915]	<< Recorder 1 (1) -> broadcast (F): active source (1000)
TRAFFIC: [          340915]	<< 1f:82:10:00
ERROR:   [          340915]	CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT failed - tx_status=00 errno=64
TRAFFIC: [          340915]	<< 1f:82:10:00
ERROR:   [          340915]	CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT failed - tx_status=00 errno=64
DEBUG:   [          340915]	failed to make 'Recorder 1' the active source. will retry later
DEBUG:   [          341915]	TransmitImageViewOn - 'image view on' already sent

After that has happened, the game is over: I can switch the TV manually to the Raspberry Pi's HDMI input, so that cec-client calms down. I can also still switch the tv on with 'on 0'. But the problem is that, 'as' command is not working anymore.

**Cec-ctl with Full KMS**

I found your reply that cec-ctl is not using libcec, which is probably in the center of this problem, so of course I gave it an opportunity:
sudo cec-ctl -d/dev/cec0 --playback -S --cec-version-1.4
Shows the topology 'almost' correctly, and shows that the TV is on. I have also tested the same with cec 2.0, but there was no change. This is the only point in the outcome that shows something incorrect:
CEC Version                : Tx, OK, Rx, Timeout
Physical Address           : Tx, OK, Rx, Timeout

sudo cec-ctl -d/dev/cec0 --to 0 --image-view-on --cec-version-1.4
This wakes the TV up, at least on the second try.

sudo cec-ctl -d/dev/cec0 --to 0 --active-source phys-addr=1.0.0.0 --cec-version-1.4
This doesn't do anything. Also there's two things wrong in the outprint:
1) Physical Address           : f.f.f.f
2) Logical Address          : Not Allocated

Here's some information on my Raspberry Pi 3 Model B:

$ uname -a
Linux rpi3 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux

$ cat /etc/rpi-issue 
Raspberry Pi reference 2023-12-05
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 446f3e34c8ceb33dfe7bd38578f1cd6b6afe19da, stage2

$ vcgencmd version
Mar 17 2023 10:52:42 
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)

I'd be grateful for any help.

popcornmix, 2024-01-08:
I might be tempted to add:
`vc4.force_hotplug=1` 
(assuming you are using first hdmi port - HDMI_A_0) to cmdline.txt (on same line).
Reboot and see if anything is different.

But this is just a guess.

SteelQuantum, 2024-01-08:
> I might be tempted to add: `vc4.force_hotplug=1` (assuming you are using first hdmi port - HDMI_A_0) to cmdline.txt (on same line). Reboot and see if anything is different.
> 
> But this is just a guess.

Thank you so much! That was the solution.

Cec-client is now working properly and switches the source rapidly by the command to the Raspberry Pi from any other source.

For some reason, cec-ctl is not working, but that's not a problem anymore. I'll go with cec-client or libcec-python.