raspberrypi/linux #5256: Hifiberry Digi2 Pro DAC not recognized by 5.15 kernel

reinisb, 2022-11-25:
### Describe the bug

I have a RPI 4 (Rev 1.2) running Raspberry Pi OS Lite (64-bit) on 5.15.76-v8+. I also have [HiFiberry Digi2 Pro (HW 2.1)](https://www.hifiberry.com/shop/boards/hifiberry-digi2-pro/). The problem is that since update to 5.15 the Hifiberry HAT is not recognized anymore.

/boot/config.txt
```
dtoverlay=
#dtparam=audio=on
dtoverlay=vc4-kms-v3d,noaudio
dtoverlay=hifiberry-digi-pro
```

```
pi@raspberry:~ $ aplay -l
aplay: device_list:274: no soundcards found...

- ```

### Steps to reproduce the behaviour

1. Clean install of Raspberry Pi OS Lite (64-bit, Bullseye).
2. Attach HiFiberry Digi2 Pro (HW 2.1)
3. Edit /boot/config.txt to have the above lines.
4.  aplay -l returns no soundcards found...

### Device (s)

Raspberry Pi 4 Mod. B

### System

System Information
------------------

Raspberry Pi 4 Model B Rev 1.2
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"

Raspberry Pi reference 2022-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 005a8c73b05a2cab394073150208bf4f069e861a, stage2

Linux raspberry 5.15.76-v8+ #1597 SMP PREEMPT Fri Nov 4 12:16:41 GMT 2022 aarch64 GNU/Linux
Revision	: c03112
Serial		: 10000000b23850c7
Model		: Raspberry Pi 4 Model B Rev 1.2
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0, libcamera interfaces=0

Videocore information
---------------------

Oct 26 2022 11:09:05 
Copyright (c) 2012 Broadcom
version c72ad6b26ff40c91ef776b847436094ee63fabee (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       30471668 1658792  27540116   6% /
devtmpfs         1776952       0   1776952   0% /dev
tmpfs            1942712       0   1942712   0% /dev/shm
tmpfs             777088    1020    776068   1% /run
tmpfs               5120       0      5120   0% /run/lock
/dev/mmcblk0p1    261108   31962    229146  13% /boot
[... 562 lines cut ...]
[    6.585515] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[    6.858941] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    6.891271] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    6.901880] cfg80211: loaded regulatory.db is malformed or signature is missing/invalid
[    7.055989] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[    7.057315] bcmgenet fd580000.ethernet eth0: Link is Down
[   11.140508] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[   11.140551] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   13.158060] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net.ipv6.neigh.eth0.retrans_time - use net.ipv6.neigh.eth0.retrans_time_ms instead
[   14.252728] Bluetooth: Core ver 2.22
[   14.252822] NET: Registered PF_BLUETOOTH protocol family
[   14.252828] Bluetooth: HCI device and connection manager initialized
[   14.252844] Bluetooth: HCI socket layer initialized
[   14.252853] Bluetooth: L2CAP socket layer initialized
[   14.252867] Bluetooth: SCO socket layer initialized
[   14.261942] Bluetooth: HCI UART driver ver 2.3
[   14.261964] Bluetooth: HCI UART protocol H4 registered
[   14.262029] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   14.262201] Bluetooth: HCI UART protocol Broadcom registered
[   14.577088] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   14.577108] Bluetooth: BNEP filters: protocol multicast
[   14.577123] Bluetooth: BNEP socket layer initialized
[   14.589869] NET: Registered PF_ALG protocol family
[   15.588481] vc4-drm gpu: [drm] Cannot find any crtc or sizes
[   31.716415] cam-dummy-reg: disabling

pelwell, 2022-11-25:
There's a relevant-looking error message in that output:
```
[    6.003265] wm8804 1-003b: Failed to read device ID: -121
[    6.004305] wm8804: probe of 1-003b failed with error -121
```
Error -121 is EREMOTEIO, which in this case looks like an I2C issue.

1. What does `i2cdetect -y 1` show?
2. And `i2cdump -y 1 0x3b`?
3. What else do you have attached to the Pi 4?
4. Are you using a decent power supply and cable?

pelwell, 2022-11-25:
By the way, you can build yourself a vcdbg equivalent with these commands:
```
$ wget https://raw.githubusercontent.com/raspberrypi/utils/master/vclog/vcdbg.c
$ gcc -o vclog vcdbg.c
```
and run it with:
```
$ sudo ./vclog -m
```

reinisb, 2022-11-25:
I don't have anything else attached to my RPi, only the Hifiberry HAT, ethernet cable and power supply cable. I'm using the official raspberry pi power supply.

I reseated hifiberry and now I get this:
```
pi@raspberry:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
ALSA lib conf.c:1970:(_snd_config_load_with_include) _toplevel_:18:1:Unexpected char
ALSA lib conf.c:3817:(config_file_open) /home/pi/.asoundrc may be old or corrupted: consider to remove or fix it
ALSA lib conf.c:3739:(snd_config_hooks_call) function snd_config_hook_load returned error: Invalid argument
ALSA lib conf.c:4203:(snd_config_update_r) hooks failed, removing configuration
aplay: device_list:283: control open (0): Invalid argument
```

Also, in raspbi-config Audio devices list hifiberry is now showing up (before reseating no device showed up):
`0 snd_rpi_hifiberry_digi `

Vclog returns this:
```
sudo ./vclog -m
005705.843: arasan: arasan_emmc_open
005928.711: brfs: File read: /mfs/sd/config.txt
005929.732: brfs: File read: 2184 bytes
005951.719: HDMI0:EDID error reading EDID block 0 attempt 0
005952.733: HDMI0:EDID giving up on reading EDID block 0
005964.724: HDMI1:EDID error reading EDID block 0 attempt 0
005965.740: HDMI1:EDID giving up on reading EDID block 0
005967.435: brfs: File read: /mfs/sd/config.txt
006762.118: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
006764.643: *** Restart logging
006764.664: brfs: File read: 2184 bytes
006770.091: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
006771.109: hdmi: HDMI0:EDID giving up on reading EDID block 0
006776.152: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
006777.167: hdmi: HDMI0:EDID giving up on reading EDID block 0
006777.186: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
006782.228: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
006783.242: hdmi: HDMI1:EDID giving up on reading EDID block 0
006788.285: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
006789.302: hdmi: HDMI1:EDID giving up on reading EDID block 0
006789.318: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
006789.332: HDMI0: hdmi_pixel_encoding: 300000000
006789.346: HDMI1: hdmi_pixel_encoding: 300000000
006905.962: dtb_file 'bcm2711-rpi-4-b.dtb'
006911.358: brfs: File read: /mfs/sd/bcm2711-rpi-4-b.dtb
006911.376: Loaded 'bcm2711-rpi-4-b.dtb' to 0x100 size 0xcc65
006924.422: brfs: File read: 52325 bytes
006936.447: brfs: File read: /mfs/sd/overlays/overlay_map.dtb
007011.454: brfs: File read: 2231 bytes
007013.522: brfs: File read: /mfs/sd/config.txt
007014.018: Ignoring HAT overlay
007014.519: brfs: File read: 2184 bytes
007031.486: brfs: File read: /mfs/sd/overlays/vc4-kms-v3d-pi4.dtbo
007097.109: Loaded overlay 'vc4-kms-v3d'
007097.121: dtparam: noaudio=true
007272.730: brfs: File read: 3914 bytes
007279.615: brfs: File read: /mfs/sd/overlays/hifiberry-digi-pro.dtbo
007295.222: Loaded overlay 'hifiberry-digi-pro'
007322.959: brfs: File read: 1125 bytes
007326.873: brfs: File read: /mfs/sd/overlays/disable-wifi.dtbo
007335.673: Loaded overlay 'disable-wifi'
007355.160: brfs: File read: 387 bytes
007356.656: brfs: File read: /mfs/sd/cmdline.txt
007356.695: Read command line from file 'cmdline.txt':
007356.707: 'console=serial0,115200 console=tty1 root=PARTUUID=c93a34f8-02 rootfstype=ext4 fsck.repair=yes rootwait'
007479.652: kernel=
007479.671: brfs: File read: 102 bytes
008102.033: brfs: File read: /mfs/sd/kernel8.img
008102.053: Loaded 'kernel8.img' to 0x80000 size 0x7d15db
009275.990: Kernel relocated to 0x200000
009276.004: Device tree loaded to 0x2eff2c00 (size 0xd3b4)
[... 611 lines cut ...]
[    7.139602] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    7.172393] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    7.184597] cfg80211: loaded regulatory.db is malformed or signature is missing/invalid
[    7.366419] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[    7.367708] bcmgenet fd580000.ethernet eth0: Link is Down
[   11.454360] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[   11.454403] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   13.472631] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net.ipv6.neigh.eth0.retrans_time - use net.ipv6.neigh.eth0.retrans_time_ms instead
[   14.569062] Bluetooth: Core ver 2.22
[   14.569197] NET: Registered PF_BLUETOOTH protocol family
[   14.569208] Bluetooth: HCI device and connection manager initialized
[   14.569232] Bluetooth: HCI socket layer initialized
[   14.569246] Bluetooth: L2CAP socket layer initialized
[   14.569272] Bluetooth: SCO socket layer initialized
[   14.584207] Bluetooth: HCI UART driver ver 2.3
[   14.584239] Bluetooth: HCI UART protocol H4 registered
[   14.584337] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   14.584611] Bluetooth: HCI UART protocol Broadcom registered
[   14.903747] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   14.903766] Bluetooth: BNEP filters: protocol multicast
[   14.903782] Bluetooth: BNEP socket layer initialized
[   14.917334] NET: Registered PF_ALG protocol family
[   16.350348] vc4-drm gpu: [drm] Cannot find any crtc or sizes
[   31.710261] cam-dummy-reg: disabling
```

pelwell, 2022-11-25:
Thanks. You can skip the raspinfo and vclog output from now on.

```
[    6.332854] wm8804 1-003b: revision E
```
That's looking much better.
```
pi@raspberry:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
ALSA lib conf.c:1970:(_snd_config_load_with_include) _toplevel_:18:1:Unexpected char
ALSA lib conf.c:3817:(config_file_open) /home/pi/.asoundrc may be old or corrupted: consider to remove or fix it
ALSA lib conf.c:3739:(snd_config_hooks_call) function snd_config_hook_load returned error: Invalid argument
ALSA lib conf.c:4203:(snd_config_update_r) hooks failed, removing configuration
aplay: device_list:283: control open (0): Invalid argument
```
Does `rm ~/.asoundrc` help?

reinisb, 2022-11-25:
Thanks, `rm ~/.asoundrc` worked!

pi@raspberry:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpihifiberry [snd_rpi_hifiberry_digi], device 0: HiFiBerry Digi+ Pro HiFi wm8804-spdif-0 [HiFiBerry Digi+ Pro HiFi wm8804-spdif-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

Do you have ideas why ~/.asoundrc was there in the first place?

reinisb, 2022-11-25:
Though after I reboot `~/.asoundrc` is back.

pelwell, 2022-11-25:
But does it still work, or is ALSA reporting corruption again? If so, post the content of .asoundrc.

reinisb, 2022-11-25:
It's strange. After a reboot ALSA reported corruption again, but now after another reboot it does not report corruption again. Audio through hifiberry works.

pelwell, 2022-11-25:
Strange indeed. If you get another corrupted one, do upload it here.

pelwell, 2022-11-25:
Is this a regular (desktop) image? As I understand it, the LXPanel plugin is creating (or arranging the creation of) the asoundrc file. And have you enabled the beta Wayland support?

reinisb, 2022-11-25:
This is not a desktop image. This is what they call a Lite image ([ref](https://www.raspberrypi.com/software/operating-systems/)). No, I have not enabled Wayland.

[image]

syborgxx, 2022-11-25:
I have a similar problem with a Pi 4 and Hifiberry DAC+. 

My DAC is recognized when a monitor is connected to HDMI. When I do a headless boot then the DAC is not found. Plug a monitor back in and boot and it is working again.

My use is with a headless Pi so I need to find a workaround.

I did not have a file called .asoundrc.

I tried hdmi_force_hotplug=1 and that didn't help.

What other info can I provide to help fix this issue?

pelwell, 2022-11-25:
Post the output of `dmesg -l err,warn` after booting when the monitor is not attached.

syborgxx, 2022-11-25:
dmesg -l err,warn output as requested:

[    1.256753] usb_phy_generic phy: supply vcc not found, using dummy regulator
[    1.256962] usb_phy_generic phy: dummy supplies not allowed for exclusive requests
[    5.351424] systemd[1]: bpf-lsm: Failed to load BPF object: No such process
[    6.517545] systemd-gpt-auto-generator[155]: File system behind root file system is reported by btrfs to be backed by pseudo-device /dev/root, which is not a valid userspace accessible device node. Cannot determine correct backing block device.
[    6.679541] systemd[141]: /usr/lib/systemd/system-generators/systemd-gpt-auto-generator failed with exit status 1.
[    7.880346] systemd-journald[179]: File /var/log/journal/fe3a3f9a2d6a476ab61fe033c6b2b7bd/system.journal corrupted or uncleanly shut down, renaming and replacing.
[    8.957784] vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
[    9.221469] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[    9.281928] bcm2835_v4l2: module is from the staging directory, the quality is unknown, you have been warned.
[    9.381119] bcm2835_isp: module is from the staging directory, the quality is unknown, you have been warned.
[    9.410178] rpivid_hevc: module is from the staging directory, the quality is unknown, you have been warned.
[    9.432568] bcm2835_codec: module is from the staging directory, the quality is unknown, you have been warned.
[    9.457235] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.bin failed with error -2
[    9.548724] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[    9.657370] [drm] No displays found. Consider forcing hotplug if HDMI is attached
[   14.621849] systemd-journald[179]: File /var/log/journal/fe3a3f9a2d6a476ab61fe033c6b2b7bd/user-1000.journal corrupted or uncleanly shut down, renaming and replacing.

pelwell, 2022-11-25:
That looks OK. How about `i2cdetect -y 1`? And `aplay -l` just to be sure?

syborgxx, 2022-11-25:
aplay -l
aplay: device_list:274: no soundcards found...

i2cdetect -y 1
Error: Could not open file `/dev/i2c-1' or `/dev/i2c/1': No such file or directory

pelwell, 2022-11-25:
Hmm. `sudo vcdbg log msg`?

syborgxx, 2022-11-25:
I don't have vcdbg and I don't get any hits with dnf provides vcdbg. Where can I get vcdbg?

pelwell, 2022-11-25:
Follow these instructions: https://github.com/raspberrypi/linux/issues/5256#issuecomment-1327282518

syborgxx, 2022-11-25:
003890.449: arasan: arasan_emmc_open
004073.582: brfs: File read: /mfs/sd/config.txt
004074.292: brfs: File read: 1323 bytes
004096.796: HDMI0:EDID error reading EDID block 0 attempt 0
004097.811: HDMI0:EDID giving up on reading EDID block 0
004109.529: HDMI1:EDID error reading EDID block 0 attempt 0
004110.545: HDMI1:EDID giving up on reading EDID block 0
004116.475: brfs: File read: /mfs/sd/config.txt
004578.355: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
004580.868: *** Restart logging
004580.888: brfs: File read: 1323 bytes
004587.377: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
004588.395: hdmi: HDMI0:EDID giving up on reading EDID block 0
004593.437: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
004594.453: hdmi: HDMI0:EDID giving up on reading EDID block 0
004594.472: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
004599.513: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
004600.528: hdmi: HDMI1:EDID giving up on reading EDID block 0
004605.570: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
004606.588: hdmi: HDMI1:EDID giving up on reading EDID block 0
004606.603: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
004606.617: HDMI0: hdmi_pixel_encoding: 300000000
004606.631: HDMI1: hdmi_pixel_encoding: 300000000
004635.119: dtb_file 'bcm2711-rpi-4-b.dtb'
004641.969: brfs: File read: /mfs/sd/efi/bcm2711-rpi-4-b.dtb
004641.986: Loaded '/efi/bcm2711-rpi-4-b.dtb' to 0x100 size 0xcc65
004655.038: brfs: File read: 52325 bytes
004665.970: brfs: File read: /mfs/sd/efi/overlays/overlay_map.dtb
004737.620: brfs: File read: 2259 bytes
004740.433: brfs: File read: /mfs/sd/config.txt
004740.704: dtwarn: overlay 'pi3-disable-bt' has been renamed 'disable-bt'
004741.195: brfs: File read: 1323 bytes
004745.776: brfs: File read: /mfs/sd/efi/overlays/disable-bt.dtbo
004760.431: Loaded overlay 'pi3-disable-bt'
004794.277: brfs: File read: 1073 bytes
004800.403: brfs: File read: /mfs/sd/efi/overlays/gpio-fan.dtbo
004805.279: Loaded overlay 'gpio-fan'
004805.291: dtparam: temp=65000
004805.703: dtparam: watchdog=on
004837.004: brfs: File read: 1312 bytes
004844.110: brfs: File read: /mfs/sd/efi/overlays/hifiberry-dac.dtbo
004852.998: Loaded overlay 'hifiberry-dac'
004877.540: brfs: File read: 655 bytes
004894.894: brfs: File read: /mfs/sd/efi/overlays/vc4-fkms-v3d-pi4.dtbo
004912.780: Loaded overlay 'vc4-fkms-v3d'
004912.793: dtparam: audio=off
004965.363: brfs: File read: 1446 bytes
004967.778: brfs: File read: /mfs/sd/efi/cmdline.txt
004967.850: Read command line from file '/efi/cmdline.txt':
004967.869: 'coherent_pool=16M dwc_otg.lpm_enable=0 earlyprintk console=serial0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=btrfs rootflags=subvol=root ro fsck.repair=yes rootwait libahci.ignore_sss=1 raid=noautodetect nortc selinux=0 usbhid.mousepoll=0 hdmi=o
005157.590: brfs: File read: 257 bytes
006972.325: brfs: File read: /mfs/sd/efi/kernel8.img
006972.347: Loaded '/efi/kernel8.img' to 0x80000 size 0x179f200
006998.389: Kernel relocated to 0x200000
006998.400: Device tree loaded to 0x2eff2a00 (size 0xd5ab)
007003.780: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
010039.678: vchiq_core: vchiq_init_state: slot_zero = 0xde000000, is_master = 1
016863.775: camsubs: Looking for camera 0: i2c_port = 0, led gpio = -1, power enable gpio = 133
017159.448: camsubs: Camera not found
017159.519: camsubs: Looking for camera 0: i2c_port = 0, led gpio = -1, power enable gpio = 133
017160.226: camsubs: Camera not found
017160.288: camsubs: Looking for camera 0: i2c_port = 0, led gpio = -1, power enable gpio = 133
017160.994: camsubs: Camera not found
017161.055: camsubs: Looking for camera 0: i2c_port = 0, led gpio = -1, power enable gpio = 133
017161.250: camsubs: Camera not found
017161.312: camsubs: Looking for camera 0: i2c_port = 0, led gpio = -1, power enable gpio = 133
017162.019: camsubs: Camera not found

pelwell, 2022-11-25:
So the hifiberry overlay loaded OK, but there's no `/dev/i2c-1`?

What does `lsmod | grep i2c` show? And does `sudo modprobe i2c-dev` cause `/dev/i2c-1` to appear?

syborgxx, 2022-11-25:
lsmod | grep i2c
i2c_i801               40960  0
i2c_smbus              20480  1 i2c_i801

After modprobe
lsmod | grep i2c
i2c_dev                24576  0
i2c_i801               40960  0
i2c_smbus              20480  1 i2c_i801

pelwell, 2022-11-25:
Is `/dev/i2c-1` there now? If so, what does `i2cdetect -y 1` show now?

syborgxx, 2022-11-25:
i2c_dev is there, but not i2c-1.

sudo i2cdetect -y 1
Error: Could not open file `/dev/i2c-1' or `/dev/i2c/1': No such file or directory

pelwell, 2022-11-28:
Assuming everything you've told me is correct, I can't think of an explanation for the failure you are seeing. There is strong evidence that the overlay has been applied - no error messages from the firmware, the I2C and I2S pins being configured - and yet the I2C interface hasn't been configured, again with no errors.

Has the i2c-bcm2835 module been loaded? Try `sudo modprobe i2c-bcm2835`, and again look for /dev/i2c-1. If that doesn't work, I'm running out of ideas, unless it is some strange power-related problem, where your HDMI cable is back-powering the Pi enough to make the difference. Which power supply are you using? Is the cable a good one?

syborgxx, 2022-11-28:
I'm using the official raspberrypi power supply. My pi works perfectly with raspian. All other computers at home use fedora. I wanted to change the os on my pi so  all computers in my household use the same distro.

bash-5.2$ sudo modprobe i2c-bcm2835
bash-5.2$ lsmod | grep i2c
snd_soc_pcm512x_i2c    16384  1
snd_soc_pcm512x        32768  1 snd_soc_pcm512x_i2c
regmap_i2c             16384  1 snd_soc_pcm512x_i2c
i2c_bcm2835            16384  0
i2c_dev                20480  0
bash-5.2$ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- UU -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --

pelwell, 2022-11-28:
So that is the problem - module auto-loading (or at least the auto-loading of the i2c-bcm2835 module) is broken.

What do the following commands return?
```
$ grep . /etc/modprobe.d/*
$ grep i2c_bcm2835 /lib/modules/`uname -r`/modules.alias
```

pelwell, 2022-11-28:
If you get a lot of output from the first command, `grep bcm2835 /etc/modprobe.d/*` should be sufficient.

syborgxx, 2022-11-28:
grep bcm2835 /etc/modprobe.d/* gets no hits

bash-5.2$ grep i2c_bcm2835 /lib/modules/`uname -r`/modules.alias
alias platform:i2c-bcm2835 i2c_bcm2835
alias of:N*T*Cbrcm,bcm2835-i2cC* i2c_bcm2835
alias of:N*T*Cbrcm,bcm2835-i2c i2c_bcm2835
alias of:N*T*Cbrcm,bcm2711-i2cC* i2c_bcm2835
alias of:N*T*Cbrcm,bcm2711-i2c i2c_bcm2835

pelwell, 2022-11-28:
Disappointingly, those are the results that I get. I was hoping for a blacklist hit in the first grep or nothing in the second.

pelwell, 2022-11-28:
> I have a similar problem with a Pi 4 and Hifiberry DAC+.

> My pi works perfectly with raspian.

Those two statements belong together, and not this many responses apart. I feel I've narrowed the problem down for you significantly, but now it's time to ask on some Fedora forum.

closed 2022-11-28 by pelwell