raspberrypi/linux #5279: rpi4: changing default CMA reservation  with kernels 4.19.x fails to initialize vc4-drm gpu

joserpintuitivo, 2022-12-14:
### Describe the bug

I have a 4 GB RPI 4 running 32 bit kernel on Nerves Linux. The CMA reservation/allocation seems to be set to 64MB by default. If I try to pass 'cma=M' to the kernel command line, the video driver cannot be assigned from CMA.

### Steps to reproduce the behaviour

1.) Compile the 4.19.x raspberrypi kernel for 32-bit mode with Nerves
2.) Boot the kernel with dtoverlay=vc4-fkms-v3d and append in the cmdline.txt file 'cma=256M'
3.) Observe the failure to start the vc4-drm gpu driver

### Device (s)

Raspberry Pi 4 Mod. B

### System

Linux version 4.19.118 (jose@jose-ThinkPad-T430) (gcc version 9.2.0 (crosstool-NG 1.24.0.71-4fa0ba1)) #2 SMP PREEMPT Tue Dec 13 16:25:21 -03 2022

CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d

### Logs

```
cmd("dmesg | grep -i cma")
[    0.000000] cma: Failed to reserve 256 MiB
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 cma=64M cma=256M  smsc95xx.macaddr=E4:5F:01:63:91:2A vc_mem.mem_base=0xec00000 vc_mem.mem_size=0x10000000  dwc_otg.lpm_enable=0 console=tty1 console=ttyS0,115200 fbcon=scrollback:1024k root=/dev/mmcblk0p2 rootfstype=squashfs rootwait consoleblank=0
[    0.000000] Memory: 3171444K/3211264K available (6144K kernel code, 384K rwdata, 1248K rodata, 2048K init, 411K bss, 39820K reserved, 0K cma-reserved, 3080192K highmem)
[    5.441143] vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
[    5.452259] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
```

```
cmd("dmesg | grep -i cma")
[ 0.000000] cma: Failed to reserve 256 MiB
[ 0.000000] Kernel command line: coherent_pool=1M [8250.nr](http://8250.nr/)_uarts=1 cma=64M bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=E4:5F:01:63:91:2A vc_mem.mem_base=0xec00000 vc_mem.mem_size=0x10000000 dwc_otg.lpm_enable=0 console=tty1 console=ttyS0,115200 fbcon=scrollback:1024k root=/dev/mmcblk0p2 rootfstype=squashfs rootwait consoleblank=0 cma=256M
[ 0.000000] Memory: 3171444K/3211264K available (6144K kernel code, 384K rwdata, 1248K rodata, 2048K init, 411K bss, 39820K reserved, 0K cma-reserved, 3080192K highmem)
[ 5.320059] vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
[ 5.330652] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
```

timg236, 2022-12-14:
Any reason for using 4.19? The legacy LTS Buster release is 5.4

joserpintuitivo, 2022-12-14:
> Any reason for using 4.19? The legacy LTS Buster release is 5.4

Yes, the reason is because I am using Gstreamer with Omx, and it only compiles for 32bits and is deprecated for later kernel versions.

joserpintuitivo, 2022-12-14:
> > Yes, the reason is because I am using Gstreamer with Omx, and it only compiles for 32bits and is deprecated for later kernel versions.
> 
> ??! OMX is still supported on 5.15 (and 6.1), but the userspace libraries aren't built with Bullseye (they are with Buster), and aren't supported at all with a 64bit userspace.

I'm using Nerves, and until this version they used 32 bits. https://github.com/nerves-project/nerves_system_rpi4/tree/v1.13.0

That's why I'm using 4.19 and it works for me, only I have a problem with the size of the CMA, and that's why I need to increase it.

joserpintuitivo, 2022-12-14:
Ok, the solution would be to go from version 4.19 to 5.4?

timg236, 2022-12-14:
I think we’d need to know why the CMA parameter has changed. That might not be suitable for a 1GB Pi4

joserpintuitivo, 2022-12-14:
> I think we’d need to know why the CMA parameter has changed. That might not be suitable for a 1GB Pi4

it is 4GB Pi4.

pelwell, 2022-12-14:
Why aren't you using the `cma-256` parameter to the `vc4-fkms-v3d` overlay, as described in the README?

pelwell, 2022-12-14:
Later kernels switched to initialising CMA from Device Tree, but 4.19 is still command-line based. You could try using fdtget and fdtset to modify the command line in your current bcm2711-rpi-4-b.dtb, but if that doesn't work it's unlikely you'll get much support for a 4-year-old kernel.

joserpintuitivo, 2022-12-14:
> Later kernels switched to initialising CMA from Device Tree, but 4.19 is still command-line based. You could try using fdtget and fdtset to modify the command line in your current bcm2711-rpi-4-b.dtb, but if that doesn't work it's unlikely you'll get much support for a 4-year-old kernel.

Yes, that is the problem I have and I don't know how to edit the devices tree.

pelwell, 2022-12-14:
Something like this (I advise you to also make a backup copy of the original .dtb file):
```
$ cp /boot/bcm2711-rpi-4-b.dtb .
$ fdtget -t s bcm2711-rpi-4-b.dtb /chosen bootargs > bootargs
$ nano bootargs
  # Change the cma value and save the file
$ fdtput -t s bcm2711-rpi-4-b.dtb /chosen bootargs `cat bootargs`
$ fdtget -t s bcm2711-rpi-4-b.dtb /chosen bootargs
$ sudo cp bcm2711-rpi-4-b.dtb /boot/
```

joserpintuitivo, 2022-12-14:
> Something like this (I advise you to also make a backup copy of the original .dtb file):
>
> ```
> $ cp /boot/bcm2711-rpi-4-b.dtb .
> $ fdtget -t s bcm2711-rpi-4-b.dtb /chosen bootargs > bootargs
> $ nano [...]

Thanks. With the commands I was able to change the value of the CMA memory, but it seems that there is something else, because when I change it to 256M or 320M, its reservation fails.

pelwell, 2022-12-15:
What is the largest value that works?

pelwell, 2022-12-15:
And do you have any mem settings in config.txt, e.g. gpu_mem?

joserpintuitivo, 2022-12-15:
> And do you have any mem settings in config.txt, e.g. gpu_mem?

```
# Default Nerves RPi 4 config.txt
#
# It's possible to override this file by using a custom fwup.conf
# configuration to pull in a replacement.
#
# Useful links:
# http://rpf.io/configtxt
# https://www.raspberrypi.org/documentation/configuration/device-tree.md
# https://github.com/raspberrypi/documentation/blob/master/configuration/device-tree.md
# https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README

# Disable the boot rainbow
disable_splash=1

# This, along with the Raspberry Pi "x" firmware is needed for the camera
# to work. The Raspberry Pi "x" firmware is selected via the Buildroot
# configuration. See Target packages->Hardware handling->Firmware.
gpu_mem=192

# Enable I2C, SPI, and audio
dtparam=i2c_arm=on
dtparam=spi=on
dtparam=audio=on

# Enable drivers for the Raspberry Pi 7" Touchscreen
#
# This makes it possible to run Scenic out of the box with the popular
# Raspberry Pi 7" Touchscreen. It appears to be harmless for non-touchscreen
# users, but if not, let us know!
# NOTE: The touchscreen seems to work fine with these commented out on the RPi4
#dtoverlay=rpi-ft5406
#dtoverlay=rpi-backlight

# Comment this in or modify to enable OneWire
# NOTE: check that the overlay that you specify is in the boot partition or
#       this won't work.
#dtoverlay=w1-gpio-pullup,gpiopin=4

# The ramoops overlay works with the pstore driver to preserve crash
# information across reboots in DRAM
dtoverlay=ramoops

# Enable the UART (/dev/ttyS0)
enable_uart=1

# Enable DRM VC4 V3D driver on top of the dispmanx display stack
#dtoverlay=vc4-fkms-v3d
#max_framebuffers=2
```

joserpintuitivo, 2022-12-15:
> And do you have any mem settings in config.txt, e.g. gpu_mem?

I'll tell you what the maximum value is.

pelwell, 2022-12-15:
Do you need that large gpu_mem value? What happens if you remove it?

joserpintuitivo, 2022-12-15:
> Do you need that large gpu_mem value? What happens if you remove it?

If I need the GPU to be able to use the OMX to do the video encondig.

pelwell, 2022-12-15:
And what do you need the CMA for m

joserpintuitivo, 2022-12-15:
> And what do you need the CMA for?

Because as I began to record videos, these errors with the application continually appeared and my rpi restarted.

```
[ 1163.448679] 2_scheduler invoked oom-killer: gfp_mask=0x6000c0(GFP_KERNEL), nodemask=(null), order=0, oom_score_adj=0
[ 1163.459534] 2_scheduler cpuset=/ mems_allowed=0
[ 1163.464301] CPU: 2 PID: 104 Comm: 2_scheduler Tainted: G C 4.19.118 #2
[ 1163.472332] Hardware name: BCM2835
[ 1163.475831] [<c021042c>] (unwind_backtrace) from [<c020bca0>] (show_stack+0x10/0x14)
[ 1163.483768] [<c020bca0>] (show_stack) from [<c077fc98>] (dump_stack+0x94/0xb4)
[ 1163.491173] [<c077fc98>] (dump_stack) from [<c02ea174>] (dump_header+0x70/0x24c)
[ 1163.498719] [<c02ea174>] (dump_header) from [<c02e92b0>] (oom_kill_process+0xb8/0x350)
[ 1163.506828] [<c02e92b0>] (oom_kill_process) from [<c02e9e20>] (out_of_memory+0x11c/0x378)
[ 1163.515201] [<c02e9e20>] (out_of_memory) from [<c02ef534>] (__alloc_pages_nodemask+0xbe8/0x11c0)
[ 1163.524195] [<c02ef534>] (__alloc_pages_nodemask) from [<c0682760>] (skb_page_frag_refill+0xe4/0xfc)
[ 1163.533507] [<c0682760>] (skb_page_frag_refill) from [<c06841b8>] (sk_page_frag_refill+0x14/0x6c)
[ 1163.542590] [<c06841b8>] (sk_page_frag_refill) from [<c06f022c>] (tcp_sendmsg_locked+0x43c/0xc18)
[ 1163.551670] [<c06f022c>] (tcp_sendmsg_locked) from [<c06f0a30>] (tcp_sendmsg+0x28/0x3c)
[ 1163.559873] [<c06f0a30>] (tcp_sendmsg) from [<c06800c4>] (sock_sendmsg+0x14/0x24)
[ 1163.567510] [<c06800c4>] (sock_sendmsg) from [<c068016c>] (sock_write_iter+0x98/0xf4)
[ 1163.575534] [<c068016c>] (sock_write_iter) from [<c0350250>] (do_iter_readv_writev+0x118/0x194)
[ 1163.584437] [<c0350250>] (do_iter_readv_writev) from [<c0350be4>] (do_iter_write+0x84/0x190)
[ 1163.593075] [<c0350be4>] (do_iter_write) from [<c0350d68>] (vfs_writev+0x78/0xe4)
[ 1163.600708] [<c0350d68>] (vfs_writev) from [<c0350e3c>] (do_writev+0x68/0xf0)
[ 1163.608023] [<c0350e3c>] (do_writev) from [<c0201000>] (ret_fast_syscall+0x0/0x4c)
[ 1163.615741] Exception stack(0xc1851fa8 to 0xc1851ff0)
[ 1163.620903] 1fa0: 00000010 b6bc00e0 00000031 b6bc00e0 00000010 00000000
[ 1163.629274] 1fc0: 00000010 b6bc00e0 b403c920 00000092 0005ad95 b6bc00d0 a9b42a0c 00082903
[ 1163.637610] 1fe0: 00000000 b403b9b0 b403c920 b6da3218
[ 1163.643240] Mem-Info:
[ 1163.645594] active_anon:19493 inactive_anon:2 isolated_anon:0
[ 1163.645594] active_file:1477 inactive_file:9296 isolated_file:0
[ 1163.645594] unevictable:0 dirty:0 writeback:5 unstable:0
[ 1163.645594] slab_reclaimable:1130 slab_unreclaimable:3224
[ 1163.645594] mapped:2121 shmem:3 pagetables:200 bounce:0
[ 1163.645594] free:755040 free_pcp:143 free_cma:15788
[ 1163.679473] Node 0 active_anon:79156kB inactive_anon:8kB active_file:2800kB inactive_file:34808kB unevictable:0kB isolated(anon):0kB isolated(file):0kB mapped:8484kB dirty:0kB writeback:20kB shmem:12kB writeback_tmp:0kB unstable:0kB all_unreclaimable? yes
[ 1163.702629] DMA free:64628kB min:636kB low:792kB high:948kB active_anon:0kB inactive_anon:0kB active_file:112kB inactive_file:20kB unevictable:0kB writepending:20kB present:131072kB managed:93180kB mlocked:0kB kernel_stack:1000kB pagetables:0kB bounce:0kB free_pcp:0kB local_pcp:0kB free_cma:63152kB
[ 1163.729648] lowmem_reserve[]: 0 0 3008 3008
[ 1163.734061] HighMem free:2958876kB min:512kB low:19644kB high:38776kB active_anon:79040kB inactive_anon:8kB active_file:2856kB inactive_file:34612kB unevictable:0kB writepending:0kB present:3080192kB managed:3080192kB mlocked:0kB kernel_stack:0kB pagetables:800kB bounce:0kB free_pcp:1556kB local_pcp:0kB free_cma:0kB
[ 1163.762704] lowmem_reserve[]: 0 0 0 0
[ 1163.766577] DMA: 87*4kB (UEHC) 55*8kB (UEHC) 15*16kB (UEHC) 21*32kB (UEHC) 8*64kB (UEC) 5*128kB (UC) 2*256kB (C) 3*512kB (C) 1*1024kB (C) 1*2048kB (C) 14*4096kB (C) = 65316kB
[ 1163.782491] HighMem: 126*4kB (UM) 120*8kB (UM) 151*16kB (UM) 52*32kB (UM) 48*64kB (UM) 22*128kB (UM) 21*256kB (UM) 19*512kB (UM) 12*1024kB (UM) 10*2048kB (UM) 708*4096kB (UM) = 2959272kB
[ 1163.799477] 2754 total pagecache pages
[ 1163.799480] 0 pages in swap cache
[ 1163.799483] Swap cache stats: add 0, delete 0, find 0/0
[ 1163.799485] Free swap = 0kB
[ 1163.799487] Total swap = 0kB
[ 1163.799490] 802816 pages RAM
[ 1163.799492] 770048 pages HighMem/MovableOnly
[ 1163.799494] 9473 pages reserved
[ 1163.799496] 16384 pages cma reserved
[ 1163.799499] Tasks state (memory values in pages):
[ 1163.799502] [ pid ] uid tgid total_vm rss pgtables_bytes swapents oom_score_adj name
[ 1163.799516] [ 88] 0 88 418 48 28672 0 0 nbtty
[ 1163.854315] [ 95] 0 95 502 19 28672 0 0 rngd
[ 1163.854321] [ 96] 0 96 483 274 32768 0 0 nbtty
[ 1163.870991] [ 97] 0 97 66114 27965 585728 0 0 beam.smp
[ 1163.870996] [ 102] 0 102 433 255 24576 0 0 erl_child_setup
[ 1163.871001] [ 127] 0 127 399 43 20480 0 0 heart
[ 1163.871005] [ 131] 0 131 454 244 32768 0 0 uevent
[ 1163.871009] [ 134] 0 134 398 41 20480 0 0 kmsg_tailer
[ 1163.871012] [ 140] 0 140 455 51 24576 0 0 if_monitor
[ 1163.871016] [ 153] 0 153 400 45 24576 0 0 muontrap
[ 1163.871020] [ 154] 0 154 577 374 28672 0 0 ntpd
[ 1163.871025] [ 176] 0 176 400 45 24576 0 0 muontrap
[ 1163.871029] [ 177] 0 177 489 58 28672 0 0 udhcpc
[ 1163.871036] Out of memory: Kill process 97 (beam.smp) score 34 or sacrifice child
[ 1163.871049] Killed process 102 (erl_child_setup) total-vm:1732kB, anon-rss:56kB, file-rss:964kB, shmem-rss:0kB
[... 334 lines cut ...]

[... cut ...]
21:01:58.625 [info]  [    202]     0   202      489       58    32768        0             0 udhcpc

21:01:58.633 [info]  [    311]     0   311      586       19    28672        0             0 epmd

21:01:58.649 [error] Out of memory: Kill process 97 (beam.smp) score 27 or sacrifice child

21:01:58.649 [error] Killed process 101 (erl_child_setup) total-vm:1732kB, anon-rss:56kB, file-rss:1092kB, shmem-rss:0kB

21:01:58.670 [info]  oom_reaper: reaped process 101 (erl_child_setup), now anon-rss:0kB, file-rss:0kB, shmem-rss:0kB
```

pelwell, 2022-12-15:
I think it's worth trying to increase CMA as much as you can to see if it makes any difference at all. So far I'm not convinced it will.

joserpintuitivo, 2022-12-15:
> I think it's worth trying to increase CMA as much as you can to see if it makes any difference at all. So far I'm not convinced it will.

In the rpi3 with the same Kernel version, I was able to increase the CMA memory without problem and these errors stopped appearing. But really I'm going to use an rpi4, I'm going to have to migrate to kernel 5.4. Thanks for your help.

6by9, 2022-12-15:
> Yes, I need the GPU to be able to use the OMX to do the video encoding.

The video encoder needs the new and 1 reference frame, and is max 1920x1080. User buffers (as configured via IL's nBufferCount on the port format) are on top of that. It also has a 4MB output FIFO.
Each video encode at 1080p is therefore taking possibly 20MB of gpu_mem.

The camera does require gpu_mem, but the default with startx=1 of 128M is usually sufficient unless asking for multiple 5/8/12MPix 24bpp or 32bpp RGB images.

Just noting from your first post
```[    0.000000] Memory: 3171444K/3211264K available ```
That makes it look like something has stolen almost 1GB of the RAM (3136MB left, so 960MB taken).
My CM4 with gpu_mem=76 reports
```[    0.000000] Memory: 3599276K/4050944K available (12288K kernel code, 1386K rwdata, 3268K rodata, 2048K init, 591K bss, 123988K reserved, 327680K cma-reserved, 3264512K highmem)```
4050944 = 3956MB.