raspberrypi/linux #5280: CM4 I2C I2C_M_IGNORE_NAK Flag

limitbit, 2022-12-14:
### Describe the bug

Hi  i am working on CM4 module and try to communicate with an i2c device, i try use I2C_M_IGNORE_NAK flag in my message but it do not respond to flag. I mention that in rpi forum and than i get an answer from an rpi engineer "It does look like the I2C_M_IGNORE_NAK flag hasn't been implemented" __

### Steps to reproduce the behaviour

Send an i2c message via kernel driver with I2C_M_IGNORE_NAK to test it.

### Device (s)

Raspberry Pi CM4

### System

Raspberry Pi reference 2022-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 005a8c73b05a2cab394073150208bf4f069e861a, stage2

version 102f1e848393c2112206fadffaaf86db04e98326 (clean) (release) (start)

Linux doly16gb 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux

6by9, 2022-12-14:
Forum link - https://forums.raspberrypi.com/viewtopic.php?t=343808

6by9, 2023-04-12:
I think the required diff is
```
diff --git a/drivers/i2c/busses/i2c-bcm2835.c b/drivers/i2c/busses/i2c-bcm2835.c
index 7408e2c136b0..45602e1acd36 100644
--- a/drivers/i2c/busses/i2c-bcm2835.c
+++ b/drivers/i2c/busses/i2c-bcm2835.c
@@ -384,7 +384,9 @@ static irqreturn_t bcm2835_i2c_isr(int this_irq, void *data)
        val = bcm2835_i2c_readl(i2c_dev, BCM2835_I2C_S);
        bcm2835_debug_add(i2c_dev, val);
 
-       err = val & (BCM2835_I2C_S_CLKT | BCM2835_I2C_S_ERR);
+       err = val & BCM2835_I2C_S_CLKT;
+       if (!(i2c_dev->curr_msg->flags & I2C_M_IGNORE_NAK))
+               err |=  val & BCM2835_I2C_S_ERR;
        if (err) {
                i2c_dev->msg_err = err;
                goto complete;
```
However having just been looking at https://forums.raspberrypi.com/viewtopic.php?t=350180, it looks like the error isn't reported until the last message of the transaction. A multi-message transaction with I2C_M_IGNORE_NAK set on one of the earlier messages is still going to report failure. I'm not sure if that can be resolved or not.

6by9, 2023-05-15:
Having just looked at this flag for another issue, it is optional, and availability is signaled to userspace via the I2C_FUNC_PROTOCOL_MANGLING flag - https://www.kernel.org/doc/html/latest/i2c/functionality.html

<html><body>

I2C_FUNC_PROTOCOL_MANGLING | Knows about the I2C_M_IGNORE_NAK, I2C_M_REV_DIR_ADDR and I2C_M_NO_RD_ACK flags (which modify the I2C protocol!)
-- | --

</body>
</html>

The i2c-bcm2835 driver doesn't set I2C_FUNC_PROTOCOL_MANGLING, therefore it is technically user error in trying to set it.

6by9, 2023-10-31:
Closing as implemented via https://github.com/raspberrypi/linux/commit/8c4f88c30fef01baa9436b4478548db6fe20b09e

closed 2023-10-31 by 6by9