raspberrypi/linux #5289: CM4: Ethernet LED0 (Speed/Activity) not controllable

agners, 2022-12-20:
### Describe the bug

On the Compute Module the Yellow/Amber LED is connected through CM4 pin 15 (Ethernet_nLED3, Active-low Ethernet activity indicator). It seems that this LED should be controllable through `eth_led0` device tree override. However, it seems not to react to any changes to that override.

The green `eth_led1` seems to be controllable, a value of 4 disables the LED as expected.

### Steps to reproduce the behaviour

1. Install latest Raspberry Pi OS on a CM4 (installed on a CM4 IO board)
2. Use the following setting in `config.txt`

```
dtparam=eth_led0=4
dtparam=eth_led1=4
```

3. Observe the amber Yellow still blinking on Ethernet activity.

### Device (s)

Raspberry Pi CM4 Lite

### System

```
$ cat /etc/rpi-issue
Raspberry Pi reference 2022-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 8a42abcd1dbd8c9c1fdfca4e0c3778255b2f9cc4, stage4
```

```
$ vcgencmd version
Oct 26 2022 11:09:05 
Copyright (c) 2012 Broadcom
version c72ad6b26ff40c91ef776b847436094ee63fabee (clean) (release) (start)
```

```
$ uname -a
Linux raspberrypi 5.15.76-v7l+ #1597 SMP Fri Nov 4 12:14:58 GMT 2022 armv7l GNU/Linux
```

pelwell, 2022-12-29:
As you say, the CM4 dts supports the `eth_led0` parameter, in the same way that Pi 4 and Pi 400 do. If there is a difference in behaviour it might be down to the use of different PHYs. I can investigate when the office reopens next week.

pelwell, 2023-01-03:
My findings agree with yours. This is due to the CM4 using a different combination of ETH LED pins than Pi 4.

Pi 4's ETH_LEDG is driven by the PHY's LED2 output, and ETH_LEDY is driven by LED1. 
CM4's ETH_LEDG is also driven by LED2, but ETH_LEDY is driven by the PHY's LED3 output.
The additional two LEDs don't exist on the Pi 4's PHY, but the two PHYs share the same ID...

I have a patch that makes both CM4 ETH LEDs controllable using the `eth_led<n>` parameters, but it breaks Pi 4. This will take more effort.

pelwell, 2023-01-03:
It turned out to be easy after all (Pi 4 failure was a problem with the testing). See https://github.com/raspberrypi/linux/pull/5300 for my suggested low-impact fix.

agners, 2023-01-24:
Sorry for the delay. Just upgraded to latest kernel and I can confirm that with that release the above `dtparm` allow to disable the Ethernet LEDs on CM4 as well. Thanks for addressing this!

closed 2023-01-24 by agners