raspberrypi/linux #5291: This is a bug report from the future.

AgentCasey, 2022-12-22:
### Describe the bug

I've Titled this as such, since currently in the Raspberry Pi OS distribution, we are currently running kernel 5.15.76-v8+. I've been testing the newer kernels all the way up to the current stable, kernel 6.1.1. Yesterday, I tested kernel 6.0.14 and all the previous kernels in that series worked as predictably as one would expect. However, I noticed something out of the ordinary, while testing kernel 6.1.1 and some of the rc kernels of that release. I've noticed that the file, /sys/class/thermal/thermal_zone0/temp is missing it no longer exists with these 6.1 kernel versions. 

To be precise, /sys/class/thermal does exist, however, the files, thermal_zone0/temp do not, any longer. Obviously, I'll be taking this report to the Linux Kernel team to see if they have any insights as to why this is, but I thought I'd bring this observation here as well.  Kernel 6.1.1 runs very well on Raspberry Pi OS and I wouldn't have even noticed the missing files had I not written a script to report the temperatures of my Raspberry Pi's.  My script depended upon the files located withing /sys/class/thermal/thermal_zone0/temp and I was alerted to the differences between all the other kernels moving towards the 6.1 series and presumably, all the others thereafter. 

Now, measuring the temperature of the Raspberry Pi can also be easily done using vcgencmd measure_temp, which still works with the new kernel 6.1.1 even with the aforementioned missing files. I've checked the configuration files of 5.15.76-v8+, 6.0.14 and 6.1.1 and except for minor differences between progressive updates, the configurations are essentially comparable. 

I'm bringing this observation to the parties concerned, since it is an issue, that, if not already met, will be so in future Linux kernel experimentation. I'm merely, trying to find the answers as to why this change has just only now occurred and my question is, is it something that has occurred by accident, or is it a deliberate act by all the developers involved, shaping a new path. 

Let me say, that my original current system kernel, version 5.15.76-v8+, is still intact and I can and do shift between kernels as I continue my experiments and discoveries.  I'm merely wondering if anyone else who like hacking and working with kernels have noticed this revelation and I'd like to hear the experiences. 

### Steps to reproduce the behaviour

In order to reproduce the behavior/behaviour, one would have to  first build the kernel version 6.1.1 obviously, it requires some steps to do so if the kernel involved, is to work on Raspberry Pi's. I have been using a rather unorthodox method to build my kernels.  I'm sure there are many ways to do so, and I'd love to hear as many of them as there are. However, the method I use, although, I think the methods may not be relevant, is  to mimic the process used by the Arch Linux Arm, Manjaro Linux Arm and Gentoo Linux Arm teams. I simply modify their scripts to suit my own ends build the kernels and install them in such a way as to not be intrusive on my Raspberry Pi OS system. I then observe the kernel and test for stability, instability or anything extraordinary.  

If the method used to build the kernel is deemed relevant, I will of course, gladly revel the process used, but I think that many people here may already know several ways to do so.  I think my main points of concern, have already been listed in the paragraphs above, and I simply wished to pass on my observations, hear yours and gain some insight which will also allow me to articulate, what I'm observing.  I hope I'll get some replies soon. I do like puzzles, so I'll be sorting this one out, until I have some answers. That is to say, even now, I'm still building and rebuilding this most recent kernel, on different Raspberry Pi's, running on different distributions and observing the results. 

From what, relatively, small  data I've amassed thus far, I'm willing to conclude, that the changes observed, has it's sources, originating from the Linux Kernel development teams and while the changes I've observed are only affecting just a few missing files that once existed, my concern is why did this occur and how many other changes are happening that are still yet unknown.  Thanks for your time and I'll be paying attention to this board, in hopes of some replies and answers. Thank you. 

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspberry Pi reference 2022-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 005a8c73b05a2cab394073150208bf4f069e861a, stage4

Oct 26 2022 11:09:05 
Copyright (c) 2012 Broadcom
version c72ad6b26ff40c91ef776b847436094ee63fabee (clean) (release) (start)

Linux dracosnovem 6.1.1-v8 #1 SMP PREEMPT Thu Dec 22 16:51:09 UTC 2022 aarch64 GNU/Linux

### Logs

[dec22 13:16] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[  +0,000000] Linux version 6.1.1-v8 (markie@freedombox) (gcc (Debian 10.2.1-6) 10.2.1 20210110, GNU ld (GNU Binutils for Debian) 2.35.2) #1 SMP PREEMPT Thu Dec 22 16:51:09 UTC 2022
[  +0,000000] random: crng init done
[  +0,000000] Machine model: Raspberry Pi 4 Model B Rev 1.4
[  +0,000000] efi: UEFI not found.
[  +0,000000] Reserved memory: created CMA memory pool at 0x000000000fc00000, size 256 MiB
[  +0,000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[  +0,000000] Zone ranges:
[  +0,000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[  +0,000000]   DMA32    [mem 0x0000000040000000-0x00000000ffffffff]
[  +0,000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[  +0,000000] Movable zone start for each node
[  +0,000000] Early memory node ranges
[  +0,000000]   node   0: [mem 0x0000000000000000-0x000000001fefffff]
[  +0,000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[  +0,000000]   node   0: [mem 0x0000000100000000-0x00000001ffffffff]
[  +0,000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[  +0,000000] On node 0, zone DMA32: 256 pages in unavailable ranges
[  +0,000000] On node 0, zone Normal: 16384 pages in unavailable ranges
[  +0,000000] percpu: Embedded 29 pages/cpu s78504 r8192 d32088 u118784
[  +0,000000] pcpu-alloc: s78504 r8192 d32088 u118784 alloc=29*4096
[  +0,000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[  +0,000000] Detected PIPT I-cache on CPU0
[  +0,000000] CPU features: detected: Spectre-v2
[  +0,000000] CPU features: detected: Spectre-v3a
[  +0,000000] CPU features: detected: Spectre-v4
[  +0,000000] CPU features: detected: Spectre-BHB
[  +0,000000] CPU features: kernel page table isolation forced ON by KASLR
[  +0,000000] CPU features: detected: Kernel page table isolation (KPTI)
[  +0,000000] CPU features: detected: ARM erratum 1742098
[  +0,000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[  +0,000000] alternatives: applying boot alternatives
[  +0,000000] Built 1 zonelists, mobility grouping on.  Total pages: 1918724
[... 482 lines cut ...]
### Additional context

This is a script that reports the temperature of the Raspberry Pi, similar to vcgencmd measure_temp. It's the script that I ran and led me to discover that /sys/class/thermal/thermal_zone0/temp, no longer existed in the same way that it does under earlier kernel versions. After running this script under 6.1.1 and later kernels the result will yield this result:
cat: /sys/class/thermal/thermal_zone0/temp: Filen eller katalogen finns inte
CPU Temperature:

(standard_in) 1: syntax error
°C

(standard_in) 1: syntax error
(standard_in) 1: syntax error
°F

#!/bin/sh

milcel=$(cat /sys/class/thermal/thermal_zone0/temp)

echo "CPU Temperature:"
echo
ctemp=$(echo "scale=3;$milcel/1000" | bc)
echo $ctemp "°C"
echo
ftemp=$(echo "scale=3;(($milcel/1000)*(9/5))+32" | bc)
echo $ftemp "°F"
echo

pelwell, 2022-12-22:
Some of all of these error messages are relevant:
```
[ +0,001268] thermal_sys: Failed to find 'trips' node
[ +0,000086] thermal_sys: Failed to find trip points for thermal id=0
[ +0,000076] bcm2711_thermal fd5d2000.avs-monitor:thermal: error -EINVAL: could not register sensor: -22
[ +0,000096] bcm2711_thermal: probe of fd5d2000.avs-monitor:thermal failed with error -22
```

AgentCasey, 2022-12-22:
Thank you pelwell, I don't know how I overlooked that.  I'll start the fix now.

AgentCasey, 2022-12-23:
I have resolved this issue.

closed 2022-12-23 by AgentCasey

wernerb, 2023-01-01:
@AgentCasey I have the same problem on rpi4 on 6.1. Can you say how you fixed the sensor?

AgentCasey, 2023-01-02:
@wernerb I fixed everything by doing this one simple move. When I made the kernel I used the command make -j4 bindeb-pkg. So long story short I copied the file /boot/bcm2711-rpi-4-b.dtb /boot/bcm2711-rpi-4-b.dtb.orig since I plan on using the old kernel if anything goes wrong, and I replaced the original /boot/bcm2711-rpi-4-b.dtb with the file with the same name, from the new kernel packages I built. That apparently was all I had to do to resolve everything.  I never realized how important those .dtb files are, I think they're referred to as device tree blob files. Anyway that's what I neglected to do and that's why I kept getting that message. So see if you can work it out in a similar manner and good luck. Thanks.