raspberrypi/linux #5292: LCD display (SPI) not working with 6.1 kernel

jolla, 2022-12-27:
### Describe the bug

Using 6.1 kernel with Waveshare LCD display there is no output to display, display powers on but remains completely white. Switching to 5.15-5.19 kernel the display works fine. Display is using SPI interface.

Display: https://www.waveshare.com/1.44inch-lcd-hat.htm

### Steps to reproduce the behaviour

Boot RPi using 6.1 kernel with Waveshare 1.44" LCD display, no output to display

### Device (s)

Raspberry Pi 4 Mod. B, Raspberry Pi CM4

### System

```
$ cat /etc/rpi-issue
Raspberry Pi reference 2022-12-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 9ecb50a577fa3772440cd37b7dbff5f327a6191f, wlanpi1
```

```
$ vcgencmd version
Oct 26 2022 11:09:36 
Copyright (c) 2012 Broadcom
version c72ad6b26ff40c91ef776b847436094ee63fabee (clean) (release) (start_db)
```

```
$ uname -a
Linux wlanpi-f06 6.1.1-v8-wlanpi+ #1 SMP PREEMPT Mon Dec 26 12:56:36 UTC 2022 aarch64 GNU/Linux
```

### Logs

```
$ dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.1.1-v8-wlanpi+ (runner@fv-az434-471) (aarch64-linux-gnu-gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #1 SMP PREEMPT Mon Dec 26 12:56:36 UTC 2022
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi Compute Module 4 Rev 1.1
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ac00000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA32: 19456 pages in unavailable ranges
[    0.000000] On node 0, zone Normal: 16384 pages in unavailable ranges
[    0.000000] percpu: Embedded 29 pages/cpu s78440 r8192 d32152 u118784
[    0.000000] pcpu-alloc: s78440 r8192 d32152 u118784 alloc=29*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v3a
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 1742098
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[... 377 lines cut ...]
[    9.392392] mt7921e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20221109110918a

[    9.421671] 8021q: 802.1Q VLAN Support v1.8
[    9.668418] mt7921e 0000:01:00.0: WM Firmware Version: ____010000, Build Time: 20221109111005
[    9.694735] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[   11.114301] systemd-journald[148]: File /var/log/journal/37992619574c471cb6afc2e30061a3e8/user-1000.journal corrupted or uncleanly shut down, renaming and replacing.
[   11.750628] Bluetooth: hci0: Device setup in 3253410 usecs
[   11.750653] Bluetooth: hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[   12.160912] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   12.160935] Bluetooth: BNEP filters: protocol multicast
[   12.160952] Bluetooth: BNEP socket layer initialized
[   12.170025] Bluetooth: MGMT ver 1.22
[   12.190277] NET: Registered PF_ALG protocol family
[   13.087128] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[   13.087168] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
```

### Additional context

SPI interfaces are present

```
$ ls /dev/spi*
/dev/spidev0.0  /dev/spidev0.1
```

6by9, 2022-12-27:
Seeing as this is a 3rd party product which is using userspace to generate the required SPI commands, then unless SPI is genuinely broken in 6.1, it isn't a kernel issue.

Exactly which bits of the [Waveshare wiki pages](https://www.waveshare.com/wiki/1.44inch_LCD_HAT) have you followed to configure the panel?

WiringPi is deprecated, so continuing to require it is going to have issues (I didn't think it even supported the CM4)
AFAIK FBCP won't play nicely with vc4-kms-v3d, so that part is also going to problematic.

jolla, 2022-12-28:
Good news, after upgrading pip from pip-22.3 to pip-22.3.1 the OLED display is now working with 6.x kernels.

FWIW we are using [luma.oled](https://luma-oled.readthedocs.io/en/latest/intro.html) to drive the display.

jolla, 2023-01-25:
Unfortunately the issue is still present, however it is intermittent, often after a reboot or power cycle the display backlight will illuminate but no output is displayed. No issues if reverting back to a 5.x kernel but same issues is present on all 6.x kernels, most recently tested with 6.1.8 and it is still present.

jolla, 2023-01-25:
I was able to finally resolve this by changing the interface type from `spi` to `gpio_cs_spi` and then defining the chip select GPIO using the `--gpio-chip-select` parameter.

closed 2023-01-25 by jolla