raspberrypi/linux #5295: Compiling the kernel with LLVM=1 fails for ARCH=arm CROSS_COMPILE=armv6l-unknown-linux-gnueabihf- kernels

eliasnaur, 2022-12-30:
### Describe the bug

The issue seems to be that `https://github.com/raspberrypi/linux/blob/rpi-5.15.y/arch/arm/lib/memcmp_rpi.S` (and perhaps others) are not compatible with the clang assembler.

### Steps to reproduce the behaviour

On a nix system with flakes enabled,

```sh
$ nix build github:eliasnaur/rpikernel-clang-issue
error: builder for '/nix/store/3zc9nww3crck85wir5fs78zfyfv1nm5q-Raspberry-Pi-Linux-kernel.drv' failed with exit code 2;
last 10 log lines:
>         ldrmib DAT5, [S_2]
>         ^
> <instantiation>:11:9: note: while in macro instantiation
>         memcmp_trailing_15bytes unaligned
>         ^
> arch/arm/lib/memcmp_rpi.S:265:6: note: while in macro instantiation
> 140: memcmp_short_inner_loop 1
>      ^
> make[1]: *** [scripts/Makefile.build:391: arch/arm/lib/memcmp_rpi.o] Error 1
> make: *** [Makefile:1900: arch/arm/lib] Error 2
For full logs, run 'nix log /nix/store/3zc9nww3crck85wir5fs78zfyfv1nm5q-Raspberry-Pi-Linux-kernel.drv'.
```

The relevant configuration and building scripts are in the [repository](https://github.com/eliasnaur/rpikernel-clang-issue/blob/d8ae540103c27c52023c306b61c3c5af4a87d851/flake.nix#L24). I'm using [clang 14](https://github.com/eliasnaur/rpikernel-clang-issue/blob/d8ae540103c27c52023c306b61c3c5af4a87d851/flake.nix#L28).

### Device (s)

Raspberry Pi Zero, Raspberry Pi Zero W / WH

### System

The issue is during building of custom kernel, not at runtime.

### Additional context

I'm building a custom kernel with LLVM because I have problems producing deterministic output from a gcc build.

The issue can be worked around by disabling `BCM2835_FAST_MEMCPY` in the kernel config.

pelwell, 2023-01-03:
Are there no further error messages? The ones you include are showing where the problem is but not what.

eliasnaur, 2023-01-03:
Here's the full `nix log` output:

[log.txt](https://github.com/raspberrypi/linux/files/10339023/log.txt)

mrturcot, 2023-01-04:
Not sure if makes any difference but for my clang kernel to compile correctly I needed to run make like

`make CC=clang LD=ld.lld NM=llvm-nm AR=llvm-ar LLVM_IAS=1 -jX`

eliasnaur, 2023-01-04:
Those arguments give me the same result, and indeed `LLVM=1` should effectively be [equivalent](https://docs.kernel.org/kbuild/llvm.html#llvm-utilities). I'm guessing you're compiling your kernel for arm64? I believe `BCM2835_FAST_MEMCPY` is only relevant for 32-bit kernels.

iucoen, 2023-03-12:
Try LLVM_IAS=0

clang doesn't like the fast_memcpy assembly files for some reason.

altimeter-130ft, 2023-11-18:
Some of the RPi-specific assembly sources in arch/arm/lib/ use the undocumented syntax accepted by GCC / GNU Binutils but not by LLVM / clang.

I have fixed those sources into the styles acceptable by both LLVM and GCC.  The builds by both LLVM ("make LLVM=1" without LLVM_IAS) and GCC have been confirmed to work.