raspberrypi/linux #5297: RPi4/armv7h panic on boot with dtoverlay=vc4-kms-v3d,cma-512

graysky2, 2023-01-01:
### Describe the bug

Booting with the following in `/boot/config.txt` causes a kernel panic.  Unfortunately, systemd does not initialize journalctl so I cannot log the storm of text that zooms by.
```
# Disable compensation for displays with overscan
display_auto_detect=1

over_voltage=6
arm_freq=2000

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]
dtoverlay=vc4-kms-v3d,cma-512
dtoverlay=
disable_overscan=1
disable_fw_kms_setup=1
```

Note that if I boot an aarch64 with the identical `/boot/config.txt` that causes the panic on armv7h, the bug does not occur/system boots as expected.

If I omit the `,cma-512` from the line, it boots as expected.

### Steps to reproduce the behaviour

Boot Arch Linux ARM armv7h image, update, add the lines above to `/boot/config.txt`

### Device (s)

Raspberry Pi 4 Mod. B

### System

* OS is Arch Linux ARM/armv7h
* `% vcgencmd version`
```Dec 12 2022 11:56:56 
Copyright (c) 2012 Broadcom
version ed6f6b8fcdc6476410b9cf75d141633461d34bdd (clean) (release) (start)
```
* `Linux workbench7 5.15.84-1-rpi-ARCH #1 SMP Mon Dec 19 13:37:50 MST 2022 armv7l GNU/Linux`

### Logs

I can try to attached a picture or movie of the boot output, but as stated above, no record of the boot is in journalctl's logs.

P33M, 2023-01-02:
>over_voltage=6
>arm_freq=2000

Overclocks are not guaranteed. Remove these two lines - do you get the same behaviour?

popcornmix, 2023-01-02:
>  Unfortunately, systemd does not initialize journalctl so I cannot log the storm of text that zooms by.

Recent kernel with ramoops overlay may help grab a panic after a reboot: 
https://forums.raspberrypi.com/viewtopic.php?t=199047

But as @P33M says, test with the overclock disabled first.

graysky2, 2023-01-02:
I removed the two overclocking lines but the problem is still present.  I will look into that ramoops overlay.

graysky2, 2023-01-02:
I tried but there isn't any data in the expected location if I am reading the thread you linked correctly, @popcornmix 

```
# zgrep PSTORE /proc/config.gz
CONFIG_PSTORE=y
CONFIG_PSTORE_DEFAULT_KMSG_BYTES=10240
CONFIG_PSTORE_DEFLATE_COMPRESS=y
# CONFIG_PSTORE_LZO_COMPRESS is not set
# CONFIG_PSTORE_LZ4_COMPRESS is not set
# CONFIG_PSTORE_LZ4HC_COMPRESS is not set
# CONFIG_PSTORE_842_COMPRESS is not set
# CONFIG_PSTORE_ZSTD_COMPRESS is not set
CONFIG_PSTORE_COMPRESS=y
CONFIG_PSTORE_DEFLATE_COMPRESS_DEFAULT=y
CONFIG_PSTORE_COMPRESS_DEFAULT="deflate"
CONFIG_PSTORE_CONSOLE=y
# CONFIG_PSTORE_PMSG is not set
# CONFIG_PSTORE_FTRACE is not set
CONFIG_PSTORE_RAM=y
# CONFIG_PSTORE_BLK is not set
```

```
# dmesg | egrep 'pstore|ramoops'
egrep: warning: egrep is obsolescent; using grep -E
[    0.038430] pstore: Registered ramoops as persistent store backend
[    0.038456] ramoops: using 0x10000@0xb000000, ecc: 0
[    2.116145] pstore: Using crash dump compression: deflate
```

```
# grep "" /sys/module/ramoops/parameters/*
/sys/module/ramoops/parameters/console_size:0
/sys/module/ramoops/parameters/dump_oops:-1
/sys/module/ramoops/parameters/ecc:0
/sys/module/ramoops/parameters/ftrace_size:0
/sys/module/ramoops/parameters/max_reason:2
/sys/module/ramoops/parameters/mem_address:184549376
/sys/module/ramoops/parameters/mem_size:65536
/sys/module/ramoops/parameters/mem_type:0
/sys/module/ramoops/parameters/pmsg_size:0
/sys/module/ramoops/parameters/record_size:16384
```

Booting with the dtoverlay enabled however does not yield a capture:
```
# tree /sys/fs/pstore
/sys/fs/pstore

0 directories, 0 files
```

```
# cat /boot/config.txt
 Disable compensation for displays with overscan
display_auto_detect=1

#over_voltage=6
#arm_freq=2000

dtoverlay=ramoops

# Uncomment to enable bluetooth
#dtparam=krnbt=on

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]
include kodi.config.txt
```

graysky2, 2023-01-02:
@popcornmix - I must not be enabling this correctly/when I trigger a panic, I do not see any data.  What am I doing wrong?
```
# zgrep PSTORE /proc/config.gz
CONFIG_PSTORE=y
CONFIG_PSTORE_DEFAULT_KMSG_BYTES=10240
CONFIG_PSTORE_DEFLATE_COMPRESS=y
# CONFIG_PSTORE_LZO_COMPRESS is not set
# CONFIG_PSTORE_LZ4_COMPRESS is not set
# CONFIG_PSTORE_LZ4HC_COMPRESS is not set
# CONFIG_PSTORE_842_COMPRESS is not set
# CONFIG_PSTORE_ZSTD_COMPRESS is not set
CONFIG_PSTORE_COMPRESS=y
CONFIG_PSTORE_DEFLATE_COMPRESS_DEFAULT=y
CONFIG_PSTORE_COMPRESS_DEFAULT="deflate"
CONFIG_PSTORE_CONSOLE=y
# CONFIG_PSTORE_PMSG is not set
# CONFIG_PSTORE_FTRACE is not set
CONFIG_PSTORE_RAM=y
# CONFIG_PSTORE_BLK is not set

# dmesg | egrep 'pstore|ramoops'
egrep: warning: egrep is obsolescent; using grep -E
[    0.038430] pstore: Registered ramoops as persistent store backend
[    0.038456] ramoops: using 0x10000@0xb000000, ecc: 0
[    2.116145] pstore: Using crash dump compression: deflate

# grep "" /sys/module/ramoops/parameters/*
/sys/module/ramoops/parameters/console_size:0
/sys/module/ramoops/parameters/dump_oops:-1
/sys/module/ramoops/parameters/ecc:0
/sys/module/ramoops/parameters/ftrace_size:0
/sys/module/ramoops/parameters/max_reason:2
/sys/module/ramoops/parameters/mem_address:184549376
/sys/module/ramoops/parameters/mem_size:65536
/sys/module/ramoops/parameters/mem_type:0
/sys/module/ramoops/parameters/pmsg_size:0
/sys/module/ramoops/parameters/record_size:16384
```

I should note that I compiled `ramoops.dtbo` as described in the thread, but I had `ramoops-pi4.dtbo` already installed:
```
# ls -lh /boot/overlays/ram*
-rwxr-xr-x 1 root root 757 Jan  2 15:16 /boot/overlays/ramoops.dtbo
-rwxr-xr-x 1 root root 741 Jan  1 15:01 /boot/overlays/ramoops-pi4.dtbo
```

And it is enabled:
```
# grep ramoo /boot/config.txt 
dtoverlay=ramoops
```

Yet, I have no data:
```
# tree /var/lib/systemd/pstore/
/var/lib/systemd/pstore/

0 directories, 0 files

# tree /sys/fs/pstore          
/sys/fs/pstore

0 directories, 0 files
```

graysky2, 2023-01-03:
I took the uSD card and successfully booted with the offending line in `/boot/config.txt` on another RPi4 B (2 GB RAM version) whereas the one affected by this bug is an 8 GB version.  I think that's either pointing to something with the larger RAM or something specific to my specimen.

graysky2, 2023-01-04:
In trying to run `memtester` it seems that it is limited to 2074 MB on armv7h?
```
% sudo memtester 7G  1000  
memtester version 4.5.1 (32-bit)
Copyright (C) 2001-2020 Charles Cazabon.
Licensed under the GNU General Public License version 2 (only).

pagesize is 4096
pagesizemask is 0xfffff000
want 3072MB (3221225472 bytes)
got  2047MB (2147479552 bytes), trying mlock ...locked.
Loop 1/1000:
  Stuck Address       : testing  15
...
```

There is plenty detected.

```
% free -m
               total        used        free      shared  buff/cache   available
Mem:            7892         159        7605           0         127        7160
Swap:              0           0           0
```

popcornmix, 2023-01-04:
Try running memtester with sudo.
"Run memtester as root so that it can mlock the memory it tests."
https://linux.die.net/man/8/memtester

I think I've found it can grab larger amounts of ram.

graysky2, 2023-01-04:
I get the same result when invoking with sudo vs running as root:
```
% sudo memtester 6G     
memtester version 4.5.1 (32-bit)
Copyright (C) 2001-2020 Charles Cazabon.
Licensed under the GNU General Public License version 2 (only).

pagesize is 4096
pagesizemask is 0xfffff000
want 2048MB (2147483648 bytes)
got  2047MB (2147479552 bytes), trying mlock ...locked.
Loop 1:
  Stuck Address       : setting   0^C
```

And running as root:
```
# memtester 6G  
memtester version 4.5.1 (32-bit)
Copyright (C) 2001-2020 Charles Cazabon.
Licensed under the GNU General Public License version 2 (only).

pagesize is 4096
pagesizemask is 0xfffff000
want 2048MB (2147483648 bytes)
got  2047MB (2147479552 bytes), trying mlock ...^C
```

pelwell, 2023-01-04:
Slightly off-topic, but can I recommend you get hold of a serial cable of some description - either one which is USB-based such as an FTDI, or simply three patch cables to another Pi - so that you can capture the kernel output in circumstances like this. Using the `earlycon=...` command line parameter means you can get output from very early in the boot process.

graysky2, 2023-01-04:
I have one but this particular RPi4 is enclosed in an Akasa GemPro case which does not offer access to the headers.  It is great for thermal control since the case itself is a beefy heat sink.  The down side is that I risk cracking surface components on the RPi4 due to thermal paste that is really sticky.  I would also have to replace all the thermal pads within it if I crack the case.  I don't want to risk damaging it given the lack of of RPi4 right now.

pelwell, 2023-01-04:
Understood. For the same reasons, the suggestion of using another off-the-shelf Pi 4 is probably also out of the question.

graysky2, 2023-01-04:
I would love to have another 8 GB model but they are going for $200+ now on Amazon here in the US.

pelwell, 2023-01-04:
I was just about to suggest what you've just added in your EDIT. I can give it a try tomorrow.

graysky2, 2023-01-04:
I will create an image for you.

> I took the uSD card and successfully booted with the offending line in /boot/config.txt on another RPi4 B (2 GB RAM version) whereas the one affected by this bug is an 8 GB version. 

I also just confirmed the the 4 GB model boots fine.

graysky2, 2023-01-04:
Tarball is here.  Just follow the standard ArchARM install instructions using my tarball instead of the official one:
https://archlinuxarm.org/platforms/armv8/broadcom/raspberry-pi-4

I will send you a private email with a link to the image.

graysky2, 2023-01-05:
@pelwell - I emailed your raspberrypi.org address.  Did you receive the link?

JamesH65, 2023-01-05:
raspberrypi.com not .org.  Although .org might still work.

pelwell, 2023-01-05:
I have something that is showing signs of life, but I had to locate a firmware, and it's lacking an initramfs file - which causes it to panic.

Can you send me a suitable initramfs file?

graysky2, 2023-01-05:
That is interesting... I disabled generation of the initramfs files because I can boot without them.  In any case, you can generate a new one by running the following as root:
```
mkinitcpio -p linux-rpi
```

pelwell, 2023-01-05:
But I can't log in...

pelwell, 2023-01-05:
Because it won't boot because it can't find an init. Is that in the initramfs?

graysky2, 2023-01-05:
That is odd ... boots for me without the initramfs file.  Tested the exact image.  CHeck you email in a few min.  WOrking on it now.

graysky2, 2023-01-05:
Email sent

pelwell, 2023-01-05:
But it doesn't seem to contain any executables.

pelwell, 2023-01-05:
That file wasn't there 5 minutes ago.

graysky2, 2023-01-05:
Problem with my `.htaccess` which is now corrected

graysky2, 2023-01-05:
I got a mail bounce/proton.me does not like me attaching the initramfs ... I upload it to the same space on my server along with a missing config file.

The conf file should go into `/etc/` and it is needed for you to generate your own initramfs.img files on kernel updates.  Not sure why it too was omitted.  

The `initramfs-linux.img` goes into `/boot/` as you know and you will need to make an edit to `/boot/config.txt` adding this line:
```
initramfs initramfs-linux.img followkernel
```

graysky2, 2023-01-05:
Are you seeing the four files ok?
[image]

pelwell, 2023-01-05:
Yes - they're downloaded and installing

graysky2, 2023-01-05:
Great!  For testing, I took the identical image you're downloading, added the initramfs img, and the line to `/boot/config.txt` to use it, but reproduced the panic on boot, so it did not seem to matter.

> I have something that is showing signs of life, but I had to locate a firmware, and it's lacking an initramfs file - which causes it to panic.

Please let me know which firmware file/files I am missing.

pelwell, 2023-01-05:
Nothing - it all seems OK now I have the correct archive. It doesn't let me log in as root (it stalls for ~10 seconds, report `double free or corruption (out)` and logs out), but the alarm user works and can su.

Switching to cma-512 in config.txt looks OK to start with then it goes haywire:
```
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 38531 entries in 113 pages
[    0.000000] ftrace: allocated 113 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000000] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000015] Switching to timer-based delay loop, resolution 18ns
[    0.000318] Console: colour dummy device 80x30
[    0.000834] printk: console [tty1] enabled
[    0.000897] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.000936] pid_max: default: 32768 minimum: 301
[    0.001054] LSM: Security Framework initializing
[    0.001226] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.001264] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.002471] cgroup: Disabling memory control group subsystem
[    0.002665] CPU: Testing write buffer coherency: ok
[    0.002722] CPU0: Spectre BHB: enabling loop workaround for all CPUs
[    0.003057] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.003962] cblist_init_generic: Setting adjustable number of callback queues.
[    0.003992] cblist_init_generic: Setting shift to 2 and lim to 1.
[    0.004121] cblist_init_generic: Setting shift to 2 and lim to 1.
[    0.004264] Setting up static identity map for 0x200000 - 0x20003c
[    0.004429] rcu: Hierarchical SRCU implementation.
[    0.004451] rcu:     Max phase no-delay instances is 1000.
[    0.005212] smp: Bringing up secondary CPUs ...
[    0.006077] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.007036] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.007976] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.008087] smp: Brought up 1 node, 4 CPUs
[    0.008149] SMP: Total of 4 processors activated (432.00 BogoMIPS).
[    0.008173] CPU: All CPU(s) started in HYP mode.
[    0.008190] CPU: Virtualization extensions available.
[    0.008787] devtmpfs: initialized
[    0.019334] VFP support v0.3: implementor 41 architecture 3 part 40 variant 8 rev 0
[    0.019529] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.019571] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.032485] pinctrl core: initialized pinctrl subsystem
[    0.033453] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.035706] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.036775] thermal_sys: Registered thermal governor 'step_wise'
[    0.037046] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.037090] hw-breakpoint: maximum watchpoint size is 8 bytes.
[    0.037437] Serial: AMBA PL011 UART driver
[    0.046565] bcm2835-mbox fe00b880.mailbox: mailbox enabled
[    0.070018] raspberrypi-firmware soc:firmware: Attached to firmware from 2022-12-12T11:56:56, variant start
[    0.080032] raspberrypi-firmware soc:firmware: Firmware hash is ed6f6b8fcdc6476410b9cf75d141633461d34bdd
[    0.121957] kprobes: kprobe jump-optimization is enabled. All kprobes are optimized if possible.
[    0.290215] raid6: neonx8   gen()  1219 MB/s
[    0.460350] raid6: neonx4   gen()  1667 MB/s
[    0.630481] raid6: neonx2   gen()  1475 MB/s
[    0.800611] raid6: neonx1   gen()  1153 MB/s
[    0.970776] raid6: int32x8  gen()   390 MB/s
[    1.140897] raid6: int32x4  gen()   446 MB/s
[    1.311032] raid6: int32x2  gen()   498 MB/s
[    1.481166] raid6: int32x1  gen()   424 MB/s
[    1.481191] raid6: using algorithm neonx4 gen() 1667 MB/s
[    1.651249] raid6: .... xor() 1161 MB/s, rmw enabled
[    1.651270] raid6: using neon recovery algorithm
[... 1854 lines cut ...]
[   27.828859] CPU2: stopping
[   27.828868] CPU: 2 PID: 0 Comm: swapper/2 Tainted: G         C         6.1.1-1-rpi-ARCH #1
[   27.828881] Hardware name: BCM2711
[   27.828888]  unwind_backtrace from show_stack+0x18/0x1c
[   27.828904]  show_stack from dump_stack_lvl+0x70/0xac
[   27.828920]  dump_stack_lvl from do_handle_IPI+0x2c0/0x2f8
[   27.828937]  do_handle_IPI from ipi_handler+0x20/0x28
[   27.828951]  ipi_handler from handle_percpu_devid_irq+0x84/0x1c0
[   27.828967]  handle_percpu_devid_irq from generic_handle_domain_irq+0x30/0x40
[   27.828980]  generic_handle_domain_irq from gic_handle_irq+0x7c/0x90
[   27.828992]  gic_handle_irq from generic_handle_arch_irq+0x34/0x44
[   27.829008]  generic_handle_arch_irq from call_with_stack+0x18/0x20
[... cut ...]
[   27.829067] dfa0: c0208150 c0208154 60000013 ffffffff
[   27.829074]  __irq_svc from arch_cpu_idle+0x40/0x44
[   27.829089]  arch_cpu_idle from default_idle_call+0x3c/0xbc
[   27.829106]  default_idle_call from do_idle+0xd8/0x14c
[   27.829118]  do_idle from cpu_startup_entry+0x20/0x24
[   27.829129]  cpu_startup_entry from secondary_start_kernel+0x130/0x138
[   27.829142]  secondary_start_kernel from 0x2017f4
[   28.221485] ---[ end Kernel panic - not syncing: System is deadlocked on memory ]---
```

graysky2, 2023-01-05:
Wow, at least you can both reproduce the bug and capture the output.  Any thoughts on causes?

About root not logging you in, I do not know.  Can you change the root password after `sudo su` and if so can you login at that point?

pelwell, 2023-01-05:
[I'll leave the root login problem for now - it's not hindering me]

I think it's to do with SWIOTLB placement. This is the relevant output from a bunch of configurations:
```
8GB cma-320:
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ac00000, size 320 MiB
[    0.000000] software IO TLB: mapped [mem 0x0000000012400000-0x0000000016400000] (64MB)
[    0.000000] Memory: 7756476K/8245248K available (12288K kernel code, 1555K rwdata, 3328K rodata, 2048K init, 647K bss, 161092K reserved, 327680K cma-reserved, 7458816K highmem)

8GB cma-448:
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000012c00000, size 448 MiB
[    0.000000] software IO TLB: mapped [mem 0x000000000a400000-0x000000000e400000] (64MB)
[    0.000000] Memory: 7625148K/8245248K available (12288K kernel code, 1555K rwdata, 3328K rodata, 2048K init, 647K bss, 161348K reserved, 458752K cma-reserved, 7458816K highmem)

8GB cma-512:
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000ec00000, size 512 MiB
[    0.000000] software IO TLB: mapped [mem 0x0000000006400000-0x000000000a400000] (64MB)
[    0.000000] Memory: 7559484K/8245248K available (12288K kernel code, 1555K rwdata, 3328K rodata, 2048K init, 647K bss, 161476K reserved, 524288K cma-reserved, 7458816K highmem)

4GB cma-320:
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ac00000, size 320 MiB
[    0.000000] software IO TLB: mapped [mem 0x0000000014890000-0x0000000018890000] (64MB)
[    0.000000] Memory: 3599036K/4050944K available (12288K kernel code, 1555K rwdata, 3328K rodata, 2048K init, 647K bss, 124228K reserved, 327680K cma-reserved, 3264512K highmem)

4GB cma-512:
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000ec00000, size 512 MiB
[    0.000000] software IO TLB: mapped [mem 0x0000000008890000-0x000000000c890000] (64MB)
[    0.000000] Memory: 3402044K/4050944K available (12288K kernel code, 1555K rwdata, 3328K rodata, 2048K init, 647K bss, 124612K reserved, 524288K cma-reserved, 3264512K highmem)
```
Notice how in the problematic 8GB cma-512 case, the TLB area ends up very low in RAM (0x6400000, i.e. 100M).

The allocated SWIOTLB  and CMA ranges (and the gaps between them) are:
```
8GB cma-320: SWIOTLB=12400000-16400000 (4800000) CMA=1ac00000-2ec00000
8GB cma-448: SWIOTLB=0a400000-0e400000 (4800000) CMA=12c00000-2ec00000
8GB cma-512: SWIOTLB=06400000-0a400000 (4800000) CMA=0ec00000-2ec00000
4GB cma-320: SWIOTLB=14890000-18890000 (2370000) CMA=1ac00000-2ec00000
4GB cma-512: SWIOTLB=08890000-0c890000 (2370000) CMA=0ec00000-2ec00000
```
This shows that the gap beteen CMA and SWIOTLB is smaller on the 4GB board, and as a result (OK, the causality is unproven) the SWIOTLB area is higher.

My working theory is that the gap between the top of the kernel code+data and the base of SWIOTLB is used for normal kernel allocations (which can't use HIGHMEM - that's mainly for applications), and with SWIOTLB at 100MB there isn't enough space left.

pelwell, 2023-01-05:
And the same for RPiOS:
```
8GB cma-512 (RPiOS):
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000ec00000, size 512 MiB
[    0.000000] software IO TLB: mapped [mem 0x0000000006400000-0x000000000a400000] (64MB)
[    0.000000] Memory: 7557548K/8245248K available (10240K kernel code, 1386K rwdata, 3308K rodata, 2048K init, 591K bss, 163412K reserved, 524288K cma-reserved, 7458816K highmem)
```
Note that this is the same placement as on Arch, but it doesn't crash. The kernel is a bit smaller, and perhaps that makes a difference - I don't know.

graysky2, 2023-01-05:
Some of that is over my head, but the patterns are interesting.  I don't know how much of an intellectual contributor I can be at this point.

As a control, I installed the distro armv7h image to a new card, updated it, and was able to boot it with the line that causes the panics on my "old" image intact.  So that points to something on my image/the image I shared with you.

```
8GB cma-512 (fresh image)
[  +0.000000] Reserved memory: created CMA memory pool at 0x000000000e800000, size 512 MiB
[  +0.000000] software IO TLB: mapped [mem 0x0000000006000000-0x000000000a000000] (64MB)
[  +0.000000] Memory: 7547560K/8245248K available (14336K kernel code, 1540K rwdata, 3936K rodata, 2048K init, 631K bss, 173400K reserved, 524288K cma-reserved, 7458816K highmem)
```

pelwell, 2023-01-06:
Although it would be nice to fix (or even understand) this, have you considered just using a 64-bit kernel on the Pi 4, i.e. dropping `kernel7l.img`? We're currently contemplating changing the firmware to make loading the 64-bit kernel the default, but allow the use of `arm_64bit=0` to revert to the 32-bit. You may possibly encounter some build problems with apps that assume a 64-bit kernel implies a 64-bit userland.

graysky2, 2023-01-06:
I understand development is moving towards 64-bit.  If there isn't an easy fix (sounds like cause is still unknown), I understand the reluctance to spend resources on this.

graysky2, 2023-01-06:
> Try running memtester with sudo.
"Run memtester as root so that it can mlock the memory it tests."

I emailed Charles, the author of memtester, with the question of why can't it allocate more RAM on arm 32-bit but it can on arm64.  His reply:

_This is normal.  A 32-bit kernel, or a 32-bit userland (as some ARM devices use, even with a 64-bit kernel) limits the amount of memory that can be allocated by *any* userland process.  On a typical Linux kernel, that will be half of a 32-bit address space - or 4GB divided by 2, for a total of 2GB.  The other half/2GB of the address space is reserved for the kernel.

This 2GB memory limitation is the reason for 64-bit machines today.

> want 4000MB (4194304000 bytes)got 2047MB (2147479552 bytes), trying mlock
> ...^C

And there you see it successfully allocating just a touch under the 2GB maximum - it's impressively close to the max, you must have pretty much nothing else running.  memtester itself needs a little memory for its own code, the system C library, I/O buffers, etc._

pelwell, 2023-01-09:
I'm slowly making progress. The 0x4800000 (72MB) gap between the SWIOTLB and CMA on an 8GB Pi is the page table. It contains one 36-byte `struct page` for each 4KB page of RAM. 8GB is 2*1024*1024*4KB, so the page table is 2*1024*1024*36 = 72MB. Naturally on a 4GB Pi the page table is smaller, but it's actually slightly less than the expected 36MB because the peripheral space masks the top 64MB (16*1024*4KB) of RAM from the ARMs, leaving 1008*1024*4KB usable, which requires a 1008*1024*36 = 0x2370000 page table.

This goes some way to explaining why things are worse with more RAM, since several things need to be in low memory and a 512MB CMA grabs a large chunk of it, but I'd still like to know exactly what kind of memory we're short of in the failing case and whether that is reasonable.

graysky2, 2023-01-09:
Thanks @pelwell - I honestly do not know how many 8GB owners will be running armv7h.

mkreisl, 2023-02-13:
I don't know if this is still of interest, but I had the problem (not me directly but a customer) years ago with a Pi4/8GB, all in 32bit. The problem only occurs with 8GB, my 4GB version doesn't have the problem. so since then i just put cma-384 in config.txt (XBian)
Obviously the bug is still there, strange that it does not seem to occur with everyone

pelwell, 2023-02-13:
And didn't make any meaningful progress beyond what is shown above, I'm closing the issue to indicate that it is not being worked on.

We're seriously considering making the 64-bit kernel the default on all Pi 4s, but for 8GB units it could make quite a performance difference in some situations.

closed 2023-02-13 by pelwell

mkreisl, 2023-02-13:
> We're seriously considering making the 64-bit kernel the default on all Pi 4s, but for 8GB units it could make quite a performance difference in some situations.

I started using the 64bit kernel about a year ago. One user told me that video playback in Kodi looks better with the 64bit kernel. But I can't really understand that.

@pelwell 
Just one last question, the performance differences with Pi4/8GB, are they positive or negative. I only have a 4GB version to develop, so unfortunately I can not collect my own experiences.

pelwell, 2023-02-13:
The difference is positive - the kernel can address all of RAM without having to set up temporary mappings, and there may be fewer levels in the page tables (I'm guessing a bit there).