raspberrypi/linux #5302: Memory leak in video driver

Flole998, 2023-01-05:
### Describe the bug

When playing video files with Kodi the kernel memory (namely kmalloc-128 and kmalloc-256) is constantly increasing until the system eventually runs OOM.

### Steps to reproduce the behaviour

Play a video with kodi and watch slabtop.

### Device (s)

Raspberry Pi 3 Mod. B+

### System

```
# cat /etc/rpi-issue
Raspberry Pi reference 2018-10-09
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7b40fd11bc13d6496bb6bd1bc02f946b7e0ac7cb, stage2
# vcgencmd version
Dec 12 2022 11:59:10
Copyright (c) 2012 Broadcom
version ed6f6b8fcdc6476410b9cf75d141633461d34bdd (clean) (release) (start)
# uname -a
Linux Raspberry-Schlafzimmer 5.15.84-v8+ #1611 SMP PREEMPT Wed Jan 4 16:19:37 GMT 2023 aarch64 GNU/Linux
# cat /etc/issue
Debian GNU/Linux 11
```

### Logs

```
# cat kmalloc-128/alloc_traces | sort -rn | head
  39453 kstrdup_const+0x3c/0x48 age=9949/204277/303649 pid=0-712 cpus=0-3
....
# cat kmalloc-256/alloc_traces | sort -rn | head
  33110 drmm_mutex_init+0x54/0x68 [drm] age=4618/195716/305391 pid=178-587 cpus=0-3
....
```

### Additional context

I am using

```
dtoverlay=vc4-kms-v3d,cma-256
```

pelwell, 2023-01-09:
I'm surprised that this problem hasn't been more widely reported. Is there a particular kind of video (size, format) that causes this? Can you point us at a specific clip that is known to cause leaks?

Flole998, 2023-01-09:
I've just tested with the sample BigBuckBunny movie from http://commondatastorage.googleapis.com/gtv-videos-bucket/sample/BigBuckBunny.mp4 which shows this behavior. I don't think it's related to a specific format or type though. After playing that video once I have 22000k for kmalloc-128 and 17500k for kmalloc-256, at the beginning those were significantly smaller (around 9000k and 8000k). Once playback stops the memory also stops increasing.

I'm not really sure why nobody else noticed it, it's leaking rather slowly so maybe others are rebooting the device often enough or aren't letting it play many hours of video. I know that this is some kind of regression as it used to work perfectly fine in the past. It would be great if you could let me know if you can reproduce it.

Reoogamal0, 2023-01-09:
Ok

Reoogamal0, 2023-01-09:
هل يمكن التوضيح

popcornmix, 2023-01-10:
I've just tried this playing h.264 1920x1080 big buck bunny on kodi on RPiOS.
HW decode (kodi shows ff-h264_v4l2mem-drm-prime) and display through DRMPrime.

I've added to cmdline.txt `slub_debug=U,kmalloc-128,kmalloc-256`

```
pi@pi4:~ $ sudo  cat /sys/kernel/debug/slab/kmalloc-128/alloc_traces  | sort -rn | head -5
     68 bus_add_driver+0x50/0x1f8 waste=3264/48 age=43772/43775/43778 pid=1 cpus=0
     63 device_add+0x4a8/0x898 waste=1260/20 age=44012/44012/44012 pid=1 cpus=0
     58 bus_add_driver+0x50/0x1f8 waste=2784/48 age=42956/43807/44203 pid=1 cpus=0,2
     44 perf_cgroup_css_alloc+0x24/0x5c waste=352/8 age=30946/42112/42807 pid=1-666 cpus=0-3
     43 mod_sysfs_setup+0x47c/0x664 waste=2408/56 age=42376/42445/42487 pid=203-225 cpus=0-3
pi@pi4:~ $ sudo  cat /sys/kernel/debug/slab/kmalloc-256/alloc_traces  | sort -rn | head -5
     16 blkg_alloc+0x38/0x1a4 age=44209/44209/44209 pid=1 cpus=0
     16 alloc_desc+0x3c/0x1e4 age=44730/44730/44730 pid=0 cpus=0
      8 blkg_alloc+0x38/0x1a4 age=44207/44207/44207 pid=1 cpus=0
      8 blk_alloc_flush_queue+0x74/0xdc age=44207/44207/44207 pid=1 cpus=0
      8 alloc_desc+0x3c/0x1e4 age=44730/44730/44730 pid=0 cpus=0
```
So I don't seem to be seeing this.
I happen to be on 6.2 kernel and pi4, but will try with 5.15 and pi3 when I get a chance (although I'd be surprised if that changed things)

Flole998, 2023-01-10:
In my case kodi is showing `ff-h264_v4l2m2m-drm_prime`, could that already make a difference?

popcornmix, 2023-01-10:
> In my case kodi is showing `ff-h264_v4l2m2m-drm_prime`, could that already make a difference?

That's what I have (with a minor typo). It's the same codec.

popcornmix, 2023-01-10:
On 5.15 and Pi4 it's a little different but I'm not seeing your leak.
```
pi@pi4:~ $ sudo  cat /sys/kernel/debug/slab/kmalloc-128/alloc_traces  | sort -rn | head -5
   1610 alloc_nfs_open_context+0x38/0x14c age=0/113910/143406 pid=1-1620 cpus=0-3
    445 device_add+0x4fc/0x8a0 age=70807/143500/144087 pid=1-1065 cpus=0-3
    267 drm_property_create+0xac/0x1a8 [drm] age=142787/142800/142855 pid=242 cpus=0,2-3
    170 bus_add_driver+0x58/0x1f8 age=142852/143859/144086 pid=1-246 cpus=0-3
     70 assoc_array_insert+0x1ec/0xa10 age=52624/140636/144001 pid=1-1392 cpus=0-3
pi@pi4:~ $ sudo  cat /sys/kernel/debug/slab/kmalloc-256/alloc_traces  | sort -rn | head -5
     79 alloc_desc+0x38/0x19c age=144477/144558/144564 pid=0-1 cpus=0-1
     61 __class_register+0x34/0x158 age=70646/143204/144561 pid=1-1122 cpus=0-2
     27 bus_register+0x30/0x214 age=143348/144457/144562 pid=1-242 cpus=0-2
     25 blkg_alloc+0x44/0x1ac age=144313/144486/144494 pid=1-92 cpus=1,3
     25 alloc_workqueue+0x8c/0x448 age=143848/144481/144564 pid=0-92 cpus=0-1,3
```

Flole998, 2023-01-10:
Alright, that's interesting then... 

Could it be related to some Kodi settings? Looking at my guisettings.xml I have these video settings:

```
    <setting id="videoscreen.monitor" default="true">Default</setting>
    <setting id="videoscreen.screen" default="true">0</setting>
    <setting id="videoscreen.resolution">18</setting>
    <setting id="videoscreen.screenmode" default="true">DESKTOP</setting>
    <setting id="videoscreen.fakefullscreen" default="true">true</setting>
    <setting id="videoscreen.blankdisplays" default="true">false</setting>
    <setting id="videoscreen.delayrefreshchange">3</setting>
    <setting id="videoscreen.dither" default="true">false</setting>
    <setting id="videoscreen.ditherdepth" default="true">8</setting>
    <setting id="videoscreen.cmsenabled" default="true">false</setting>
    <setting id="videoscreen.cmsmode" default="true">0</setting>
    <setting id="videoscreen.cms3dlut" default="true" />
    <setting id="videoscreen.displayprofile" default="true" />
    <setting id="videoscreen.cmswhitepoint" default="true">0</setting>
    <setting id="videoscreen.cmsprimaries" default="true">0</setting>
    <setting id="videoscreen.cmsgammamode" default="true">0</setting>
    <setting id="videoscreen.cmsgamma" default="true">220</setting>
    <setting id="videoscreen.cmslutsize" default="true">6</setting>
    <setting id="videoscreen.hwscalingfilter" default="true">true</setting>
    <setting id="videoscreen.limitguisize" default="true">0</setting>
    <setting id="videoscreen.limitedrange" default="true">false</setting>
    <setting id="videoscreen.whitelist" default="true" />
    <setting id="videoscreen.whitelistpulldown" default="true">false</setting>
    <setting id="videoscreen.whitelistdoublerefreshrate" default="true">false</setting>
    <setting id="videoscreen.stereoscopicmode" default="true">0</setting>
    <setting id="videoscreen.preferedstereoscopicmode" default="true">100</setting>
```
Is the "triple-buffering" correct or is double-buffering the correct setting? I tried both and it doesn't seem to affect the issue.

I also tried disabling the DRM-PRIME-Decoder in Kodi settings, same result. It even leaks if I press a button and the GUI renders a new view.

Is there anything I could try to get more information on what might be happening? Or could this be an issue that only appears on "older" pi's and not on the 4?

popcornmix, 2023-01-10:
```
pi@b3plus:~ $ uname -a
Linux b3plus 5.15.84-v7+ #1613 SMP Thu Jan 5 11:59:48 GMT 2023 armv7l GNU/Linux
pi@b3plus:~ $ sudo  cat /sys/kernel/debug/slab/kmalloc-128/alloc_traces  | sort -rn | head -5
  19163 drmm_mutex_init+0x50/0x54 [drm] age=5295/37416/109394 pid=223-876 cpus=0-3
   1650 alloc_nfs_open_context+0x38/0x14c age=0/98095/110148 pid=1-1240 cpus=0-3
    370 device_add+0x4fc/0x8a0 age=95537/110633/111047 pid=1-891 cpus=0-3
    232 drm_property_create+0xac/0x1a8 [drm] age=109330/109340/109394 pid=223 cpus=1-3
     96 bus_add_driver+0x58/0x1f8 age=109394/110600/111046 pid=1-227 cpus=0-3
pi@b3plus:~ $ sudo  cat /sys/kernel/debug/slab/kmalloc-256/alloc_traces  | sort -rn | head -5
    200 alloc_desc+0x38/0x19c age=111441/111943/111946 pid=0-79 cpus=0
     56 __class_register+0x34/0x158 age=95320/111443/111943 pid=1-952 cpus=0-3
     25 bus_register+0x30/0x214 age=110318/111788/111944 pid=1-223 cpus=0-3
     25 blkg_alloc+0x44/0x1ac age=111585/111845/111857 pid=1-35 cpus=1,3
     23 alloc_workqueue+0x8c/0x448 age=110996/111827/111946 pid=0-35 cpus=0-1,3
pi@b3plus:~ $ sudo  cat /sys/kernel/debug/slab/kmalloc-64/alloc_traces  | sort -rn | head -5
  24944 kstrdup_const+0x38/0x3c age=3968/58309/115146 pid=0-957 cpus=0-3
   7031 nfs_readdir_add_to_array+0x2c/0x154 age=9823/103569/114210 pid=1-1208 cpus=0-3
   4070 nfs_access_add_cache+0x34/0x23c age=9423/107707/114261 pid=1-1217 cpus=0-3
   1240 safe_name+0x90/0xb4 age=115144/115145/115146 pid=1 cpus=0
   1078 load_module+0x229c/0x294c age=98512/112380/114197 pid=1-957 cpus=0-3
```

Interesting. Pi3 is showing a leak. For some reason I'm getting the leak in different sized bins from you. 
The allocations don't disappear when video stops, or even when kodi exits.

@jc-kynesim @6by9 any ideas? Can't think of any code that is different on Pi3 and Pi4.

Flole998, 2023-01-10:
> For some reason I'm getting the leak in different sized bins from you.

I guess 32 bit vs 64 bit kernel? I need more space to fit the addresses into so the struct get larger?

popcornmix, 2023-01-10:
@jc-kynesim 
```
./hello_drmprime ~/dell/bbb_sunflower_1080p_30fps_normal.mp4
```
seems to have the `kstrdup_const` leak.

jc-kynesim, 2023-01-10:
Is the drm_property_create entry a leak or just current alloc?

jc-kynesim, 2023-01-10:
hello_drmprime probably isn't a model of careful cleanup - are you suggesting the leak is userspace assisted? or are you just noting that it fails there too?

popcornmix, 2023-01-10:
`drm_property_create` doesn't increase while video is playing, or between runs.
The allocations were there before I starting playing video.

popcornmix, 2023-01-10:
> hello_drmprime probably isn't a model of careful cleanup - are you suggesting the leak is userspace assisted? or are you just noting that it fails there too?

Kodi and hello_drmprime appear to behave identically (note: I've edited previous post - both leaks are present in both cases).
I suspect the leak is in the kernel driver (but that is my guess).
I'm just giving you an example more reproducible than kodi.

popcornmix, 2023-01-10:
That command line doesn't appear to increase the leak numbers.
If I add `vout_drm` in place of the `null` then I see the leak again.

jc-kynesim, 2023-01-10:
Then in all likelyhood its some part of the dmabuf import process

jc-kynesim, 2023-01-10:
I'll see if I can narrow it down tomorrow - I've got to go now.

6by9, 2023-01-10:
drmm_mutex_init is only used in a few places.

https://github.com/torvalds/linux/commit/72f1beeee46b863a947c1a25b8f14319d79d7ccb switched vc4_create_object (and a few others in vc4) to using it. That initially feels wrong as AIUI it ties the allocation to the lifetime of the drm_device, not necessarily of just the buffer.
It being in vc4_bo also feels like a match with it only showing up on Pi3, as Pi0-3 advertise ```.gem_create_object = vc4_create_object,```, but Pi4 doesn't.
Oddly there doesn't appear to be a counterpart to gem_create_object for freeing objects in struct drm_driver

Having outstanding drmm allocs also follows with the kstrdup_const leak, as drmm_kmalloc and __drmm_add_action both use it to save the alloc name.

@mripard Am I reading the situation correctly?

6by9, 2023-01-10:
It looks like lima_gem_create_object and panfrost_gem_create_object both have the same issue of calling mutex_init but not calling mutex_destroy, so this isn't unique to vc4.

Flole998, 2023-01-10:
I assume I can't just disable the "vc4_create_object" on Pi 3 somehow so it behaves the same way as the Pi 4 as a temporary workaround? I guess you guys are close to finding a proper fix though already, which is really impressive. I also found that commit when I did some research before opening this bug report but I thought it was a fix for this issue, previously it was never free()'d and now it should apparently be free()'d somewhere/somehow but isn't.

6by9, 2023-01-10:
In the case of Kodi you may get away with dropping 
https://github.com/raspberrypi/linux/blob/rpi-5.15.y/drivers/gpu/drm/vc4/vc4_drv.c#L215
```.gem_create_object = vc4_create_object,```
and recompiling the kernel, but that is very likely to break any usage of the 3D side of the vc4 driver.

popcornmix, 2023-01-10:
> In the case of Kodi you may get away with dropping

That didn't go well. I booted, but running ffmpeg (with drm output) produced:
http://sprunge.us/a8mHop

mripard, 2023-01-11:
> drmm_mutex_init is only used in a few places.
> 
> [torvalds/linux@72f1bee](https://github.com/torvalds/linux/commit/72f1beeee46b863a947c1a25b8f14319d79d7ccb) switched vc4_create_object (and a few others in vc4) to using it. That initially feels wrong as AIUI it ties the allocation to the lifetime of the drm_device, not necessarily of just the buffer. It being in vc4_bo also feels like a match with it only showing up on Pi3, as Pi0-3 advertise `.gem_create_object = vc4_create_object,`, but Pi4 doesn't. Oddly there doesn't appear to be a counterpart to gem_create_object for freeing objects in struct drm_driver
> 
> Having outstanding drmm allocs also follows with the kstrdup_const leak, as drmm_kmalloc and __drmm_add_action both use it to save the alloc name.
> 
> @mripard Am I reading the situation correctly?

Yes, I think you do. In the typical case, mutex_destroy() won't do anything (it's a nop: https://elixir.bootlin.com/linux/latest/source/include/linux/mutex.h#L89) but drmm_* functions typically allocate a small structure to store a copy of the resources it needs to free the resource later. That structure will indeed be freed (along with the resource) when the DRM device goes away. So I think what we see is an accumulation of those drmm structures, rather than mutexes themselves.

I think we should partially revert the commit you pointed to for vc4_create_object, and add the mutex_destroy call at the end of  vc4_free_object.

I'll send a PR

mripard, 2023-01-11:
> drmm_mutex_init is only used in a few places.
>
> [torvalds/linux@72f1bee](https://github.com/torvalds/linux/commit/72f1beeee46b863a947c1a25b8f14319d79d7ccb) switched vc4_create_object (and a few [...]

Yes, I think you do. In the typical case, mutex_destroy() won't do anything (it's a nop: https://elixir.bootlin.com/linux/latest/source/include/linux/mutex.h#L89) but drmm_* functions typically allocate a small structure to store a copy of the resources it needs to free the resource later. That structure will indeed be freed (along with the resource) when the DRM device goes away. So I think what we see is an accumulation of those drmm structures, rather than mutexes themselves.

I think we should partially revert the commit you pointed to for vc4_create_object, and add the mutex_destroy call at the end of  vc4_free_object.

I'll send a PR

PR #5313 (2023-01-11): drm/vc4: bo: Fix drmm_mutex_init memory hog
Commit 374146cad469 ("drm/vc4: Switch to drmm_mutex_init") converted, among other functions, vc4_create_object() to use drmm_mutex_init().

However, that function is used to allocate a BO, and therefore the mutex needs to be freed much sooner than when the DRM device is removed from the system.

For each buffer allocation we thus end up allocating a small structure as part of the DRM-managed mechanism that is never freed, eventually leading us to no longer having any free memory anymore.

Let's switch back to mutex_init/mutex_destroy to deal with it  [... cut]
closed 2023-01-11 by pelwell