raspberrypi/linux #5305: 6.1 fails to compile without CONFIG_PWM=y

jolla, 2023-01-05:
### Describe the bug

When attempting to compile the latest 6.1 kernel it fails with the following errors. 

After adding `CONFIG_PWM=y` to bcm2711_defconfig compilation is able to continue. 

### Steps to reproduce the behaviour

Attempt to compile 6.1 kernel, it will fail without CONFIG_PWM=y

### Device (s)

Other

### System

Device used to compile: Linux 5.4.0-135-generic #152-Ubuntu SMP Wed Nov 23 20:19:22 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

### Logs

```
Value requested for CONFIG_SENSORS_PWM_FAN not in final .config
Requested value:  CONFIG_SENSORS_PWM_FAN=m
Actual value:     

Value requested for CONFIG_IR_PWM_TX not in final .config
Requested value:  CONFIG_IR_PWM_TX=m
Actual value:     

Value requested for CONFIG_BACKLIGHT_PWM not in final .config
Requested value:  CONFIG_BACKLIGHT_PWM=m
Actual value:     

Value requested for CONFIG_BACKLIGHT_LM3630A not in final .config
Requested value:  CONFIG_BACKLIGHT_LM3630A=m
Actual value:     

Value requested for CONFIG_LEDS_PWM not in final .config
Requested value:  CONFIG_LEDS_PWM=y
Actual value:     

Value requested for CONFIG_PWM_BCM2835 not in final .config
Requested value:  CONFIG_PWM_BCM2835=m
Actual value:     

Value requested for CONFIG_PWM_PCA9685 not in final .config
Requested value:  CONFIG_PWM_PCA9685=m
Actual value:     

Value requested for CONFIG_PWM_RASPBERRYPI_POE not in final .config
Requested value:  CONFIG_PWM_RASPBERRYPI_POE=m
Actual value:     

[ ERROR ] Error updating defconfig. See above log to check which configs had conflicts.

```

### Additional context

https://github.com/raspberrypi/linux/blob/rpi-6.1.y/arch/arm64/configs/bcm2711_defconfig

pelwell, 2023-01-05:
I'm in the middle of forward-porting recent changes to 6.1, and that will be fixed in the process (I've just had to do it for 6.0). I'm guessing something was back-ported from 6.2.

jolla, 2023-01-05:
Thank you, FWIW, I don't have this issue with the 6.2 branch as `CONFIG_PWM=y` exists in bcm2711_defconfig
https://github.com/raspberrypi/linux/blob/rpi-6.2.y/arch/arm64/configs/bcm2711_defconfig#L1439

pelwell, 2023-01-05:
Yes, I added it earlier this week.

pelwell, 2023-01-06:
The config fix-up patch is now in rpi-6.1.y, which should resolve your issue,

closed 2023-01-14 by jolla