raspberrypi/linux #5307: Kernel version mismatch

spynetcamera, 2023-01-06:
### Describe the bug

My Raspberry Pi 1 B+, after apt update/full-upgrade, is still 5.15.76+, while the official sources are now 5.15.84+.
I downloaded and bult according to the documentation in https://www.raspberrypi.com/documentation/computers/linux_kernel.html.

### Steps to reproduce the behaviour

git clone --depth=1 https://github.com/raspberrypi/linux

cd linux
KERNEL=kernel
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- bcmrpi_defconfig

make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- zImage modules dtbs

### Device (s)

Raspberry Pi 1 Mod. B+

### System

Raspberry Pi reference 2022-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 8a42abcd1dbd8c9c1fdfca4e0c3778255b2f9cc4, stage2

Oct 26 2022 11:10:35
Copyright (c) 2012 Broadcom
version c72ad6b26ff40c91ef776b847436094ee63fabee (clean) (release) (start)

Linux raspberrypi 5.15.76+ #1597 Fri Nov 4 12:11:43 GMT 2022 armv6l GNU/Linux

pelwell, 2023-01-06:
You are reporting that the apt kernel package, built from the `stable` branch of the rpi-firmware repo, doesn't include the very latest kernel commits? That's a feature, not a bug.

pelwell, 2023-01-06:
If instead you are saying that you built the latest sources but they don't appear to be running, then did you actually install the built kernel?
```
sudo make modules_install
sudo cp arch/arm/boot/dts/*.dtb /boot/
sudo cp arch/arm/boot/dts/overlays/*.dtb* /boot/overlays/
sudo cp arch/arm/boot/dts/overlays/README /boot/overlays/
sudo cp arch/arm/boot/zImage /boot/kernel.img
```

6by9, 2023-01-06:
And don't forget to reboot after installing all the files.