raspberrypi/linux #5311: net/AX.25 socket termination issue

kq6up, 2023-01-09:
### Describe the bug

When user space applications utilize the native Linux AX.25 stack, the socket does not go
away cleanly after a connection is made and terminated.  This has already been has already been fixed and committed upstream as I check for deltas between 5.1.84 and 6.1.4 and they match the deltas listed in the fix here:

https://lore.kernel.org/lkml/20220530152158.108619-1-duoming@zju.edu.cn/

### Steps to reproduce the behaviour

The AX.25 is for licensed ham radio operators, but someone could theoretically do this over FRS or CB.  It is a bit involved, and since it has already been committed I am not sure this bit is necessary.  I did test the patches on the current raspberry Pi sources by pulling/patching/compiling/installing/ and testing and verified that they are working on the Pi.  See my work here:

https://groups.io/g/KM4ACK-Pi/topic/howto_patch_the_kernel_for/95904470?p=,,,20,0,0,0::recentpostdate/sticky,,,20,2,0,95904470,previd%3D9223372036854775807,nextid%3D1672690066927215010&previd=9223372036854775807&nextid=1672690066927215010

It would be great not to have to patch/compile for each new pathchlevel that gets pushed out in the updates.

### Device (s)

Raspberry Pi 3 Mod. B+

### System

 cat /etc/rpi-issue
Raspberry Pi reference 2022-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 005a8c73b05a2cab394073150208bf4f069e861a, stage4

vcgencmd version
Jan  5 2023 10:46:54 
Copyright (c) 2012 Broadcom
version 8ba17717fbcedd4c3b6d4bce7e50c7af4155cba9 (clean) (release) (start)

uname -a
Linux PORTApi 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux

pelwell, 2023-01-10:
In my opinion that commit should have been backported to 5.15 (an LTS branch that is still maintained) by the kernel maintainers as a matter of course because of the Fixes tags. I've cherry-picked it back to rpi-5.15.y (see 9ad0d2c58bcf5) - it's already in our other active branches, including the beta builds of rpi-6.1.y installable with `sudo rpi-update next`.

kq6up, 2023-01-10:
I am curious as to why I see changes in the rpi-5.15y version on github,
but when I pull according to the kernel building instructions on the rpi
site there is no delta when I diff against unpatched source.  Sorry if
these questions seem naive or awkwardly stated.  It had been probably 20
years since I have last compiled a kernel.

pelwell, 2023-01-10:
Without seeing the commands you have typed I'm going to have to guess, but the most likely explanation would be that you didn't "fetch" the latest commits from our linux GitHub repo.

git maintains local copies of the contents of "remotes" - repositories stored on other servers (or different directories in the same server, in some cases). You can checkout branches from those remotes as often as you like, but unless you use `git fetch <remote>` (or some flavours of `git pull`) you won't see any commits that have been added to the remote.

kq6up, 2023-01-10:
I checked again this morning and they are in what I pulled from the server
using the same git command.  So no delta from my patched code (which tells
me that those deltas have been committed to the tree I pulled this
morning).  So all is good on the source side, when would I expect the
binaries to come downstream in general kernel update to everyone?

-Chris Maness

pelwell, 2023-01-10:
Actually, you might get lucky - there have been a few post-launch tweaks for the V3 camera driver, which makes a new release more likely then usual.

kq6up, 2023-01-10:
Nice, thanks for your help, Phil.

-Chris KQ6UP