raspberrypi/linux #5320: Raspberry Pi 3B and slow internet downloads

JG24, 2023-01-14:
### Describe the bug

A few days ago I decided to make something like a NAS out of my Raspberry Pi 3 Model B Rev 1.2. Okay, I know, big words. I simply connected the SSD with a USB cable, configured the system and it worked. I have one problem though. I don't expect miracles from this set. Raspberry has a lan with a speed of up to 100mbit, so I can realistically achieve results at the level of 86mbit. Writing to an SSD via USB 2.0 is also not a surprise to me, but I consider 28.6 MB/s a really good result.

So where does the problem occur? Well, when I try to download anything (from the Internet), the download speed does not exceed 1.4 MB/s (i.e. plus or minus 10Mbit per second). Is there any solution for this?

I'm pretty sure there's a bug somewhere, but I can't find it.

### Steps to reproduce the behaviour

Downloading a file with wget:
```
root@rpi:~# wget https://speed.hetzner.de/1GB.bin
2023-01-14 08:55:55 (1.49 MB/s) - ‘1GB.bin’ saved [1048576000/1048576000]
```

And downloading the same file using axel:
```
root@rpi:~# axel -n 5 https://speed.hetzner.de/1GB.bin
Downloaded 1000 Megabyte(s) in 3:01 minute(s). (5638.71 KB/s)
```

Almost 5x faster!

root@rpi:~# speedtest
```
Idle Latency: 4.42 ms (jitter: 0.14ms, low: 4.31ms, high: 4.53ms)
Download: 86.27 Mbps (data used: 46.5 MB)
7.94 ms (jitter: 2.61ms, low: 4.46ms, high: 27.04ms)
Upload: 88.38 Mbps (data used: 45.4 MB)
10.48 ms (jitter: 2.84ms, low: 5.98ms, high: 29.98ms)
Packet Loss: 0.0%
```

Any ideas?

### Device (s)

Raspberry Pi 3 Mod. B

### System

System Information
------------------

Raspberry Pi 3 Model B Rev 1.2
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"

Raspberry Pi reference 2022-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 005a8c73b05a2cab394073150208bf4f069e861a, stage2

Linux rpi 5.15.76-v8+ #1597 SMP PREEMPT Fri Nov 4 12:16:41 GMT 2022 aarch64 GNU/Linux
Revision	: a02082
Serial		: 000000009507a5a9
Model		: Raspberry Pi 3 Model B Rev 1.2
Throttled flag  : throttled=0x0
Camera          : supported=1 detected=1, libcamera interfaces=0

Videocore information
---------------------

Oct 26 2022 11:11:31 
Copyright (c) 2012 Broadcom
version c72ad6b26ff40c91ef776b847436094ee63fabee (clean) (release) (start_x)

alloc failures:     0
compactions:        0
[... 561 lines cut ...]
[   18.600293] 8021q: 802.1Q VLAN Support v1.8
[   18.989539] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   19.029826] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   19.107848] cfg80211: loaded regulatory.db is malformed or signature is missing/invalid
[   19.387374] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k FS
[   19.667337] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   19.822384] smsc95xx 1-1.1:1.0 eth0: Link is Down
[   22.746109] ax88179_178a 1-1.3:1.0 eth1: ax88179 - Link status is: 1
[   22.775435] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   24.005776] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net.ipv6.neigh.eth1.retrans_time - use net.ipv6.neigh.eth1.retrans_time_ms instead
[   25.430722] Bluetooth: Core ver 2.22
[   25.430849] NET: Registered PF_BLUETOOTH protocol family
[   25.430858] Bluetooth: HCI device and connection manager initialized
[   25.430887] Bluetooth: HCI socket layer initialized
[   25.430913] Bluetooth: L2CAP socket layer initialized
[   25.430945] Bluetooth: SCO socket layer initialized
[   25.465647] Bluetooth: HCI UART driver ver 2.3
[   25.465678] Bluetooth: HCI UART protocol H4 registered
[   25.465770] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   25.466041] Bluetooth: HCI UART protocol Broadcom registered
[   25.777884] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   25.777910] Bluetooth: BNEP filters: protocol multicast
[   25.777934] Bluetooth: BNEP socket layer initialized
[   25.801466] NET: Registered PF_ALG protocol family
[   33.763658] cam-dummy-reg: disabling

pelwell, 2023-01-16:
Running the following on a 3B (to take the local storage speed out of the equation) gives me an average download speed of 10.2M (better than I expected):
```
pi@raspberrypi:~$ curl https://speed.hetzner.de/1GB.bin >/dev/null
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1000M  100 1000M    0     0  10.2M      0  0:01:37  0:01:37 --:--:-- 11.0M
```
I did find bandwidth from the download site to be variable, getting only 7MB/s when I tried on server here.

Looking at your kernel log, the following entries are relevant: 
```
[ 3.816752] smsc95xx v2.0.0
[ 3.916672] SMSC LAN8700 usb-001:003:01: attached PHY driver (mii_bus:phy_addr=usb-001:003:01, irq=POLL)
[ 3.924723] smsc95xx 1-1.1:1.0 eth0: register 'smsc95xx' at usb-3f980000.usb-1.1, smsc95xx USB 2.0 Ethernet, m.m.m.m
...
[ 16.244351] ax88179_178a 1-1.3:1.0 eth1: register 'ax88179_178a' at usb-3f980000.usb-1.3, ASIX AX88179 USB 3.0 Gigabit Ethernet, m.m.m.m
[ 16.244654] usbcore: registered new interface driver ax88179_178a
...
[ 19.667337] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[ 19.822384] smsc95xx 1-1.1:1.0 eth0: Link is Down
[ 22.746109] ax88179_178a 1-1.3:1.0 eth1: ax88179 - Link status is: 1
[ 22.775435] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
```
Observe that the onboard Ethernet is Down (not usable), while eth1 - an external Gigabit Ethernet - is up, which suggests that you aren't using the onboard interface at all.

JG24, 2023-01-22:
More trials and tests:

`root@rpi:~# wget https://speed.hetzner.de/100MB.bin > /dev/null`
Avg. speed is: **1.33MB/s**

`root@rpi:~# axel -n 5 https://speed.hetzner.de/100MB.bin`
Avg. speed is: **5.8MB/s**

Speedtest result: https://www.speedtest.net/result/c/e6a2b14a-139c-4de3-8f0f-27d4403b4879

And LAN download:
`wget http://192.168.1.81/test/test.tar.gz`
--2023-01-18 18:28:21-- http://192.168.1.81/test/test.tar.gz
2023-01-18 18:28:41 (**10.5 MB/s**) - ‘test.tar.gz’ saved [222223484/222223484]

Wget and curl are the slowest. Axel makes world records. Speedtest shows exactly the same result that we can get by downloading a file on the internal network.

I only dream about 10MB/s. I don't understand where these speed differences between downloading from the local network and the Internet come from. My internet connection is 980Mbit. Since the Raspberry can download from the local network at a speed of 10.5 MB/s, it should download from the Internet at exactly the same speed.

Here are the tests done on an old Intel(R) Core(TM)2 Duo CPU E8200 @ 2.66GHz (is standing next to Raspberry PI):

>    Speedtest by Ookla
> 
>      Server: NSYS s.r.o. - Náchod (id = 38880)
>         ISP: Netia SA
>     Latency:    12.57 ms   (0.07 ms jitter)
>    Download:   939.88 Mbps (data used: 923.8 MB)
>      Upload:   279.95 Mbps (data used: 187.9 MB)
> Packet Loss:     0.0%
>  Result URL: https://www.speedtest.net/result/c/3c76eb13-189d-4997-bc6a-86285ca6e95c

And download test:

> root@M58P:~# wget https://speed.hetzner.de/1GB.bin
> --2023-01-22 08:18:49--  https://speed.hetzner.de/1GB.bin
> Translacja speed.hetzner.de (speed.hetzner.de)... 88.198.248.254, 2a01:4f8:0:59ed::2
> Łączenie się z speed.hetzner.de (speed.hetzner.de)|88.198.248.254|:443... połączono.
> Żądanie HTTP wysłano, oczekiwanie na odpowiedź... 200 OK
> Długość: 1048576000 (1000M) [application/octet-stream]
> Zapis do: `1GB.bin'
> 
> 1GB.bin                                         100%[==================================>]   1000M  60,7MB/s    w 19s
> 
> 2023-01-22 08:19:08 (52,1 MB/s) - zapisano `1GB.bin' [1048576000/1048576000]

So internet download speed is absolutely no problem here.

Fresh dmesg: https://pastebin.com/raw/DSFQeCMM

JG24, 2023-01-25:
Test's on Ubuntu:

> root@rpi:/# wget https://speed.hetzner.de/1GB.bin -O /dev/null
> --2023-01-25 20:02:19--  https://speed.hetzner.de/1GB.bin
> Resolving speed.hetzner.de (speed.hetzner.de)... 88.198.248.254, 2a01:4f8:0:59ed::2
> Connecting to speed.hetzner.de (speed.hetzner.de)|88.198.248.254|:443... connected.
> HTTP request sent, awaiting response... 200 OK
> Length: 1048576000 (1000M) [application/octet-stream]
> Saving to: ‘/dev/null’
> 
> /dev/null                                       100%[===============================================>]   1000M  6.00MB/s    in 2m 48s
> 
> 2023-01-25 20:05:07 (**5.96 MB/s**) - ‘/dev/null’ saved [1048576000/1048576000]

Something is really wrong with the Raspberry Pi OS.

popcornmix, 2023-01-26:
It looks like you are comparing (RPiOS):
```
root@rpi:~# wget https://speed.hetzner.de/100MB.bin > /dev/null
Avg. speed is: 1.33MB/s
```
to (Ubuntu)
```
root@rpi:/# wget https://speed.hetzner.de/1GB.bin -O /dev/null
Avg. speed is: 5.96MB/s
```
The first will be writing file to local directory, and the second discarding the data.
It's best to compare like with like.

pelwell, 2023-01-26:
Well spotted. The important observation is that you turned my request to run `curl https://speed.hetzner.de/1GB.bin >/dev/null` into `wget https://speed.hetzner.de/100MB.bin > /dev/null`. Aside from the size of the file, the main difference between these two commands is that `curl` writes to stdout by default while `wget` creates a file with an appropriate name - the redirection to `/dev/null` does nothing useful here.

JG24, 2023-01-26:
Oh sorry, my oversight. Of course, this does not change the test results in any way, but to be 100% accurate, I am pasting the measurement made a moment ago.

Raspberry PI OS:

> root@rpi:/# curl https://speed.hetzner.de/100MB.bin >/dev/null
>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>                                  Dload  Upload   Total   Spent    Left  Speed
> 100  100M  100  100M    0     0  2007k      0  0:00:51  0:00:51 --:--:-- 1666k

Ubuntu for Raspberry PI (Linux rpi 5.19.0-1004-raspi #10-Ubuntu SMP PREEMPT Mon Oct 17 11:31:26 UTC 2022 armv7l armv7l armv7l GNU/Linux
):

> root@rpi:/# curl https://speed.hetzner.de/100MB.bin >/dev/null
>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>                                  Dload  Upload   Total   Spent    Left  Speed
> 100  100M  100  100M    0     0  5687k      0  0:00:18  0:00:18 --:--:-- 6258k

popcornmix, 2023-01-26:
Just tried on Pi3B and got:
```
pi@pi3:~ $ curl https://speed.hetzner.de/100MB.bin >/dev/null
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  100M  100  100M    0     0  10.0M      0  0:00:09  0:00:09 --:--:-- 10.4M
pi@pi3:~ $ uname -a
Linux pi3 5.15.84-v7+ #1226 SMP Fri Jan 13 15:36:35 GMT 2023 armv7l GNU/Linux
```

JG24, 2023-01-26:
My linux version is:

root@rpi:/# uname -a
Linux rpi 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux

popcornmix, 2023-01-27:
No real change with 64-bit OS on Pi3:
```
pi@pi3:~ $ curl https://speed.hetzner.de/100MB.bin >/dev/null
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  100M  100  100M    0     0  10.3M      0  0:00:09  0:00:09 --:--:-- 10.8M
pi@pi3:~ $ uname -a
Linux pi3 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux
```

pelwell, 2023-01-27:
My tests also show the expected performance, suggesting that there is something different about your environment. To eliminate some more differences, please install a clean vanilla RPi OS to a spare SD card, boot it with no external USB hardware, and try your performance test again.

JG24, 2023-01-27:
Ok, clean system, Rasperry Pi 3b with sd card and with sd card + mounted sdd:

> root@rpi:/# curl https://speed.hetzner.de/100MB.bin >/dev/null
>   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
>                                  Dload  Upload   Total   Spent    Left  Speed
> 100  100M  100  100M    0     0  10.8M      0  0:00:09  0:00:09 --:--:-- 11.1M

Same system, same command, boot from USB SSD and instead 11MB/s I have 1666k :(

pelwell, 2023-01-27:
The Ethernet interface on the 3B shares the same USB controller as the external USB ports, and it is an interface that places a significant interrupt load on the CPU. My theory is that the servicing of the SSD - probably as the result of file activity on the root FS - is stealing bandwidth (USB and CPU) from the Ethernet interface.

JG24, 2023-01-27:
Ok, another test, clean Ubuntu on SSD (Linux rpi 5.19.0-1004-raspi #10-Ubuntu SMP PREEMPT Mon Oct 17 11:31:26 UTC 2022 armv7l armv7l armv7l GNU/Linux):

root@rpi:/# curl https://speed.hetzner.de/100MB.bin >/dev/null
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  100M  100  100M    0     0  5449k      0  0:00:18  0:00:18 --:--:-- 6194k

Only Raspberry Pi OS is so slow.

> root@rpi:/# lsusb
> Bus 001 Device 004: ID 125f:a35a A-DATA Technology Co., Ltd. HD650
> Bus 001 Device 003: ID 0424:ec00 Microchip Technology, Inc. (formerly SMSC) SMSC9512/9514 Fast Ethernet Adapter
> Bus 001 Device 002: ID 0424:9514 Microchip Technology, Inc. (formerly SMSC) SMC9514 Hub
> Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

> root@rpi:/# cat /etc/fstab
> LABEL=writable  /       ext4    discard,errors=remount-ro       0 1
> LABEL=system-boot       /boot/firmware  vfat    defaults        0       1

pelwell, 2023-01-27:
Does adding the following to config txt make a difference?
```
dtoverlay=dwc2,dr_mode=host
```

JG24, 2023-01-28:
The difference is that Raspberry does not boot at all ;)

pelwell, 2023-01-30:
> The difference is that Raspberry does not boot at all ;

I find that surprising - it works fine for me, although since you are running a 64-bit kernel it's not actually going to do anything useful.

So far it's just you reporting this issue, and you have a workaround (run Ubuntu), so it's going on the back burner.

JG24, 2023-01-30:
Perhaps only few people use Raspberry Pi 3b + SSD + Raspberry Pi OS 64 at the same time. Or maybe few people pay attention to the speed of downloading files, assuming that it will be slow anyway.

All I can do is wait for a solution.

pelwell, 2023-01-30:
You might get some useful information from iostat:
```
$ sudo apt install sysstat
$ sudo iostat /dev/sda 5
```
This will show IO statistics for the indicated device every 5 seconds.

1. How busy is your SSD when you boot from it?
2. Does downloading the test file change the stats significantly? If so, perhaps there is some logging enabled somewhere that is  keeping the SSD active.
3 & 4. As above, but booting from SD and only mounting the SSD.

pelwell, 2023-01-31:
This is possibly a red herring, but are you running headless (i.e. no monitor attached)? And booting to desktop or CLI? I ask because I saw an issue yesterday where a headless Pi was keeping the SD card busy because the lightdm login process was unhappy. Using raspi-config to configure the system to boot to console CLI solved the problem.

seamusdemora, 2023-02-01:
FWIW:

I can make a more formal submission of this issue if that's helpful, but I thought I'd do this first:

My RPi 4B has recently developed a debilitating reduction in WiFi bandwidth/download speed (haven't tested over Ethernet). This occurred *approximately*  the same time as my last EEPROM update - which included a VL805 firmware update (I follow the *critical* stream). 

During `apt update` on my RPi 4B, I am now seeing this: 
> Fetched 11.4 MB in 7min 12s (26.4 kB/s)

On my RPi 3B+, I see this:
>Fetched 25.9 MB in 1min 11s (366 kB/s)  

The 4B still runs *buster*, the 3B+ runs *buster*. Both run the *Lite* distro with no routine tasks that have heavy network utilization. They both connect to the same WiFi AP. I realize there are numerous variables, but over years, I've never seen the 4B show such sluggish network connections. These aren't one-time measurements; they've been consistent over approximately a week now. 

FWIW:
```bash
$ cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

<<snip>>

Hardware	: BCM2711
Revision	: b03111
Serial		: 100000006cce8fc1
Model		: Raspberry Pi 4 Model B Rev 1.1
```  

One odd thing is this: I'm not sure what vl805 firmware version I have. I get different indications. After updating with the `-a` option, I see this:

```bash
$ rpi-eeprom-update
BOOTLOADER: up to date
   CURRENT: Wed 11 Jan 17:40:52 UTC 2023 (1673458852)
    LATEST: Wed 11 Jan 17:40:52 UTC 2023 (1673458852)
   RELEASE: stable (/lib/firmware/raspberrypi/bootloader/stable)
            Use raspi-config to change the release.

  VL805_FW: Dedicated VL805 EEPROM
     VL805: version unknown. Try sudo rpi-eeprom-update
   CURRENT:
    LATEST: 000138c0
``` 

If I then try to update with the `-A vl805` option, I get a different report: 

```bash
$ sudo rpi-eeprom-update -A vl805
BOOTLOADER: up to date
   CURRENT: Wed 11 Jan 17:40:52 UTC 2023 (1673458852)
    LATEST: Wed 11 Jan 17:40:52 UTC 2023 (1673458852)
   RELEASE: stable (/lib/firmware/raspberrypi/bootloader/stable)
            Use raspi-config to change the release.

  VL805_FW: Dedicated VL805 EEPROM
     VL805: up to date
   CURRENT: 000138c0
    LATEST: 000138c0
``` 

***BUT*** after a reboot, `rpi-eeprom-update` once again reports `version unknown`. 

????? Any ideas or recommendations appreciated.

pelwell, 2023-02-01:
You are posting about WiFi performance and VL805 firmware versions on a 4B in a thread about Ethernet performance on a 3B. Well done.

drankinatty, 2023-10-24:
Strange similar slow download from raspberrypi.org on 3b over ethernet, e.g.

```none
Get:1 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:3 http://archive.raspberrypi.org/debian buster/main armhf Packages [400 kB]
Get:4 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Fetched 13.5 MB in 2min 59s (75.1 kB/s)
```

This is over a 300Mbs connection. Related or is the resolver for raspberrypi.org on the fritz?