raspberrypi/linux #5324: RPi 4B - 6.1.y/6.2.y branch: thermal sensor isn't recognised

puchu, 2023-01-16:
### Describe the bug

up to the 6.0.y branch, it's possible to access the cpu thermal sensor value under /sys/devices/virtual/thermal/thermal_zone0/hwmon0/temp1_input, starting with 6.1.y, the thermal sensor device isn't recognised anymore

### Steps to reproduce the behaviour

create 6.1.y/6.2.y branch kernel config with "make bcm2711_defconfig" and compile

### Device (s)

Raspberry Pi 4 Mod. B

### System

.

pelwell, 2023-01-17:
Try this:
```
$ sudo rpi-update next
$ sudo reboot
...
$ uname -a
$ grep -s . /sys/devices/virtual/thermal/thermal_zone0/hwmon0/*
```

puchu, 2023-01-17:
with the 6.0 branch:
# uname -a
Linux foo 6.0.19-v8+ #175 SMP Sun Jan 15 15:21:08 CET 2023 aarch64 GNU/Linux
# grep -s . /sys/devices/virtual/thermal/thermal_zone0/hwmon0/*
/sys/devices/virtual/thermal/thermal_zone0/hwmon0/name:cpu_thermal
/sys/devices/virtual/thermal/thermal_zone0/hwmon0/temp1_input:43329

with the 6.1 branch:
# uname -a
Linux foo 6.1.6-v8+ #2 SMP PREEMPT Tue Jan 17 13:05:51 CET 2023 aarch64 GNU/Linux
# grep -s . /sys/devices/virtual/thermal/thermal_zone0/hwmon0/*
#

puchu, 2023-01-17:
with the 6.0 branch:
$ uname -a
Linux foo 6.0.19-v8+ https://github.com/raspberrypi/linux/issues/175 SMP Sun Jan 15 15:21:08 CET 2023 aarch64 GNU/Linux
$ grep -s . /sys/devices/virtual/thermal/thermal_zone0/hwmon0/*
/sys/devices/virtual/thermal/thermal_zone0/hwmon0/name:cpu_thermal
/sys/devices/virtual/thermal/thermal_zone0/hwmon0/temp1_input:43329

with the 6.1 branch:
$ uname -a
Linux foo 6.1.6-v8+ https://github.com/raspberrypi/linux/issues/2 SMP PREEMPT Tue Jan 17 13:05:51 CET 2023 aarch64 GNU/Linux
$ grep -s . /sys/devices/virtual/thermal/thermal_zone0/hwmon0/*
$

pelwell, 2023-01-17:
Have you done what I asked you to and run `sudo rpi-update next`?
```
pi@raspberrypi:~$ grep -s . /sys/devices/virtual/thermal/thermal_zone0/hwmon0/*
/sys/devices/virtual/thermal/thermal_zone0/hwmon0/name:cpu_thermal
/sys/devices/virtual/thermal/thermal_zone0/hwmon0/temp1_crit:110000
/sys/devices/virtual/thermal/thermal_zone0/hwmon0/temp1_input:55991
pi@raspberrypi:~$ uname -a
Linux raspberrypi 6.1.6-v8+ #1618 SMP PREEMPT Mon Jan 16 16:35:56 GMT 2023 aarch64 GNU/Linux
```

puchu, 2023-01-17:
i didn't use the script as it doesn't work on my distro, but i switched to the next firmware branch and now it works.

Thanks.

Is there a timeline when next will become stable?

puchu, 2023-01-17:
i couldn't use the script as it doesn't work on my distro, but i switched to the "next" firmware branch and now it works.

Thanks.

Is there a timeline when the "next" branch will become "stable"?

pelwell, 2023-01-17:
If `rpi-update next` works for you but not your own build then there is something wrong with your build or installation.

There is no timeline for changing our `master` branch yet, let alone our `stable` branch. It's notable that the kernel devs haven't selected the next LTS branch - I was expecting 6.1 to be one, but that's looking less and less likely. We won't be choosing our next master kernel until that happens.

puchu, 2023-01-17:
rpi-update didn't work because of the distro i use but i read the code and switched the firmware to the "next" branch which fixed the bug for me.

thank you very much!

closed 2023-01-17 by puchu