raspberrypi/linux #5325: Please add support for MPU9250 sensor?

Jibun-no-Kage, 2023-01-17:


pelwell, 2023-01-17:
I've added `mpu6050` and `mpu9250` as parameters to the `i2c-sensor` overlay, deprecating the standalone `mpu6050` overlay (although it hasn't been deleted yet). Let me know if you have any problems with it.

Jibun-no-Kage, 2023-01-17:
Will do!  I tested the old overlay last night just to get familiar with the MPU6050.  When is the change likely to be available?  Is it visible via rpi-update?  Or is there a repo update pending soon?

pelwell, 2023-01-17:
I expect there to be an rpi-update release by the end of the week, but it's not guaranteed.

You can always compile the overlays yourself:
```
$ sudo apt install git bc bison flex libssl-dev make
$ git clone --depth=1 https://github.com/raspberrypi/linux
$ cd linux
$ make ARCH=arm bcm2711_defconfig
$ make ARCH=arm dtbs
$ sudo cp arch/arm/boot/dts/overlays/i2c-sensor.dtbo /boot/overlays
```
It would normally be a simple matter of running `dtc`, but the overlay uses `#include` which requires pre-processing with `cpp`.

popcornmix, 2023-01-18:
rpi-update kernel should contain support for this

Jibun-no-Kage, 2023-01-18:
Great, thanks!

closed 2023-01-18 by Jibun-no-Kage

Jibun-no-Kage, 2023-01-20:
Testing the mpu6050, while awaiting my 9250, using the new overlay definition, ran into something odd.  I can read the 'matrix' registers, the 'sampling' registers, but none of the 'raw' registers.  Moreover, if I move the mpu6050, I never see the 'matrix' values change. 

\# cat /boot/config.txt | grep mpu
dtoverlay=i2c-sensor,mpu6050

\# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

\# cat /sys/bus/i2c/devices/1-0068/iio:device0/*matrix
0, 0, 0; 0, 0, 0; 0, 0, 0
1, 0, 0; 0, 1, 0; 0, 0, 1
1, 0, 0; 0, 1, 0; 0, 0, 1
0, 0, 0; 0, 0, 0; 0, 0, 0

\# cat /sys/bus/i2c/devices/1-0068/iio:device0/*raw
cat: '/sys/bus/i2c/devices/1-0068/iio:device0/in_accel_x_raw': Invalid argument
cat: '/sys/bus/i2c/devices/1-0068/iio:device0/in_accel_y_raw': Invalid argument
cat: '/sys/bus/i2c/devices/1-0068/iio:device0/in_accel_z_raw': Invalid argument
cat: '/sys/bus/i2c/devices/1-0068/iio:device0/in_anglvel_x_raw': Invalid argument
cat: '/sys/bus/i2c/devices/1-0068/iio:device0/in_anglvel_y_raw': Invalid argument
cat: '/sys/bus/i2c/devices/1-0068/iio:device0/in_anglvel_z_raw': Invalid argument
cat: '/sys/bus/i2c/devices/1-0068/iio:device0/in_temp_raw': Invalid argument

Under the older dtoverlay I could see this working, i.e. I could read the raw values....

\# cat /boot/config.txt | grep mpu
\#dtoverlay=i2c-sensor,mpu6050
dtoverlay=mpu6050

\# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

\# cat /sys/bus/i2c/devices/1-0068/iio:device0/*raw
8362
6830
13872
-27
-8
14
-3810

\# cat /sys/bus/i2c/devices/1-0068/iio:device0/in_accel_z_raw
5170
\# cat /sys/bus/i2c/devices/1-0068/iio:device0/in_accel_z_raw
5138
\# cat /sys/bus/i2c/devices/1-0068/iio:device0/in_accel_z_raw
5110
\# cat /sys/bus/i2c/devices/1-0068/iio:device0/in_accel_z_raw
5164
\# cat /sys/bus/i2c/devices/1-0068/iio:device0/in_accel_z_raw
4950
\# cat /sys/bus/i2c/devices/1-0068/iio:device0/in_accel_z_raw
5026
\# cat /sys/bus/i2c/devices/1-0068/iio:device0/in_accel_z_raw
5070
\# cat /sys/bus/i2c/devices/1-0068/iio:device0/in_accel_z_raw
4988

The only time via the older dtoverlay, I cannot read the raw values, is if I try to read the registers too fast.  Otherwise they update as expected.  But the newer dtoverlay, I can't seem to read the raw values at all.  I consistently get the invalid argument error.

Still testing, but something odd is happening.

pelwell, 2023-01-20:
The only difference between the two overlays should be the polarity of the interrupt line, but the driver appears to adjust the hardware behaviour to match the declaration of the interrupt GPIO so it _shouldn't_ matter.

Can you try this build instead?
https://drive.google.com/file/d/1eWeA3oRhVTrYI1Y57HLoXzy_dhXmQOd8/view?usp=share_link

Jibun-no-Kage, 2023-01-20:
Actually, finally got the older dtoverly to error out as well, so then the behavior was similar.  When I first tested, the older method everything was fine.  Then I moved to the new overlay and things got weird.  Then finally, the weird issues happened with the older method.  This implied hardware not software issue at that point.  So, I pulled  the Adafruit driver, and that had a few errors as well, not as many as just driving from bash but enough.  So stripped down the circuit and rebuilt it.  Sure enough things stabilized.   I just finished testing python, so next the overlays.