raspberrypi/linux #5332: Newer version(s) of gstreamer causes issues w/ bcm2835-codec

bluestang2006, 2023-01-23:
### Describe the bug

I was advised to post this specific issue here after I posted initially in the [forums](https://forums.raspberrypi.com/viewtopic.php?t=346223):

Details of the initial bug report filed w/ gstreamer [here](https://gitlab.freedesktop.org/gstreamer/gstreamer/-/issues/1729) because I narrowed this down to a specific commit that causes this issue.

### Steps to reproduce the behaviour

A sample of the app can be downloaded from this [post](https://gitlab.freedesktop.org/gstreamer/gstreamer/-/issues/1729#note_1735417) w/ quick steps to produce.

Need to use any gstreamer tagged version 1.21.3 and up. The branch main is easiest.

meson cmd to build:
```
meson setup --prefix=/usr -Dgpl=enabled -Dugly=disabled -Drtsp_server=disabled -Dges=disabled -Ddevtools=disabled -Dqt5=disabled -Dtls=disabled -Dgst-plugins-good:rpicamsrc=disabled -Dgst-plugins-bad:opencv=disabled -Dgst-plugins-bad:webrtc=disabled -Dgst-plugins-bad:webrtcdsp=disabled -Dgst-plugins-bad:v4l2codecs=enabled -Dgst-plugins-base:gl_winsys=wayland,gbm build-test
```

The application is being ran in the TTY, so no X11/Wayland.

### Device (s)

Raspberry Pi 4 Mod. B

### System

```
Raspberry Pi reference 2022-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 005a8c73b05a2cab394073150208bf4f069e861a, stage4
```
```
Jan 18 2023 12:12:12
Copyright (c) 2012 Broadcom
version 658f02cc8edcb68a568273f05d2b6ceede181e15 (clean) (release) (start)
```
```
Linux raspberrypi 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux
```

### Logs

Kernel Log:
```
[ 1549.252779] ------------[ cut here ]------------
[ 1549.252787] WARNING: CPU: 3 PID: 15781 at drivers/media/common/videobuf2/videobuf2-core.c:2035 __vb2_queue_cancel+0x1b8/0x220 [videobuf2_common]
[ 1549.252855] Modules linked in: xpad ff_memless joydev rfcomm cmac algif_hash aes_arm64 aes_generic algif_skcipher af_alg bnep hci_uart btbcm bluetooth ecdh_generic ecc libaes 8021q garp stp llc sg vc4 snd_soc_hdmi_codec drm_display_helper cec drm_dma_helper brcmfmac rpivid_hevc(C) drm_kms_helper bcm2835_codec(C) brcmutil snd_soc_core bcm2835_isp(C) v4l2_mem2mem v3d cfg80211 videobuf2_dma_contig snd_bcm2835(C) snd_compress bcm2835_v4l2(C) snd_pcm_dmaengine bcm2835_mmal_vchiq(C) gpu_sched snd_pcm videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 rfkill videobuf2_common drm_shmem_helper raspberrypi_hwmon i2c_brcmstb videodev vc_sm_cma(C) snd_timer mc snd syscopyarea sysfillrect sysimgblt fb_sys_fops uio_pdrv_genirq uio nvmem_rmem drm i2c_dev fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
[ 1549.253219] CPU: 3 PID: 15781 Comm: retrofe Tainted: G         C         6.1.7-v8+ #1619
[ 1549.253234] Hardware name: Raspberry Pi 4 Model B Rev 1.2 (DT)
[ 1549.253242] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[ 1549.253257] pc : __vb2_queue_cancel+0x1b8/0x220 [videobuf2_common]
[ 1549.253297] lr : __vb2_queue_cancel+0x3c/0x220 [videobuf2_common]
[ 1549.253335] sp : ffffffc008b0bae0
[ 1549.253342] x29: ffffffc008b0bae0 x28: ffffffec231d2938 x27: 0000000000000000
[ 1549.253366] x26: 0000000040045613 x25: ffffff80431b5d28 x24: 0000000000000000
[ 1549.253388] x23: ffffff8041ffeb00 x22: ffffffec310aa758 x21: ffffff804303c298
[ 1549.253408] x20: ffffff804303c298 x19: ffffff804303c298 x18: ffffff8049840028
[ 1549.253429] x17: 756f656d6954203a x16: ffffffec30498f50 x15: ffffff80ab8e30c8
[ 1549.253450] x14: 0000000000000000 x13: ffffff8049840028 x12: ffffff80ab8e2fe8
[ 1549.253471] x11: 0000000000000000 x10: ffffff8049840030 x9 : ffffffec304ea67c
[ 1549.253491] x8 : 000000000010000f x7 : ffffffc008b0c000 x6 : ffffffc008b08000
[ 1549.253511] x5 : ffffffc008b0c000 x4 : 0000000000000000 x3 : ffffffec31559000
[ 1549.253531] x2 : ffffffec315590f0 x1 : 0000000100000000 x0 : 0000000000000002
[ 1549.253552] Call trace:
[ 1549.253558]  __vb2_queue_cancel+0x1b8/0x220 [videobuf2_common]
[ 1549.253599]  vb2_core_streamoff+0x2c/0xc8 [videobuf2_common]
[ 1549.253638]  vb2_streamoff+0x24/0x78 [videobuf2_v4l2]
[ 1549.253673]  v4l2_m2m_streamoff+0x44/0x130 [v4l2_mem2mem]
[ 1549.253715]  v4l2_m2m_ioctl_streamoff+0x20/0x30 [v4l2_mem2mem]
[ 1549.253750]  v4l_streamoff+0x2c/0x38 [videodev]
[ 1549.253850]  __video_do_ioctl+0x184/0x3d8 [videodev]
[ 1549.253938]  video_usercopy+0x2d8/0x740 [videodev]
[... 64 lines cut ...]
[ 1563.876784] bcm2835_mmal_vchiq: vchiq_mmal_component_init: failed to create component -62 (Not enough GPU mem?)
[ 1563.876804] bcm2835-codec bcm2835-codec: bcm2835_codec_create_component: failed to create component ril.video_decode
[ 1567.716791] bcm2835_mmal_vchiq: timed out waiting for sync completion
[ 1569.188623] usb 2-2: USB disconnect, device number 6
[ 1569.241058] usb 1-1.2: USB disconnect, device number 10
[ 1569.241102] usb 1-1.2.1: USB disconnect, device number 11
[ 1569.242282] usb 1-1.2.3: USB disconnect, device number 12
[ 1569.242314] usb 1-1.2.3.1: USB disconnect, device number 14
[ 1569.446305] usb 1-1.2.3.3: USB disconnect, device number 15
[ 1569.734886] usb 1-1.2.3.4: USB disconnect, device number 16
[ 1569.735134] xpad 1-1.2.3.4:1.0: xpad_try_sending_next_out_packet - usb_submit_urb failed with result -19
[ 1569.795965] usb 1-1.2.5: USB disconnect, device number 13
[ 1570.788796] bcm2835_mmal_vchiq: timed out waiting for sync completion
[ 1571.049220] usb 2-2: new SuperSpeed USB device number 7 using xhci_hcd
[ 1571.108313] usb 2-2: New USB device found, idVendor=2109, idProduct=0817, bcdDevice= 6.34
[ 1571.108329] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1571.108336] usb 2-2: Product: USB3.0 Hub
[ 1571.108342] usb 2-2: Manufacturer: VIA Labs, Inc.
[ 1571.108348] usb 2-2: SerialNumber: 000000000
[ 1571.110631] hub 2-2:1.0: USB hub found
[ 1571.110904] hub 2-2:1.0: 4 ports detected
[ 1573.860827] bcm2835_mmal_vchiq: timed out waiting for sync completion
[ 1576.932893] bcm2835_mmal_vchiq: timed out waiting for sync completion
[ 1580.004927] bcm2835_mmal_vchiq: timed out waiting for sync completion
```

bluestang2006, 2023-01-25:
Closing issue as the bug is in gstreamer and PR has been submitted.

closed 2023-01-25 by bluestang2006

nicolasb565, 2023-12-20:
@bluestang2006 What is the status of the PR in gstreamer? I have a similar problem with gstreamr 1.20.7 and linux kernel 5.15.
Thanks!

6by9, 2023-12-20:
The PR was https://gitlab.freedesktop.org/gstreamer/gstreamer/-/merge_requests/3786, and backported to 1.20 in https://gitlab.freedesktop.org/gstreamer/gstreamer/-/merge_requests/3795.
It looks like it should have been in 1.20.6.

nicolasb565, 2023-12-20:
@6by9 thanks, look like I should have that fix so something else is going on.