raspberrypi/linux #5334: Please add or extend dtoverlay sht3x to include newer sht4x senors?

Jibun-no-Kage, 2023-01-24:
Please add or extend dtoverlay sht3x to include newer sht4x.  Just got the newer sht4x sensors, and tried the sht3x support, does not seem to work, which seems a bit odd since address 0x44 is used for both.  Willing to help test of course.  Thanks.

\# cat /boot/config.txt | grep dtoverlay
dtoverlay=i2c-sensor,sht3x

\# lsmod | grep sht
sht3x                  16384  0
crc8                   16384  1 sht3x

\# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- 44 -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

pelwell, 2023-01-24:
Issues or requests related to specific overlays should be opened in this repo (raspberrypi/linux), not the firmware.

Your request should be satisfied by https://github.com/raspberrypi/linux/commit/c8c99191e1419062ac8b668956d19e788865912a and https://github.com/raspberrypi/linux/commit/636cdc62014a04ac1673286b11a063dfdaa09bbe.

Jibun-no-Kage, 2023-01-24:
Oh, that is right.  Sorry about that.  Thanks for expanding the sensor support, really appreciate it.