raspberrypi/linux #5338: spi-nand kernel module is missing

mkshevetskiy, 2023-01-26:
### Describe the bug

A lot of spi-nand flashes is supported linux kernel. It will be very helpful to add a possibility to read/write/erase such flashes using standard linux support. Unfortunately spi-nand support is not enabled in raspberry linux kernel. 

example: recovery bootloader on the router board

- Raspberry Pi configuration

    Recompile raspberry pi linux kernel to enable spi-nand driver (linux-5.10 or above is required). Make sure spinand, mtd and nandcore are built as modules.
    sudo raspi-config, go to “Interface Options → SPI” and enable spi support
    edit /boot/config.txt file and enable/add the following lines

    dtparam=spi=on
    dtoverlay=anyspi:spi0-0,dev="spi-nand",speed=5000000

    install the following packages: mtd-utils
    reboot board to apply changes

-  Connect spi-nand chip to raspberry board    

    Testpoint   Meaning   Where to connect

3.3V   3.3V power supply   pin 17 of raspberry pi connector
DI   SPI Data In   pin 19 of raspberry pi connector
DO   SPI Data Out   pin 21 of raspberry pi connector
CLK   SPI Clock   pin 23 of raspberry pi connector
GND   Ground   pin 25 of raspberry pi connector
CS#   SPI Chip Select   pin 24 of raspberry pi connector
WP#   Write Protect   short with HOLD# and 3.3V testpoints
HOLD#   Hold   short with WP# and 3.3V testpoints
    

- restart spinand and mtd modules

    rmmod ofpart spinand mtd nandcore
    modprobe spinand
    modprobe mtd

    make sure the flash was detected

    dmesg | tail -n 6
    [  237.055291] spi-nand: probe of spi0.0 failed with error -524
    [  383.913569] spi-id: 00, ef, aa, 21
    [  383.913729] spi-id: ef, aa, 21, 00
    [  383.913878] spi-id: ef, aa, 21, 00
    [  383.913936] spi-nand spi0.0: Winbond SPI NAND was found.
    [  383.913965] spi-nand spi0.0: 128 MiB, block size: 128 KiB, page size: 2048, OOB size: 64

- Recovering bootloader

    Add partitions of your choice using

    mtdpart add /dev/mtd0 <PART_NAME> <START> <SIZE>

    Erase & flash partition

    flash_erase /dev/mtd<NUMBER> 0 0
    nandwrite /dev/mtd<NUMBER> image_file

    Disconnect flash from raspberry board and test it

### Steps to reproduce the behaviour

modprobe spinand

### Device (s)

Raspberry Pi 1 Mod. B+

### System

1) any raspberry OS image
2) any firmware version
3) linux-5.10 and above

pelwell, 2023-01-30:
If you don't mind trying beta software which has had minimal testing, you can install a build of the updated kernel with:
```
$ sudo rpi-update pulls/5339
```

mkshevetskiy, 2023-01-30:
I'll try it when I returns from business trip.
Thanks!

mkshevetskiy, 2023-02-17:
It works! Thank you!

closed 2023-02-17 by mkshevetskiy