raspberrypi/linux #5342: Built-in BTRFS support

solskogen, 2023-02-01:
### Describe the bug

BTRFS is, as of now, enabled as a module. This means that to be able to use BTRFS as your root filesystem you must use initrd ramdisk, and this part is a bit tricky to get to work (but it can be done).  With BTRFS compiled in the kernel, this part would a lot easier. 

### Steps to reproduce the behaviour

.

### Device (s)

Raspberry Pi 4 Mod. B

### System

Linux raspberrypi 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64

uniartisan, 2023-12-11:
I can not agree more, and I wish I could have btrfs built into the kernel when it was built

solskogen, 2023-12-11:
With the newest Raspberry PI OS, using btrfs is quite easy. Closing the issue.

closed 2023-12-11 by solskogen

Fantu, 2023-12-11:
hi, I not tried rapsberry based on debian 12 yet.
rpi kernel now have btrfs built-in or is only faster/easier workaround for kernel/bootloader/firmware changes to debian based that is better?

brian6932, 2023-12-11:
But this isn't an issue with Pi OS, it's an issue with rpi-kernel. I'm on Void Linux, and use this kernel, I had to use an initramfs to have btrfs root.

pelwell, 2023-12-11:
> I had to use an initramfs (on an installation via xchroot to install booster) to have btrfs root.

That is the intended behaviour. Bookworm RPiOS makes that easy, which is what @solskogen was referring to.

brian6932, 2023-12-11:
The issue states that it's a module and not built into the kernel, how's the issue resolved?

XECDesign, 2023-12-11:
Wouldn't it be more appropriate to request that whoever maintains the relevant kernel package in void linux enables the option on their end? I'm not sure it makes sense to enable it in the defconfigs here.