raspberrypi/linux #5343: USB boot system disconnects when connects android device

Ninoh-FOX, 2023-02-01:


timg236, 2023-02-01:
Moved to Linux because this appears to be happening after the Linux kernel was already loaded from USB.

Ninoh-FOX, 2023-02-01:
[  154.384462] usb 1-1.2.4: new high-speed USB device number 6 using xhci_hcd
[  154.504720] usb 1-1.2.4: device descriptor read/64, error -32
[  154.775294] usb 1-1.2.4: New USB device found, idVendor=18d1, idProduct=4ee7, bcdDevice= 4.04
[  154.775327] usb 1-1.2.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  154.775348] usb 1-1.2.4: Product: Redmi Note 9
[  154.775368] usb 1-1.2.4: Manufacturer: Xiaomi
[  154.775385] usb 1-1.2.4: SerialNumber: XXXXXXXXXXX

Ninoh-FOX, 2023-02-01:
I use the official raspberry transformer to power the rpi 400, as for the devices I have a usb 3.1 multiport (05e3:0626 Genesys Logic, Inc. USB3.1 Hub)

This would be the usb log before the system disconnected:

Bus 002 Device 003: ID 05e3:0626 Genesys Logic, Inc. USB3.1 Hub
Bus 002 Device 002: ID 0781:5583 SanDisk Corp. Ultra Fit
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 04d9:0007 Holtek Semiconductor, Inc. Raspberry Pi Internal Keyboard
Bus 001 Device 004: ID 045e:0737 Microsoft Corp. Compact Optical Mouse 500
Bus 001 Device 006: ID 18d1:4ee7 Google Inc. Nexus/Pixel Device (charging + debug)
Bus 001 Device 003: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Ninoh-FOX, 2023-02-01:
this is my /boot/cmdline.txt config:

`console=serial0,115200 console=tty1 root=PARTUUID=da24f968-02 usb-storage.quirks=0781:5583:u rootfstype=ext4 fsck.mode=force fsck.repair=yes rootwait
`

and this my /boot/config.txt

`# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 1800 MHz is the default.
#arm_freq=2000

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
dtparam=i2s=on
dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
#dtparam=audio=on

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Run in 64-bit mode
#arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
#otg_mode=1

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-fkms-v3d,cma-512,audio=off
max_framebuffers=2
hdmi_enable_4kp60=1

[all]
dtoverlay=hifiberry-dac
dtparam=act_led_trigger=actpwr
gpu_mem=128
force_eeprom_read=0
#dtoverlay=i2c1
#dtoverlay=dacberry400
#gpio=26=op,dh
`