raspberrypi/linux #5347: Yet another Kernel Panic

mihaimiculescu, 2023-02-07:
### Describe the bug

This is slightly different from [this one](https://github.com/raspberrypi/linux/issues/3465l)
_Use case:_
A Raspberry Pi4 booting from a SSD USB drive and having also a SD card as fallback, just in case the boot from SSD fails 
_The onset:_
After a brownout (very short power outage) the **boot** partition from the SSD gave up the ghost. It was corrupted beyond repair. The Pi didn't boot anymore.
Had to unplug it and boot from the SD. Plugged the SSD back in after boot.

### Steps to reproduce the behaviour

_The repair attempt:_
Since I didn't boot for a while from the SD card, it asked for installing updates, which I did. **BIG MISTAKE!!** as I understood later.
Then, using Gparted, I deleted the SSD's boot partition, created it back exactly how it was supposed to be, All good.
`ls /media/<username>` correctly lists
**`boot   rootfs`**
`sudo fdisk -l` gives me this output:
```
Device     Boot  Start       End   Sectors   Size Id Type
/dev/sda1  *      8192    532479    524288   256M  c W95 FAT32 (LBA)
/dev/sda2       532480 937701375 937168896 446.9G 83 Linux
```
Then copied the freshly updated content of the SD boot partition to the SSD's boot partition. And then `sudo reboot now`
There it was: Kernel Panic (see picture attached). Notice that the `unknown-block` is `(8,1)` different from `unknown-block(8,7)` as appeared in [the other issue](https://github.com/raspberrypi/linux/issues/3465)
My guess: by updating the SD card and copying the boot content on the SSD, I have now something on the SSD's **boot** partition that is "ahead" of something from the un-updated **root** partition. Hence the Kernel Panic.
[image]

### Device (s)

Raspberry Pi 4 Mod. B

### System

raspinfo
Can't issue the command, since it doesn't boot

pelwell, 2023-02-07:
Do you think the kernel is behaving incorrectly in some way? The "panic" is a sensible response to the lack of a usable root filing system. This is not the place to ask for help with a corrupt installation.

mihaimiculescu, 2023-02-08:
I'm not sure I can answer correctly to your question.
The **root** partition was healthy all the time.
The kernel Panic occurred after re-creating the **boot** partition.

pelwell, 2023-02-08:
If the root fs isn't mountable it's not usable, whether or not it is corrupt.

mihaimiculescu, 2023-02-08:
The SSD's **root** fs was mountable all the time. 
For instance, when booting from the SD, the SSD's **root** partition did mount.

pelwell, 2023-02-08:
The boot partition and contents you have assembled doesn't seem to think it's bootable, and that's what counts.

All of the following refer to the boot partition on the SSD:
1. As mentioned in the other thread, try adding `framebuffer_height=2160` to config.txt.
2. Remove `quiet` from cmdline.txt, and add `loglevel=8` to be sure.
3. Check the `root=` in the cmdline.txt is correct for the SSD.
4. Try booting.

mihaimiculescu, 2023-02-09:
3. was correctly specified. I always pay attention to this in the cmdline.txt

Meanwhile I took another SD card, not updated, that I keep for testing purposes.
Emptied the boot partition of the SSD, copied the content of the boot partition of this SD to the SSD.
It booted from the SSD. Some services failed to start (docker among them), but, hey, back in business!
Afterwards, `sudo apt update` and `sudo apt upgrade` did the trick. After reboot all was perfect. All is up and running.

As a good practice suggestion, maybe it's a good idea to keep a copy of the content of the boot partition in a dedicated folder on the root partition. And always update it by re-copying, after each software update, if the update did affect anything related to the boot partition. I know I'll do that from now on.

Thanks for your patience!

closed 2023-02-09 by pelwell