raspberrypi/linux #5352: CM4: Kernel panic with PI7C9X2G404EV PCIe switch

iluminat23, 2023-02-17:


timg236, 2023-02-17:
Transferred to Linux since the firmware is not involved with PCIe

nbuchwitz, 2023-02-17:
Afaik @6by9 has successfully used the pericom / diodes switch with the CM4. Do you remember if there were any hoops you had to jump? Does your card had the eeprom attached to the switch?

6by9, 2023-02-17:
I've a PI7C9X2G304 1:2 PCIe switch built in to a Hauppauge WinTV QuadHD DVB-T2/C tuner card. That needed ```dtoverlay=pcie-32bit-dma```, but down to CX23888 chips on the other side.

I've also used a ASMedia ASM1184e 1:4 PCIe switch, which had no issues. I've not tested it recently though.

My NAS is a CM4IO running with a Pericom PCIe switch (12d8:2404), VL805 USB3 controller and Marvell 9215 SATA controller. It's still running a self-built 5.10.74-v8 kernel on Buster as I haven't rebuilt it for a while.
```
pi@cm4nas:~ $ dmesg | grep GT/s
[    1.254526] brcm-pcie fd500000.pcie: link up, 2.5 GT/s PCIe x1 (SSC)
[    1.259976] pci 0000:01:00.0: 2.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s PCIe x1 link at 0000:00:00.0 (capable of 4.000 Gb/s with 5.0 GT/s PCIe x1 link)
[    1.274156] pci 0000:04:00.0: 2.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s PCIe x1 link at 0000:00:00.0 (capable of 4.000 Gb/s with 5.0 GT/s PCIe x1 link)
[    1.278425] pci 0000:05:00.0: 2.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s PCIe x1 link at 0000:00:00.0 (capable of 4.000 Gb/s with 5.0 GT/s PCIe x1 link)
```
No issues there.

nbuchwitz, 2023-02-17:
Any chance you could share the kernel and your firmware / eeprom version, so we can check with the diodes/pericom eval board?

6by9, 2023-02-20:
```
pi@cm4nas:~ $ vcgencmd version
Dec  1 2021 15:02:25 
Copyright (c) 2012 Broadcom
version 71bd3109023a0c8575585ba87cbb374d2eeb038f (clean) (release) (start_db)

pi@cm4nas:~ $ vcgencmd bootloader_version
Sep  3 2020 13:11:43
version c305221a6d7e532693cc7ff57fddfc8649def167 (release)
timestamp 1599135103
update-time 0
capabilities 0x00000000

pi@cm4nas:~ $ vcgencmd bootloader_config
[all]
WAKE_ON_GPIO=0
POWER_OFF_ON_HALT=1
DISABLE_HDMI=0
BOOT_ORDER=0xf41

# Example debug configuration
# USB MSD -> Network -> eMMC boot -> Repeat
#[gpio7=0]
#BOOT_ORDER=0xf124
#BOOT_UART=1

pi@cm4nas:~ $ uname -a
Linux cm4nas 5.10.74-v8+ #1 SMP PREEMPT Sun Nov 7 16:34:20 GMT 2021 aarch64 GNU/Linux
```
Kernel config was the base bcm2711_defconfig + CONFIG_SATA_AHCI=m.

Looking at the board, the chip is labelled P17C9X2G 404SLBFDE 1712GT.
Board is labelled ```Name: PCIe1To3 PN:90.PCIE3.V10 Date: 20170503```. Bought off ebay in Sept 2020, but the link is now dead. It looks very similar to https://www.ebay.co.uk/itm/304669598600

It is on a CM4Lite, so I'll see if I get time to make up a fresh SD card and try booting with it.

6by9, 2023-02-21:
Tested on Bullseye 64bit Lite with kernel/firmware as per the Sept image, apt dist-upgrade'd, and rpi-update. Rebooted half a dozen times with each and had no issues.

About the only thing I didn't update was my bootloader, which is still on 
```
Sep  3 2020 13:11:43
version c305221a6d7e532693cc7ff57fddfc8649def167 (release)
timestamp 1599135103
```

Not much more I can add - for me it has been just plug and play.

nbuchwitz, 2023-02-21:
Thanks a lot! We will test with with your firmware version and also check if we can get our hands on one of the above mentioned cards.

6by9, 2023-02-21:
The Ebay link above is nearly identical to my board, although I can't guarantee it.

iluminat23, 2023-02-21:
Hmm your dmesg output shows, that your card is only running Gen1 (2,5 GT/s). Have you pinned this to Gen1 in the devicetree or anything like that?

6by9, 2023-02-21:
> Have you pinned this to Gen1 in the devicetree or anything like that?

No, this was literally a vanilla Bullseye Lite 64bit image which I updated. No changes to config.txt or device tree at all.

iluminat23, 2023-02-21:
OK, so there is something fishy here, as it should get a Gen2 link. But it seems to falls to Gen1 while doing link training.

6by9, 2023-02-22:
Checking on my Buster 5.10.74 kernel, the more complete lspci output for the host and Pericom upstream port:
```
pi@cm4nas:~ $ sudo lspci -vvv
00:00.0 PCI bridge: Broadcom Limited Device 2711 (rev 20) (prog-if 00 [Normal decode])
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 0
	Bus: primary=00, secondary=01, subordinate=05, sec-latency=0
	I/O behind bridge: 00000000-00000fff
	Memory behind bridge: c0000000-c01fffff
	Prefetchable memory behind bridge: 00000000fff00000-00000000000fffff
	Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort+ <SERR- <PERR-
	BridgeCtl: Parity- SERR+ NoISA- VGA- MAbort- >Reset- FastB2B-
		PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
	Capabilities: [48] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=1 PME-
	Capabilities: [ac] Express (v2) Root Port (Slot-), MSI 00
		DevCap:	MaxPayload 512 bytes, PhantFunc 0
			ExtTag- RBE+
		DevCtl:	Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
			RlxdOrd+ ExtTag- PhantFunc- AuxPwr+ NoSnoop+
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <2us, L1 <4us
			ClockPM+ Surprise- LLActRep- BwNot+ ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes Disabled- CommClk-
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 2.5GT/s, Width x1, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt+
		RootCtl: ErrCorrectable- ErrNon-Fatal- ErrFatal- PMEIntEna- CRSVisible+
		RootCap: CRSVisible+
		RootSta: PME ReqID 0000, PMEStatus- PMEPending-
		DevCap2: Completion Timeout: Range ABCD, TimeoutDis+, LTR+, OBFF Via WAKE# ARIFwd-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR+, OBFF Disabled ARIFwd-
		LnkCtl2: Target Link Speed: 2.5GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete-, EqualizationPhase1-
			 EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
	Capabilities: [100 v1] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		AERCap:	First Error Pointer: 00, GenCap- CGenEn- ChkCap- ChkEn-
	Capabilities: [180 v1] Vendor Specific Information: ID=0000 Rev=0 Len=028 <?>
	Capabilities: [240 v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
			  PortCommonModeRestoreTime=8us PortTPowerOnTime=10us
		L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
			   T_CommonMode=1us LTR1.2_Threshold=0ns
		L1SubCtl2: T_PwrOn=10us

01:00.0 PCI bridge: Pericom Semiconductor Device 2404 (rev 05) (prog-if 00 [Normal decode])
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Bus: primary=01, secondary=02, subordinate=05, sec-latency=0
	I/O behind bridge: 0000f000-00000fff
	Memory behind bridge: c0000000-c01fffff
	Prefetchable memory behind bridge: 00000000fff00000-00000000000fffff
	Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- <SERR- <PERR-
	BridgeCtl: Parity- SERR+ NoISA- VGA- MAbort- >Reset- FastB2B-
		PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
	Capabilities: [40] Power Management version 3
		Flags: PMEClk- DSI- D1+ D2+ AuxCurrent=375mA PME(D0+,D1+,D2+,D3hot+,D3cold+)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [5c] Vital Product Data
[... 22 lines cut ...]
			 EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
	Capabilities: [100 v1] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES- TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		AERCap:	First Error Pointer: 00, GenCap+ CGenEn- ChkCap+ ChkEn-
	Capabilities: [140 v1] Virtual Channel
		Caps:	LPEVC=0 RefClk=100ns PATEntryBits=4
		Arb:	Fixed- WRR32- WRR64- WRR128-
		Ctrl:	ArbSelect=Fixed
		Status:	InProgress-
		VC0:	Caps:	PATOffset=04 MaxTimeSlots=64 RejSnoopTrans-
			Arb:	Fixed+ WRR32- WRR64- WRR128+ TWRR128- WRR256-
			Ctrl:	Enable+ ID=0 ArbSelect=Fixed TC/VC=ff
			Status:	NegoPending- InProgress-
			Port Arbitration Table <?>
	Capabilities: [20c v1] Power Budgeting <?>
	Capabilities: [230 v1] Latency Tolerance Reporting
		Max snoop latency: 0ns
		Max no snoop latency: 0ns
```
Checking /proc/device-tree/scb/pcie@7d500000/max-link-speed, it is set to 1.
I'll check again on the 6.1 kernel.....

6by9, 2023-02-22:
6.1.12 is running as Gen2 on all active links, and no kernel panics seen.
[dmesg.txt](https://github.com/raspberrypi/linux/files/10801842/dmesg.txt)
```
pi@raspberrypi:~ $ sudo lspci -vvv
00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2711 PCIe Bridge (rev 20) (prog-if 00 [Normal decode])
	Device tree node: /sys/firmware/devicetree/base/scb/pcie@7d500000/pci@0,0
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 0
	Bus: primary=00, secondary=01, subordinate=05, sec-latency=0
	I/O behind bridge: 00000000-00000fff [size=4K]
	Memory behind bridge: c0000000-c01fffff [size=2M]
	Prefetchable memory behind bridge: 00000000fff00000-00000000000fffff [disabled]
	Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort+ <SERR- <PERR-
	BridgeCtl: Parity- SERR- NoISA- VGA- VGA16- MAbort- >Reset- FastB2B-
		PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
	Capabilities: [48] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=1 PME-
	Capabilities: [ac] Express (v2) Root Port (Slot-), MSI 00
		DevCap:	MaxPayload 512 bytes, PhantFunc 0
			ExtTag- RBE+
		DevCtl:	CorrErr- NonFatalErr- FatalErr- UnsupReq-
			RlxdOrd+ ExtTag- PhantFunc- AuxPwr+ NoSnoop+
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <2us, L1 <4us
			ClockPM+ Surprise- LLActRep- BwNot+ ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes, Disabled- CommClk-
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 5GT/s (ok), Width x1 (ok)
			TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt+
		RootCap: CRSVisible+
		RootCtl: ErrCorrectable- ErrNon-Fatal- ErrFatal- PMEIntEna- CRSVisible-
		RootSta: PME ReqID 0000, PMEStatus- PMEPending-
		DevCap2: Completion Timeout: Range ABCD, TimeoutDis+ NROPrPrP- LTR+
			 10BitTagComp- 10BitTagReq- OBFF Via WAKE#, ExtFmt- EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS- LN System CLS Not Supported, TPHComp- ExtTPHComp- ARIFwd-
			 AtomicOpsCap: Routing- 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR+ OBFF Disabled, ARIFwd-
			 AtomicOpsCtl: ReqEn- EgressBlck-
		LnkCap2: Supported Link Speeds: 2.5-5GT/s, Crosslink- Retimer- 2Retimers- DRS+
		LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
			 EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: unsupported, DRS-
			 DownstreamComp: Link Down - Not Determined
	Capabilities: [100 v1] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 00, ECRCGenCap- ECRCGenEn- ECRCChkCap- ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 00000000 00000000 00000000 00000000
		RootCmd: CERptEn- NFERptEn- FERptEn-
		RootSta: CERcvd- MultCERcvd- UERcvd- MultUERcvd-
			 FirstFatal- NonFatalMsg- FatalMsg- IntMsg 0
		ErrorSrc: ERR_COR: 0000 ERR_FATAL/NONFATAL: 0000
	Capabilities: [180 v1] Vendor Specific Information: ID=0000 Rev=0 Len=028 <?>
	Capabilities: [240 v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
			  PortCommonModeRestoreTime=8us PortTPowerOnTime=10us
		L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
			   T_CommonMode=1us LTR1.2_Threshold=0ns
[... 373 lines cut ...]
		DevCap2: Completion Timeout: Range B, TimeoutDis+ NROPrPrP- LTR-
			 10BitTagComp- 10BitTagReq- OBFF Not Supported, ExtFmt- EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS- TPHComp- ExtTPHComp-
			 AtomicOpsCap: 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR- OBFF Disabled,
			 AtomicOpsCtl: ReqEn-
		LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis+
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
			 EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: unsupported
	Capabilities: [100 v1] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
[... cut ...] lover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 00, ECRCGenCap- ECRCGenEn- ECRCChkCap- ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 00000000 00000000 00000000 00000000
	Kernel driver in use: xhci_hcd
```
(Port 02:01.0 had an accident where I dropped the switch with a card plugged in and it pulled the plastic shell of the connector off rendering it useless, so it's not connected to anything. That is likely why it is reported as degraded to 2.5GT/s)

antonbo, 2023-04-19:
I also have problems with very similar setup and crash but looks a bit different
So I have: 
CM4 + CM4 IO board as base.
PCIe connected to switch ASMedia Technology Inc. ASM1184e PCIe Switch
3 of 4 slots on switch is occupied with
1. USB3 controller: Renesas Technology Corp. uPD720202 USB 3.0 Host Controller (rev 02)
2. SATA controller: Marvell Technology Group Ltd. 88SE9215 PCIe 2.0 x1 4-port SATA 6 Gb/s Controller (rev 11)
3. SATA controller: Marvell Technology Group Ltd. 88SE9215 PCIe 2.0 x1 4-port SATA 6 Gb/s Controller (rev 11) (second one)

In case if all disks are connected only to a single SATA PCIe controller system is very stable. Uptime now 18 days with 24/7 usage. But as soon as I connect disks to the second SATA controller the system starting to crash. Usually quite easy to trigger with shell command 'sync'. But might happen itself after some time.
I was unlucky and second controller were installed at the same time when kernel was updated from 5.x to 6x so I reverted back to 5.x version. It worked for some days and than started to crash immediately after boot up.

I'm using BTRFS on all disks (but no RAID or similar).

Solution tried
dtoverlay=pcie-32bit-dma in the config.txt
and pci=nomsi in cmdline.txt

Crash screenshot (sorry, do not have serial log attached so took screen shot from monitor)
[image]

antonbo, 2023-04-19:
The crash is from 5.15, but 6.1.19 crash as well

P33M, 2023-08-07:
One proposed patch on the linux-pci list:

https://lore.kernel.org/linux-pci/385baf9dbfb6b65112803998dfc0cd6f84a5e6ba.1691296765.git.lukas@wunner.de/

I think the author is correct in that accesses when the link is doing a retrain result in fatal errors, but I also think the race condition is unavoidable. Bad signal integrity could cause link-down at any time, and there's no notification that state has changed.

It's also half the story - there is an actual bug in how the probe code is enabling CRS software visibility, which results in fatal errors if an endpoint _does_ have an active link but has not yet completed internal initialisation and responds with Retry to a config read. The call to pci_enable_crs() which suppresses that particular error response is done before the link is established, so PERST# is asserted and root complex configuration registers are held in reset.

I'm waiting on hardware that should let me replicate the failure case that the mailing list alludes to.

l1k, 2023-08-13:
@P33M The patch you've linked to makes enumeration work reliably with the ASMedia ASM1184e, but timeouts (and thus kernel panics) persist with the Pericom PI7C9X2G404SL (occasionally) and with the Pericom PI7C9X2G404SV (very often).

With those chips, particularly with the latter, enumeration sometimes initially works but subsequently timeouts occur out of the blue when accessing devices downstream of that switch. Also, sometimes the switch itself is enumerated but downstream devices are not.

Since the issue is easy to reproduce with the PI7C9X2G404SV, I recommend testing with that one. The chip is interesting because it consumes less power than the others and conforms to industrial requirements (temperature and otherwise). Unfortunately eval boards with the PI7C9X2G404SV are hard to come by and expensive. @prosenberger has a single one, but it's loaned, so he's very hesitant to give it away. That eval board absolutely does not work reliably in the CM4 IO board, but shows no issues at all in a Sitara AM64 EVM or in an x86 PC. So there must be something fishy going on with the CM4 PCIe controller.

By accessing the switch's registers after a kernel panic on enumeration (over SMBUS from a different machine) it has been discovered that the Common Clock Configuration bit in the Link Control Register of the Switch Upstream Port is *zero*. In the working case, i.e., without kernel panic on enumeration, that bit is *set*. So it seems that the switch has difficulty recovering the reference clock presented by the BCM2711 PCIe controller.

Analysis of that clock with a Keysight high-speed oscilloscope has indeed exposed issues: For one, the clock changes multiple times after power-on, oscillating around the expected frequency. Second, there's a "capacitor effect" visible whereby the clock takes a while to "come down" to the expected frequency, presumably because the AC coupling needs time to charge.

It has been tried to work around the unclean clock signal by delaying release of PERST of the switch. That didn't help. It has also been tried to not show the clock signal to the switch until it has stabilized (i.e. keeping the clock visible to the switch in High-Z for a while). Didn't help either. It has been tried to shift the clock voltage up a little to open the "eye" of the signal. Didn't help.

Analysis of the communication between the CM4 and the PI7C9X2G404SV with a Teledyne Lecroy protocol analyzer initially didn't work at all. After desoldering the AC coupling on the CM4, the analyzer started to see communication, but there were no TLPs visible that would point to an error. It seems the issue is at the physical layer, not the TLP error.

@CyrilBrulebois has commented on #5455 and on the mailing list that updating the CM4 EEPROM helps. This was tried as well. It didn't help with the PI7C9X2G404SV.

Again, no issues if the switch eval board is plugged into an AM64 EVM or x86 PC. It's only the CM4 that exhibits these issues.

nbuchwitz, 2023-08-13:
Thanks for the summary of the status quo and esp. the things we tried so far, @l1k !

l1k, 2023-08-13:
[image]

Reference clock exposed by the BCM2711 PCIe controller, showing a "capacitor effect" (green signal, top-left corner) and showing that the clock oscillates between 96 MHz and 106 MHz before settling at 100 MHz after about 391 usec (which is well after PERST has been released, yellow signal at the bottom).

l1k, 2023-08-13:
I forgot to mention that forcing de-emphasis to -6dB (instead of the default -3.5dB) in the Link Control 2 register of the BCM2711 Root Port seemed to slightly improve reliability with the PI7C9X2G404SV. (The kernel would predictably crash when flashing firmware onto a Renesas USB-Controller attached below the switch with -3.5dB, but not with de-emphasis at -6dB. Nevertheless crashes would still occur later on.)

This seems to further suggest that there's a signal issue at play.

florincosta, 2023-09-18:
Hi,
@l1k I'm also having problems with the CM4 and a similar PCIe switch, the PI7C9X2G404EV, I think only the IC package differs and the PI7C9X2G404EV it's basically the same as PI7C9X2G404SV.
I've developed my on board and the problem is that the board sometimes hangs during boot, I've only tested it for a couple of days, so far no kernel panics, running latest 64 bit Raspberry pi OS and latest CM4 EEPROM image.

I've probed the CM4 PERST and PCIE REFCLK, it seems that REFCLK is enabled about 500us before de-asserting PERST, see scope capture below:
  
[image]

I've noticed that in the PI7C9X2G404EV datasheet the REFCLK should already be stable for about 100ms when PERST is de-asserted:

[image]

The issue is similar to the one here: https://community.toradex.com/t/verdin-imx8mm-clock-reset-sequencing/15851/11

I want to add the 100ms of REFCLK stabilization before de-asserting PERST, I'm not familliar with the Broadcom PCIe kernel driver, can somebody point me in the right direction?
I could also add the hardware RC delay.

danclive, 2023-09-19:
@florincosta 
Hi, the 100ms delay has been added in version 6.2.y. I don't have the hardware, so I'm not sure if the problem is still there!

https://github.com/raspberrypi/linux/blob/rpi-6.2.y/drivers/pci/controller/pcie-brcmstb.c#L1043-L1047

pelwell, 2023-09-19:
@P33M Any concerns about us back-porting https://github.com/raspberrypi/linux/commit/3ae140ad827b359bc4fa7c7985691c4c1e3ca8f4 to rpi-6.1.y? It applies cleanly.

P33M, 2023-09-19:
No, but there's some terminology in this thread that needs clearing up - 

Referring to the CEM specification v4 rev 1.0, figure 8, the critical time periods are 1 through 5.

1. 3v3aux power on to SMBUS activity - not applicable, the CM4IO board doesn't connect SMBUS, and in any case there is a protracted boot process required before the hardware is available for software to use.
2. Main power available to PERST# deassert (Tpvperl) - 100ms - satisfied as the boot process, whether or not the PCIe core is accessed from the firmware or Linux, takes >5 seconds.
3. Refclk stable to PERST# deassert (Tperst-clk) - 100us - this delay is maintained by an internal state machine within the PCIe root complex.
4. PERST# inactive to link out of electrical idle - not specified in the CEM but >0 seconds and <20ms in the base specification. This sequence is maintained by an internal state machine within the PCIe root complex.
5. PERST# inactive to JTAG activity - not applicable, the CM4IO board doesn't connect JTAG pins.

The delay referred to in the patch is inaccurately described/placed. After the previous line, the link will have autonomously activated and in general will be in L0 on the first or second iteration of the loop checking for link-up. 

Referring to the base specification v4.0 6.6.1, the context describes a time period after the link is active where software must not send any configuration requests to the downstream device, unless a Readiness Notification is received. I doubt the RC can generate an interrupt for a RN therefore the 100ms delay is required - but this should be after L0. In practice it doesn't matter.

There is an ultimate timeout of 1 second after the first attempt to issue a configuration request for the device for it to return a successful completion. This delay is catered for in the various enumeration mechanisms in the upper Linux PCI layer.

P33M, 2023-09-19:
If the switch really needs 1000x the minimum required Tperst-clk, then there's an issue - there's no easy way to extend this time.

pelwell, 2023-09-19:
Ta.

florincosta, 2023-09-19:
> If the switch really needs 1000x the minimum required Tperst-clk, then there's an issue - there's no easy way to extend this time.

Yeah, it seems the PI7C9X2G404EV switch needs a Tperst-clk of 100ms, although in the CEM spec this Tperst-clk is only specified as a minimum of 100us, maximum not specified, so the PI7C9X2G404EV requires 100ms of REFCLK stabilization **before** de-asserting the PERST# signal.. 
The 100ms from 6f634d7efb8876e5953c30c0a613aaa5f575fe05, are added **after** the PERST# is de-asserted, so this does not help in this case.

@P33M thanks for the info regarding Tperst-clk 100us and where is handled.

I still did try @danclive suggestion with 6.2.16-v8+ but got the same results:
[image]

So I guess now there is only on way to fix this, add the 100ms delay the hardway :)

[image]

P33M, 2023-09-19:
I think I may have found a way to optionally delay PERST# deassertion with a magic register bit in PCIE_MISC_HARD_DEBUG - bit 3 directly controls the output pad. If I set this bit prior to the call to `pcie->perst_set(pcie, 0)` and add a `msleep(n)` before resetting this bit, the refclk becomes active approximately n milliseconds before I see the reset line rise.

It may be the case that the RC is trying to initiate training during this delay which would violate constraint 4 above but that is probably tolerable.

P33M, 2023-09-19:
Nope, delaying the deassert in this manner doesn't affect when the RC starts link training (~300us after). It's a pretty tall ask to make the driver do this for every endpoint (thus delaying boot) so another devicetree knob is needed.

P33M, 2023-09-20:
On a CM4 try testing `sudo rpi-update pulls/5609` which should add the required delay.

florincosta, 2023-09-20:
> On a CM4 try testing `sudo rpi-update pulls/5609` which should add the required delay.

Tried it, it works, there is a Tperst-clk of aprox 100ms, 97ms to be precise.
Thanks!

[image]

 I will now test it on my CM4 carrier board.

iluminat23, 2023-09-27:
Hi,

I've tested `sudo rpi-update pulls/5609` with this setup:
* CM4 IO-Board
* PI7C9X2G404EV eval board
  *  Microchip LAN4730
  * Intel Corporation 82572EI Gigabit Ethernet Controller
  * Renesas Technology Corp. uPD720201 USB 3.0

We tested this with ~3800 cold boots.
* all devices where detected
* not panic
* no other errors

We also tested this with ~80 reboots:
* the LAN7430 is missing after every reboot
* no panic
* no other errors

We had no time to investigate the LAN7430 issue any further. But I wanted to report back our test results.

P33M, 2023-10-26:
Can you re-do `sudo rpi-update pulls/5609` then add `dtparam=pcie_tperst_clk_ms=100` to /boot/firmware/config.txt, then redo your reboot test? It moves the hack to a configurable override.

theodric, 2023-10-30:
I merged the changes that were present in #5609 as of 24-Oct-2023 into 5.15.92 and have been beating the snot out of it for 6 days with no further issues. I've sent 5+ terabytes across the switch each way between USB3-attached disks and SATA-attached disks. Previously I was getting crashes at least every few hours/few hundred gigs when attempting this. 

I'm running the following on a CM4 + official I/O board:
```
00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2711 PCIe Bridge (rev 20)
01:00.0 PCI bridge: Pericom Semiconductor PI7C9X2G304 EL/SL PCIe2 3-Port/4-Lane Packet Switch (rev 05)
02:01.0 PCI bridge: Pericom Semiconductor PI7C9X2G304 EL/SL PCIe2 3-Port/4-Lane Packet Switch (rev 05)
02:02.0 PCI bridge: Pericom Semiconductor PI7C9X2G304 EL/SL PCIe2 3-Port/4-Lane Packet Switch (rev 05)
03:00.0 SATA controller: ASMedia Technology Inc. Device 1166 (rev 02)
04:00.0 USB controller: VIA Technologies, Inc. VL805 USB 3.0 Host Controller (rev 01)
```

iluminat23, 2023-10-31:
@P33M This didn't change any thing for the reboot issue.

cold boot:
```
pi@raspberrypi:~ $ dmesg | grep -a -E "pci 0000:[0-9][0-9]:[0-9][0-9]\.0: \["
[    1.380599] pci 0000:00:00.0: [14e4:2711] type 01 class 0x060400
[    1.602178] pci 0000:01:00.0: [12d8:b404] type 01 class 0x060400
[    1.603560] pci 0000:02:01.0: [12d8:b404] type 01 class 0x060400
[    1.604571] pci 0000:02:02.0: [12d8:b404] type 01 class 0x060400
[    1.605536] pci 0000:02:03.0: [12d8:b404] type 01 class 0x060400
[    1.609895] pci 0000:03:00.0: [1912:0014] type 00 class 0x0c0330
[    1.611158] pci 0000:04:00.0: [1055:7430] type 00 class 0x020000
[    1.612654] pci 0000:05:00.0: [8086:107d] type 00 class 0x020000
```

reboot:
```
pi@raspberrypi:~ $ dmesg | grep -a -E "pci 0000:[0-9][0-9]:[0-9][0-9]\.0: \["
[    1.382958] pci 0000:00:00.0: [14e4:2711] type 01 class 0x060400
[    1.606017] pci 0000:01:00.0: [12d8:b404] type 01 class 0x060400
[    1.607403] pci 0000:02:01.0: [12d8:b404] type 01 class 0x060400
[    1.608417] pci 0000:02:02.0: [12d8:b404] type 01 class 0x060400
[    1.609371] pci 0000:02:03.0: [12d8:b404] type 01 class 0x060400
[    1.613736] pci 0000:03:00.0: [1912:0014] type 00 class 0x0c0330
[    1.615353] pci 0000:05:00.0: [8086:107d] type 00 class 0x020000
```

```
pi@raspberrypi:~ $ cat /boot/.firmware_revision 
9ffba7f9e53ab64d6e3583d7bf4ece60a19c6237
```

```
pi@raspberrypi:~ $ sudo fdtdump /sys/firmware/fdt | grep tperst

**** fdtdump is a low-level debugging tool, not meant for general use.
**** If you want to decompile a dtb, you probably want
****     dtc -I dtb -O dts <filename>

            brcm,tperst-clk-ms = <0x00000064>;
        pcie_tperst_clk_ms = [00 00 00 3e 62 72 63 6d 2c 74 70 65 72 73 74 2d 63 6c 6b 2d 6d 73 3a 30 00];
```

P33M, 2023-10-31:
Is the behaviour the same if you don't use a soft reboot, and instead forcibly reset the CM4 via the RUN pin?

iluminat23, 2023-11-02:
My test setup failed me and I'm not able to do any tests for now. I'm also not in the office for the next weeks and will not be able to fix the setup.

P33M, 2023-11-23:
Regarding the disappearing LAN7430, I've found that the default CM4 boot order includes NVMe probe, which will wake up the link without the requisite Tperst_clk time. Please confirm whether or not 0x4 or 0x6 appears anywhere in your BOOT_ORDER eeprom config.

iluminat23, 2023-11-29:
Yes on the most of my CMs I have not touched the boot order and thus NVMe boot is enabled.