raspberrypi/linux #5353: Screen blanking not working

snawgj, 2023-02-17:
### Describe the bug

I've been asked to log this issue here by the team at info@raspberrypi.com.
Since running the updates released on 14 Feb 2023 screen blanking is failing to run. The desktop continues to appear and no blanking takes place. 
This behaviour has now happened on three separate Raspberry Pis - there were no issues prior to the 14 Feb update. The systems are:
Pi 3b running Bullseye 32 bit
Pi 4 running Bullseye 32 bit
Pi 400 running Bullseye 64 bit
I just used the Pi 3b (last updated prior to 14 Feb) to confirm my previous observations. Booted up, waited 10 minutes and the screen blanked as expected. Ran the updates (including 14 Feb updates), rebooted and screen blanking is not kicking in.

### Steps to reproduce the behaviour

I've found this behaviour to be totally reproducible- now happened on 3 different and separate systems.

### Device (s)

Raspberry Pi 3 Mod. B, Raspberry Pi 4 Mod. B, Raspberry Pi 400

### System

Raspberry Pi OS Bullseye 32 and 64 bit versions. Not tested on Buster.
Jan 5 2023 10:48:31
5.15.84-v7+ #1613

popcornmix, 2023-02-18:
See [url=https://forums.raspberrypi.com/viewtopic.php?p=2082114#p2082114]here[/url]

sudo-splinter-cell, 2023-03-08:
"xset dpms" command returns:
"server does not have extension for dpms option".

"xset -q" command returns:
"DPMS (Energy Star):
  Server does not have the DPMS Extension"

Uro1, 2023-03-18:
I was also afflicted by this screen blanking issue, the latest updates pulled via apt today [18/03/2023] appear to have corrected it with my 7" touchpanel Pi4b running PiOS Bullseye arm64 now sucessfully blanking the screen again.

bembudo, 2023-12-08:
screen blanking also does not work on Raspberry pi 5, PI OS Bookworm

popcornmix, 2023-12-08:
I'll close this issue, as it was caused by an update of the xserver-xorg-core package, which was subsequently fixed (back in March).

There are issues for other causes of blanking failure (e.g. hotplug waggling waking up display).
See also: https://github.com/raspberrypi/bookworm-feedback/issues/67