raspberrypi/linux #5364: rpi-proto sound does not work with kernel 6.1.x

moroslantia, 2023-02-28:
### Describe the bug

config.txt configured to enable rpi-proto support

`
dtoverlay=i2s-mmap
dtoverlay=rpi-proto
`
System reports

`
root@pi:/root# cat /proc/asound/cards
--- no soundcards ---

root@pi:/root# cat /sys/kernel/debug/devices_deferred
soc:sound
`

### Steps to reproduce the behaviour

* Enable rpi-proto support in config.txt
* load a 6.1.x kernel

### Device (s)

Raspberry Pi CM3+

### System

Raspberry Pi reference 2022-01-14

Dec  1 2021 15:07:06
Copyright (c) 2012 Broadcom
version 71bd3109023a0c8575585ba87cbb374d2eeb038f (clean) (release) (start)

Linux pi 6.1.13-rt7-v7+ #1 SMP PREEMPT_RT Sun Feb 26 18:19:58 UTC 2023 armv7l GNU/Linux

pelwell, 2023-02-28:
For the casual observer, this is a continuation of a discussion on the Linux 6.1 topic of the RPi Forums, starting here: https://forums.raspberrypi.com/viewtopic.php?p=2082693#p2082693

pelwell, 2023-02-28:
Found it. RPi-Proto (not one of ours) support was broken in rpi-5.19.y as a result of this commit: 9dc15f81baf273b5aaaa3302ee8faacd78f361fd
For the reasons specified in the commit message, this split the user-facing config setting for WM8731 into two - SND_SOC_WM8731_I2C and SND_SOC_WM8731_SPI. The author appears to have been careful and gone through the Kconfig rules, changing uses of SND_SOC_WM8731 to one or both of the new settings, but because our soundcard support is a downstream addition it wasn't updated. Also, my inter-branch defconfig checks didn't spot the change because the original setting is implied by the setting for the RPi Proto card, which means it never appeared in the defconfigs.

https://github.com/raspberrypi/linux/pull/5365 changes the Kconfig settings for the Rpi-PROTO and audioinjector.net Pi (Hi, @flatmax) to select the I2C mode. Once the auto-builds have completed (give it an hour), you should be able to install the updated kernels by running `sudo rpi-update pulls/5365` (remembering that this is at-best beta software and should be treated with caution).

pelwell, 2023-02-28:
The auto-builds are complete now and ready to try.

moroslantia, 2023-02-28:
Applied changes work and i get the soundcard up and running
Thank's a lot for the quick help.

closed 2023-02-28 by moroslantia